openwrt/openwrt #13841: compile dwarves fails to build

datahost, 2023-10-31:
### Describe the bug

While trying to build Openwrt for my Dynalink WRX36 today, I was surprised by an error. I checked out OpenWRT via git, updated the feeds and selected my target and device via "make menuconfig. In the "Advanced configuration options" I only selected "Compile all host tools" and "Download folder", "Log folder" and "ccache".

[build_error.log](https://github.com/openwrt/openwrt/files/13220868/build_error.log)

### OpenWrt version

r24264-c997634c01

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Dynalink DL-WRX36

### Image kind

Self-built image

### Steps to reproduce

git clone https://git.openwrt.org/openwrt/openwrt.git october-last-build
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
make V=sc

### Diffconfig

```text
CONFIG_TARGET_qualcommax=y
CONFIG_TARGET_qualcommax_ipq807x=y
CONFIG_TARGET_qualcommax_ipq807x_DEVICE_dynalink_dl-wrx36=y
CONFIG_DEVEL=y
CONFIG_BUILD_ALL_HOST_TOOLS=y
CONFIG_BUILD_LOG=y
CONFIG_BUILD_LOG_DIR="/home/builduser/openwrt/buildlog"
CONFIG_CCACHE=y
CONFIG_CCACHE_DIR="/home/builduser/openwrt/.ccache"
CONFIG_DOWNLOAD_FOLDER="/home/builduser/openwrt/download"
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

darksh71, 2023-10-31:
Here's a concise troubleshooting checklist:

🖥️ Check system resources (CPU, RAM, storage).
📜 Review the terminal output for errors or stuck points.
💾 Verify sufficient disk space in your build environment.
🧹 Consider running a clean build with make clean and make dirclean.
🧰 Ensure you have the necessary build dependencies installed.
🔄 Check for existing background build processes.
📋 Examine log files, like "build_error.log," for details.
💻 Try building on a different system or clean environment.
🌐 Consult OpenWrt community resources for solutions.
🐞 Report the issue on the OpenWrt GitHub repository with detailed information.

xlighting2017, 2023-11-01:
adding logs of the same issue that happen to me

```
[27/57] Building C object CMakeFiles/dwarves.dir/elf_symtab.c.o
[28/57] Building C object CMakeFiles/dwarves.dir/rbtree.c.o
[29/57] Linking C shared library libdwarves.so.1.0.0
FAILED: libdwarves.so.1.0.0 
: && /workdir/openwrt/staging_dir/host/bin/gcc -fPIC -O2 -I/workdir/openwrt/staging_dir/host/include -pthread -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wall -O2  -L/workdir/openwrt/staging_dir/host/lib -shared -Wl,-soname,libdwarves.so.1 -o libdwarves.so.1.0.0 CMakeFiles/bpf.dir/lib/bpf/src/bpf.c.o CMakeFiles/bpf.dir/lib/bpf/src/bpf_prog_linfo.c.o CMakeFiles/bpf.dir/lib/bpf/src/btf.c.o CMakeFiles/bpf.dir/lib/bpf/src/btf_dump.c.o CMakeFiles/bpf.dir/lib/bpf/src/gen_loader.c.o CMakeFiles/bpf.dir/lib/bpf/src/hashmap.c.o CMakeFiles/bpf.dir/lib/bpf/src/libbpf.c.o CMakeFiles/bpf.dir/lib/bpf/src/libbpf_errno.c.o CMakeFiles/bpf.dir/lib/bpf/src/libbpf_probes.c.o CMakeFiles/bpf.dir/lib/bpf/src/linker.c.o CMakeFiles/bpf.dir/lib/bpf/src/netlink.c.o CMakeFiles/bpf.dir/lib/bpf/src/nlattr.c.o CMakeFiles/bpf.dir/lib/bpf/src/relo_core.c.o CMakeFiles/bpf.dir/lib/bpf/src/ringbuf.c.o CMakeFiles/bpf.dir/lib/bpf/src/str_error.c.o CMakeFiles/bpf.dir/lib/bpf/src/strset.c.o CMakeFiles/bpf.dir/lib/bpf/src/usdt.c.o CMakeFiles/dwarves.dir/dwarves.c.o CMakeFiles/dwarves.dir/dwarves_fprintf.c.o CMakeFiles/dwarves.dir/gobuffer.c.o CMakeFiles/dwarves.dir/ctf_loader.c.o CMakeFiles/dwarves.dir/libctf.c.o CMakeFiles/dwarves.dir/btf_encoder.c.o CMakeFiles/dwarves.dir/btf_loader.c.o CMakeFiles/dwarves.dir/dwarf_loader.c.o CMakeFiles/dwarves.dir/dutil.c.o CMakeFiles/dwarves.dir/elf_symtab.c.o CMakeFiles/dwarves.dir/rbtree.c.o  -Wl,-rpath,:::::::::::::::::::::::::::::::::::::  -Wl,-Bstatic  -ldw  -lelf  -lz  -Wl,-Bdynamic && :
make[3]: *** [Makefile:36: /workdir/openwrt/build_dir/host/dwarves-1.25/.built] Error 1
/usr/bin/ld: /workdir/openwrt/staging_dir/host/lib/libdw.a(dwarf_error.o): relocation R_X86_64_TPOFF32 against `global_error' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[3]: Leaving directory '/workdir/openwrt/tools/dwarves'
time: tools/dwarves/compile#10.64#0.94#11.56
    ERROR: tools/dwarves failed to build.
make[2]: *** [tools/Makefile:230: tools/dwarves/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [tools/Makefile:226: /workdir/openwrt/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyyyynyyyyyyynyyynnnynyynnynnyyyyyynyyyyyyyyyynynynnyyyyyyy] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:232: world] Error 2
Error: Process completed with exit code 2.
```

btw: target ipq40xx does not have this, because tools/dwarves is not included in ipq40xx, I presume?

ynezz, 2023-11-01:
@datahost @xlighting2017 @mirthlesslk can you provide details about your build host, what Linux distro is that? Just a wild guess, it might be related to the GCC/linker version? It builds fine here with `gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0`, but seems to fail with Debian 11 https://github.com/openwrt/openwrt/actions/runs/6697935677/job/18199086487#step:7:159

@nbd168 any idea? I suspect 5331e85d96

@datahost @xlighting2017 @mirthlesslk can you try to `git revert 5331e85d96`? It might be a workaround for that issue.

xlighting2017, 2023-11-01:
Hi ynezz:
please see my 2 cents below:

1. my log comes from github actions / ubuntu-latest, which  should be Ubuntu 22.04.3 LTS with gcc11;
2. with the same config, I build on Oct 30 works fine(which is equal to b3d2008f92f15ff776ad6c4f6d113b0cda1bd2d0), so I presume the revert should fix it...

help it helps

ynezz, 2023-11-01:
For anyone interested looking into it, here is a reliable reproducer: 

```
podman run --rm -it \
  ghcr.io/openwrt/buildbot/buildworker-v3.8.0:v9 sh -c " \
    cd $HOMEDIR && git clone https://github.com/openwrt/openwrt && cd openwrt && \
    printf 'CONFIG_DEVEL=y\nCONFIG_BUILD_ALL_HOST_TOOLS=y\nCONFIG_BUILD_LOG=y\n' > .config && \
    make defconfig && make -j$(nproc) V=sc tools/compile ; \
    ./.github/workflows/scripts/show_build_failures.sh ; bash"
```

robimarko, 2023-11-01:
Reverting:
1d6f9bdfc052b83434dc3fa809e1e6ee6329864f
and
5331e85d96346be4200ce206b2f4373e23af0f6f

Fixes the issue

rockdrilla, 2023-11-01:
Still hitting FTBFS:

```
$ make -j 1 tools/dwarves/compile V=sc
# <...>
[29/57] : && /tmp/fork.openwrt/staging_dir/host/bin/gcc -fPIC -O2 -I/tmp/fork.openwrt/staging_dir/host/include -pthread -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wall -O2  -L/tmp/fork.openwrt/staging_dir/host/lib -shared -Wl,-soname,libdwarves.so.1 -o libdwarves.so.1.0.0 CMakeFiles/bpf.dir/lib/bpf/src/bpf.c.o CMakeFiles/bpf.dir/lib/bpf/src/bpf_prog_linfo.c.o CMakeFiles/bpf.dir/lib/bpf/src/btf.c.o CMakeFiles/bpf.dir/lib/bpf/src/btf_dump.c.o CMakeFiles/bpf.dir/lib/bpf/src/gen_loader.c.o CMakeFiles/bpf.dir/lib/bpf/src/hashmap.c.o CMakeFiles/bpf.dir/lib/bpf/src/libbpf.c.o CMakeFiles/bpf.dir/lib/bpf/src/libbpf_errno.c.o CMakeFiles/bpf.dir/lib/bpf/src/libbpf_probes.c.o CMakeFiles/bpf.dir/lib/bpf/src/linker.c.o CMakeFiles/bpf.dir/lib/bpf/src/netlink.c.o CMakeFiles/bpf.dir/lib/bpf/src/nlattr.c.o CMakeFiles/bpf.dir/lib/bpf/src/relo_core.c.o CMakeFiles/bpf.dir/lib/bpf/src/ringbuf.c.o CMakeFiles/bpf.dir/lib/bpf/src/str_error.c.o CMakeFiles/bpf.dir/lib/bpf/src/strset.c.o CMakeFiles/bpf.dir/lib/bpf/src/usdt.c.o CMakeFiles/dwarves.dir/dwarves.c.o CMakeFiles/dwarves.dir/dwarves_fprintf.c.o CMakeFiles/dwarves.dir/gobuffer.c.o CMakeFiles/dwarves.dir/ctf_loader.c.o CMakeFiles/dwarves.dir/libctf.c.o CMakeFiles/dwarves.dir/btf_encoder.c.o CMakeFiles/dwarves.dir/btf_loader.c.o CMakeFiles/dwarves.dir/dwarf_loader.c.o CMakeFiles/dwarves.dir/dutil.c.o CMakeFiles/dwarves.dir/elf_symtab.c.o CMakeFiles/dwarves.dir/rbtree.c.o  -Wl,-rpath,::::::::::::::::::::::::::::::::::::::::::  -Wl,-Bstatic  -ldw  -lelf  -lz  -Wl,-Bdynamic && :
FAILED: libdwarves.so.1.0.0
: && /tmp/fork.openwrt/staging_dir/host/bin/gcc -fPIC -O2 -I/tmp/fork.openwrt/staging_dir/host/include -pthread -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wall -O2  -L/tmp/fork.openwrt/staging_dir/host/lib -shared -Wl,-soname,libdwarves.so.1 -o libdwarves.so.1.0.0 CMakeFiles/bpf.dir/lib/bpf/src/bpf.c.o CMakeFiles/bpf.dir/lib/bpf/src/bpf_prog_linfo.c.o CMakeFiles/bpf.dir/lib/bpf/src/btf.c.o CMakeFiles/bpf.dir/lib/bpf/src/btf_dump.c.o CMakeFiles/bpf.dir/lib/bpf/src/gen_loader.c.o CMakeFiles/bpf.dir/lib/bpf/src/hashmap.c.o CMakeFiles/bpf.dir/lib/bpf/src/libbpf.c.o CMakeFiles/bpf.dir/lib/bpf/src/libbpf_errno.c.o CMakeFiles/bpf.dir/lib/bpf/src/libbpf_probes.c.o CMakeFiles/bpf.dir/lib/bpf/src/linker.c.o CMakeFiles/bpf.dir/lib/bpf/src/netlink.c.o CMakeFiles/bpf.dir/lib/bpf/src/nlattr.c.o CMakeFiles/bpf.dir/lib/bpf/src/relo_core.c.o CMakeFiles/bpf.dir/lib/bpf/src/ringbuf.c.o CMakeFiles/bpf.dir/lib/bpf/src/str_error.c.o CMakeFiles/bpf.dir/lib/bpf/src/strset.c.o CMakeFiles/bpf.dir/lib/bpf/src/usdt.c.o CMakeFiles/dwarves.dir/dwarves.c.o CMakeFiles/dwarves.dir/dwarves_fprintf.c.o CMakeFiles/dwarves.dir/gobuffer.c.o CMakeFiles/dwarves.dir/ctf_loader.c.o CMakeFiles/dwarves.dir/libctf.c.o CMakeFiles/dwarves.dir/btf_encoder.c.o CMakeFiles/dwarves.dir/btf_loader.c.o CMakeFiles/dwarves.dir/dwarf_loader.c.o CMakeFiles/dwarves.dir/dutil.c.o CMakeFiles/dwarves.dir/elf_symtab.c.o CMakeFiles/dwarves.dir/rbtree.c.o  -Wl,-rpath,::::::::::::::::::::::::::::::::::::::::::  -Wl,-Bstatic  -ldw  -lelf  -lz  -Wl,-Bdynamic && :
/usr/bin/ld: /tmp/fork.openwrt/staging_dir/host/lib/libdw.a(dwarf_error.o): relocation R_X86_64_TPOFF32 against `global_error' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[2]: *** [Makefile:36: /tmp/fork.openwrt/build_dir/host/dwarves-1.25/.built] Error 1
make[2]: Leaving directory '/tmp/fork.openwrt/tools/dwarves'
time: tools/dwarves/compile#6.99#0.51#7.55
    ERROR: tools/dwarves failed to build.
make[1]: *** [tools/Makefile:228: tools/dwarves/compile] Error 1
make[1]: Leaving directory '/tmp/fork.openwrt'
make: *** [/tmp/fork.openwrt/include/toplevel.mk:232: tools/dwarves/compile] Error 2
```

rockdrilla, 2023-11-01:
PS: `tools/elfutils` and `tools/dwarves` are building now.

datahost, 2023-11-01:
I can confirm, thank you

closed 2023-11-01 by nbd168