openwrt/openwrt #13847: 23.05.0 IPv4 PPPoE software flow offload not working

InsaneKnight, 2023-11-01:
### Describe the bug

The issue described in #10224 remains in 23.05.0.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v2

### Image kind

Self-built image

### Steps to reproduce

See #10224

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-01:
Please check with tcpdump (wireshark) on the client if replying SYN+ACK packets have MSS adjusted down from LAN default in SYN. The linked issue looks very sporadic at best with 3x dispersed results claimed then slow other time fast.

jow-, 2023-11-01:
Duplicate of #10224

closed 2023-11-01 by jow-