openwrt/openwrt #13859: iptables-nft: Problem with NFT modules in 23.05.0

LupeChristoph, 2023-11-02:
### Describe the bug

I'm trying to upgrade my router from 21.02.2. I'm not using the OpenWRT firewall but a script generated by fwbuilder that has thrived since I converted from FreeBSD to OpenWRT Backfire in 2011.

The script uses the conntrack module. I tried several ways to add rules to NFT without writing them in NFT (fwbuilder has been more or less unmaintained for years now, so I can't expect a new version supporting NFT directly any day now :frowning: ).

The shell script generated by fwbuilder that works on OpenWRT 21.02.2 fails on 23.05.0. I'm looking for a way to make it work.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ramips/mt7621

### Device

Linksys EA7300 v2

### Image kind

Official downloaded image

### Steps to reproduce

This command fails:
```# iptables-nft -w -A INPUT -m conntrack --ctstate ESTABLISHED,RELATED -j ACCEPT```

### Actual behaviour

```iptables v1.8.8 (nf_tables): Couldn't load match `conntrack':No such file or directory```
It fails because of a missing /usr/lib/iptables/libxt_conntrack.so. Here is the relevant part of an strace:
```
statx(AT_FDCWD, "/usr/lib/iptables/libipt_conntrack.so", AT_STATX_SYNC_AS_STAT, STATX_BASIC_STATS, 0x7fcd1300) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/usr/lib/iptables/libxt_conntrack.so", AT_STATX_SYNC_AS_STAT, STATX_BASIC_STATS, 0x7fcd1300) = -1 ENOENT (No such file or directory)
writev(2, [{iov_base="iptables v1.8.8 (nf_tables): ", iov_len=29}, {iov_base=NULL, iov_len=0}], 2iptables v1.8.8 (nf_tables): ) = 29
writev(2, [{iov_base="Couldn't load match `conntrack':"..., iov_len=58}, {iov_base=NULL, iov_len=0}], 2Couldn't load match `conntrack':No such file or directory
) = 58
```
iptables-nft does not care about the module name, it just looks it up.

Incidentally, this line works with Debian Bullseye (11) and Ubuntu Lunar Lobster (23.04):
```
# iptables-nft -w -A INPUT -m conntrack --ctstate ESTABLISHED,RELATED -j ACCEPT
# iptables-save | grep conntrack
-A INPUT -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
```

### Expected behaviour

iptables-nft should generate the NFTables code and load it into the ruleset.

### Additional info

iptables-translate does not generate useable NFT code. I'm creating a separate ticket for that.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-03:
Restore this file like in /etc/rc.local via `nft -f /etc/ipt-compat-nft.txt`
[ipt-compat-nft.txt](https://github.com/openwrt/openwrt/files/13248813/ipt-compat-nft.txt)

obtained from debian 10 via iptables-legacy-restore with all 5 default chains empty /iptables-legacy-save iptables-nft-restore ; nft list ruleset

[empty-iptables.txt](https://github.com/openwrt/openwrt/files/13248856/empty-iptables.txt)

(dont close this, it is worth adding blank tables to make iptables-nft work at all)

jow-, 2023-11-03:
On a freshly booted OpenWrt 23.05.0 VM:

```
root@OpenWrt:/# iptables-nft -w -A INPUT -m conntrack --ctstate ESTABLISHED,RELATED -j ACCEPT
iptables v1.8.8 (nf_tables): Couldn't load match `conntrack':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:/# opkg install kmod-ipt-conntrack
Installing kmod-ipt-conntrack (5.15.134-1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/x86/64/packages/kmod-ipt-conntrack_5.15.134-1_x86_64.ipk
Configuring kmod-ipt-conntrack.
[  106.838546] kmodloader: loading kernel modules from /etc/modules.d/*
[  106.853433] kmodloader: done loading kernel modules from /etc/modules.d/*
root@OpenWrt:/# iptables-nft -w -A INPUT -m conntrack --ctstate ESTABLISHED,RELATED -j ACCEPT
root@OpenWrt:/# iptables-save | grep conntrack
-A INPUT -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
root@OpenWrt:/# 
```

closed 2023-11-03 by jow-

LupeChristoph, 2023-11-03:
Trying the rules file @brada4 sent:
```
root@splat:~# nft -f ipt-compat-nft.txt 
root@splat:~# iptables-nft -w -A INPUT -m conntrack --ctstate ESTABLISHED,RELATED -j ACCEPT
iptables v1.8.8 (nf_tables): Couldn't load match `conntrack':No such file or directory
```

Nope.

---
Thanks @jow, that's very interesting because:
```
root@splat:~# opkg list-installed | grep conntrack
kmod-nf-conntrack - 5.15.134-1
kmod-nf-conntrack6 - 5.15.134-1
```
So let's try to remove and install again:
```
root@splat:~# opkg remove kmod-nf-conntrack kmod-nf-conntrack6
No packages removed.
Collected errors:
 * print_dependents_warning: Package kmod-nf-conntrack is depended upon by packages:
 * print_dependents_warning:    kmod-nf-flow
 * print_dependents_warning:    kmod-nf-nat
 * print_dependents_warning:    kmod-nf-conntrack6
 * print_dependents_warning: These might cease to work if package kmod-nf-conntrack is removed.
```
OK, OK, those too...

Adding them gives just more and more dependencies. I'll just force-remove all NFT packages.
```
root@splat:~# opkg --force-depends  remove xtables-nft nftables-json $(opkg list-installed | awk '$1 ~ /kmod-nft/ { print $1 }')
Removing package xtables-nft from root...
Removing package nftables-json from root...
Removing package kmod-nft-compat from root...
Removing package kmod-nft-core from root...
Removing package kmod-nft-fib from root...
Removing package kmod-nft-nat from root...
Removing package kmod-nft-offload from root...
```
... and reinstall ...
```
root@splat:~# opkg install xtables-nft nftables-json $(opkg list-installed | awk '$1 ~ /kmod-nft/ { print $1 }')
... mumble ...
```
Now we see if this has helped.
```
root@splat:~# iptables-nft -w -A INPUT -m conntrack --ctstate ESTABLISHED,RELATED -j ACCEPT
iptables v1.8.8 (nf_tables): Couldn't load match `conntrack':No such file or directory
```
Nope. I can only guess which architecture that VM has, so I'll assume it's x86 or x86-64. Which may explain the difference. @jow, can you strace the iptables-nft invocation and upload it, please?

jow-, 2023-11-03:
`kmod-nf-conntrack` != `kmod-ipt-conntrack`

brada4, 2023-11-03:
the brief summary - ?tables-nft relies on xt_ style kmods for proper syntax check. Seems status quo with desktop linux as of today.

LupeChristoph, 2023-11-08:
@jow- kmod-nf-conntrack != kmod-ipt-conntrack
Really?!?

LupeChristoph, 2023-11-08:
@brada4 the brief summary - ?tables-nft relies on xt_ style kmods for proper syntax check. Seems status quo with desktop linux as of today.
I'm not sure what you want to say with that.

brada4, 2023-11-08:
You need -ipt- kmod for iptables-nft to render iptables rules into nft rules usable by -nft- module.