openwrt/openwrt #13860: iptables-nft: Problem with iptables-translate

LupeChristoph, 2023-11-02:
### Describe the bug

I'm trying to upgrade my router from 21.02.2. I'm not using the OpenWRT firewall but a script generated by fwbuilder that has thrived since I converted from FreeBSD to OpenWRT Backfire in 2011.

The script uses the conntrack module. I tried several ways to add rules to NFT without writing them in NFT (fwbuilder has been more or less unmaintained for years now, so I can't expect a new version supporting NFT directly any day now :frowning: ).

The shell script generated by fwbuilder that works on OpenWRT 21.02.2 fails on 23.05.0. I'm looking for a way to make it work.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ramips/mt7621

### Device

Linksys EA7300 v2

### Image kind

Official downloaded image

### Steps to reproduce

I tried to use iptables-translate but the result does not work:
```
root@splat:~# iptables-translate -w -A INPUT   -m conntrack --ctstate ESTABLISHED,RELATED -j ACCEPT
nft add rule ip filter INPUT ct state related,established counter accept
```

### Actual behaviour

```
root@splat:~# nft add rule ip filter INPUT ct state related,established counter accept
Error: Could not process rule: No such file or directory
add rule ip filter INPUT ct state related,established counter accept
            ^^^^^^
root@splat:~# nft list ruleset
```
While iptables-nft does not care about the module name and just tried to look it up, iptables-translate *does* care about the module name and requires arguments for conntrack.

### Expected behaviour

Generate correct NFT code.

### Additional info

I'm also creating a [ticket for the iptables problem](https://github.com/openwrt/openwrt/issues/13859).

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2023-11-02:
```
nft add table ip filter
nft add chain ip filter INPUT '{ type filter hook input priority filter; policy drop; }'
```
Then add your rules. There are no predefined tables or chains in nftables.

jow-, 2023-11-03:
This is indeed expected, while iptables-translate will generate proper rules, the surrounding default chains still need to be created somehow.

closed 2023-11-03 by jow-

brada4, 2023-11-03:
@jow- you cannot have conntrack/ctstate/ct inserted by iptables-legacy or iptables-nft, even having infrastructure chains built.

jow-, 2023-11-03:
Freshly booted OpenWrt x86/64 VM:

```
BusyBox v1.36.1 (2023-10-09 21:45:35 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0, r23497-6637af95aa
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:/# op^C

root@OpenWrt:/# uci set network.lan.proto=dhcp
root@OpenWrt:/# ifup lan
[   44.872046] br-lan: port 1(eth0) entered disabled state
[   44.877917] device eth0 left promiscuous mode
[   44.880027] br-lan: port 1(eth0) entered disabled state
[   45.295718] 8021q: adding VLAN 0 to HW filter on device eth0
[   45.301708] br-lan: port 1(eth0) entered blocking state
[   45.303722] br-lan: port 1(eth0) entered disabled state
[   45.305808] device eth0 entered promiscuous mode
root@OpenWrt:/# [   45.389398] device eth0 left promiscuous mode
[   45.391893] br-lan: port 1(eth0) entered disabled state
[   45.776024] 8021q: adding VLAN 0 to HW filter on device eth0
[   45.782217] br-lan: port 1(eth0) entered blocking state
[   45.784364] br-lan: port 1(eth0) entered disabled state
[   45.786675] device eth0 entered promiscuous mode
[   47.820606] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   47.831601] br-lan: port 1(eth0) entered blocking state
[   47.835287] br-lan: port 1(eth0) entered forwarding state
[   47.845982] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

root@OpenWrt:/# 
root@OpenWrt:/# 
root@OpenWrt:/# opkg update
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/x86/64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/telephony/Packages.sig
Signature check passed.
root@OpenWrt:/# opkg install iptables
Installing iptables-nft (1.8.8-1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/x86/64/packages/iptables-nft_1.8.8-1_x86_64.ipk
[... 202 lines cut ...]
		type route hook output priority mangle; policy accept;
	}

	chain mangle_forward {
		type filter hook forward priority mangle; policy accept;
	}
}
# Warning: table ip filter is managed by iptables-nft, do not touch!
table ip filter {
	chain INPUT {
		type filter hook input priority filter; policy accept;
		counter packets 73 bytes 26334 accept
		xt match "conntrack" counter packets 0 bytes 0 accept
	}
}
# Warning: table ip nat is managed by iptables-nft, do not touch!
table ip nat {
	chain POSTROUTING {
		type nat hook postrouting priority srcnat; policy accept;
		counter packets 21 bytes 1632 xt target "MASQUERADE"
	}
}
root@OpenWrt:/# 

```

LupeChristoph, 2023-11-03:
@jow this looks like firewall4 ran. Did you indeed disable it to keep it from interfering?

jow-, 2023-11-03:
Doesn't matter, but for completeness here's a transcript after disabling fw4:

```
BusyBox v1.36.1 (2023-10-09 21:45:35 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0, r23497-6637af95aa
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:/# fw4 stop
root@OpenWrt:/# nft list ruleset
root@OpenWrt:/# uci set network.lan.proto=dhcp
root@OpenWrt:/# ifup lan
[   40.412154] br-lan: port 1(eth0) entered disabled state
[   40.451097] device eth0 left promiscuous mode
[   40.453392] br-lan: port 1(eth0) entered disabled state
[   40.862461] 8021q: adding VLAN 0 to HW filter on device eth0
[   40.869000] br-lan: port 1(eth0) entered blocking state
[   40.872574] br-lan: port 1(eth0) entered disabled state
[   40.875481] device eth0 entered promiscuous mode
[   40.884490] br-lan: port 1(eth0) entered blocking state
[   40.886674] br-lan: port 1(eth0) entered forwarding state
[   40.984816] br-lan: port 1(eth0) entered disabled state
[   40.993024] device eth0 left promiscuous mode
[   40.995157] br-lan: port 1(eth0) entered disabled state
[   41.402565] 8021q: adding VLAN 0 to HW filter on device eth0
[   41.408723] br-lan: port 1(eth0) entered blocking state
[   41.410697] br-lan: port 1(eth0) entered disabled state
[   41.412984] device eth0 entered promiscuous mode
[   41.420985] br-lan: port 1(eth0) entered blocking state
[   41.422904] br-lan: port 1(eth0) entered forwarding state
[   41.447089] br-lan: port 1(eth0) entered disabled state
[   41.491515] device eth0 left promiscuous mode
[   41.493769] br-lan: port 1(eth0) entered disabled state
[   41.882374] 8021q: adding VLAN 0 to HW filter on device eth0
[   41.888149] br-lan: port 1(eth0) entered blocking state
[   41.889988] br-lan: port 1(eth0) entered disabled state
[   41.891924] device eth0 entered promiscuous mode
[   43.927172] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   43.934614] br-lan: port 1(eth0) entered blocking state
[   43.937346] br-lan: port 1(eth0) entered forwarding state
[   43.946311] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

root@OpenWrt:/#
root@OpenWrt:/# opkg update
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/x86/64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/x86_64/packages/Packages.sig
Signature check passed.
[... 66 lines cut ...]
:PREROUTING ACCEPT [0:0]
:INPUT ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
:POSTROUTING ACCEPT [0:0]
-A POSTROUTING -j MASQUERADE
COMMIT
# Completed on Fri Nov  3 12:01:30 2023
root@OpenWrt:/# nft list ruleset
# Warning: table ip filter is managed by iptables-nft, do not touch!
table ip filter {
	chain INPUT {
		type filter hook input priority filter; policy accept;
		xt match "conntrack" counter packets 3 bytes 229 accept
	}
}
# Warning: table ip nat is managed by iptables-nft, do not touch!
table ip nat {
	chain POSTROUTING {
		type nat hook postrouting priority srcnat; policy accept;
		counter packets 1 bytes 61 xt target "MASQUERADE"
	}
}
root@OpenWrt:/# 

```

LupeChristoph, 2023-11-08:
I have a spare Netgear R6220. I'll set that up with a clean 23.05.0 and see how that looks.

LupeChristoph, 2023-11-08:
Since I only tried iptables-translate because I did not yet know how to fix the problem with IPTables modules (just install the right kmod-**ipt** packages), this ticket has a somewhat lower priority for me noe, but I *will* still check with the Netgear router when I have time.