openwrt/openwrt #13862: ssh connection problem

tuftjd, 2023-11-03:
### Describe the bug

I use openvpn in openwrt to connect to servers elsewhere. I can ping the gateway and all servers in the LAN, but I cannot connect to the server via ssh. Then I captured the packet and analyzed it. Only part of the ssh protocol packets were received in openwrt. What is the reason?

### OpenWrt version

R23.7.7 GDQ V2[2023] Compiled by eSir

### OpenWrt target/subtarget

x86/64

### Device

Intel Default string/NUC19D - Intel(R) Celeron(R) CPU J1900 @ 1.99GHz : 4C4T

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2023-11-03:
Please use the forum for general support questions. Maybe you do have an MTU issue.

closed 2023-11-03 by jow-

tuftjd, 2023-11-03:
I checked, it's not an MTU issue,and I'm not even sure what the issue is.