openwrt/openwrt #13863: odhcp6c not running after DSL disconnect

skyfmmf, 2023-11-03:
### Describe the bug

I run OpenWrt on a VDSL connection. The router I am using has a DSL modem that is (mostly) supported by OpenWrt. To get Internet access over the DSL connection, I need to run DHCPv6 inside a PPPoE session inside VLAN 7 on the DSL interface. For IPv4 connectivity, there is a DS-Lite tunnel established when the DHCPv6 is successful. This setup is working and provides stable Internet connection during the day using the following config (already including the debugging options for this issue): 

```
config dsl 'dsl'
        option annex 'b'
        option tone 'b'
        option ds_snr_offset '0'
        option xfer_mode 'ptm'
        option line_mode 'vdsl'

config device
        option name 'dsl0'
        option macaddr 'xx:xx:xx:xx:xx:xx'

config interface 'wan'
        option proto 'pppoe'
        option username 'xxx@yyy'
        option password 'xxxxxxxx'
        option ipv6 '1'
        option device 'dsl0.7'
        option peerdns '0'
        option pppd_options 'debug'

config interface 'wan6'
        option device 'pppoe-wan'
        option proto 'dhcpv6'
        option iface_dslite 'wan4'
        option reqaddress 'try'
        option reqprefix '56'
        option peerdns '0'
        option verbose '1'
```

Every night, the ISP resets the DSL connection forcing the router to reconnect. OpenWrt performs a reconnect of the DSL and reestablishes the PPPoE session. However, it does not run DHCPv6 on that fresh session leaving the system without a global IPv6 address, without the delegated prefixes to connect clients in the local networks and without an DS-Lite tunnel for IPv4 connectivity. I observe a very similar behavior when I run `ifup wan`, i.e., restart the pppoe interface. 

After the connection reset, there is no `odhcp6c` running any more. Normally, I see it running continuously on the system. However, I was not able to figure out what actually caused `odhcp6c` not running. I have log files with debug logging on the pppoe interface and from odhcp6c, see below. I'm happy to help debug this further on my system (or test patches), but I have no idea how to debug this further right now. 

I have packet captures from the dsl0 interface from a successful connection directly after boot, where DHCPv6 packets are visible directly after setting up the PPPoE session to get an address, prefix and DS-Lite AFTR endpoint; and after restarting the pppoe interface, where no DHCPv6 packets are visible.

A working state can be restored by running `/etc/init.d/network restart`. Just restarting `wan` or `wan6` does not resolve the issue. 

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ipq40xx/generic

### Device

AVM FRITZ!Box 7530 (the actual device is a AVM FRITZ!Box 7520 which has the same hardware)

### Image kind

Self-built image

### Steps to reproduce

I observe this behavior every night when my ISP resets the DSL connection. While debugging, I found that the behavior can also be triggered by just restarting the pppoe interface using `ifup wan`. 

### Actual behaviour

After the DSL connection is reset or I restart the pppoe interface, I do not have Internet connectivity anymore. The system does not get a global IPv6 address, no IPv6 prefixes, and does not set up the DS-Lite tunnel again. `odhcp6c` is not running after the connection reset. There are no DHCPv6 packets leaving the DSL interface. 

<details>
<summary>Log from a nightly disconnect</summary>
MAC addresses and IPv6 addresses (xx for local, yy for remote) and few private details anonymized.
[... 1055 lines cut ...]
# CONFIG_PACKAGE_rtl8192ce-firmware is not set
# CONFIG_PACKAGE_rtl8192cu-firmware is not set
# CONFIG_PACKAGE_rtl8192de-firmware is not set
# CONFIG_PACKAGE_rtl8192se-firmware is not set
# CONFIG_PACKAGE_rtl8821ae-firmware is not set
# CONFIG_PACKAGE_snapshot-tool is not set
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_umbim is not set
# CONFIG_PACKAGE_uqmi is not set
# CONFIG_PACKAGE_wil6210-firmware is not set
# CONFIG_PACKAGE_wl12xx-firmware is not set
# CONFIG_PACKAGE_wl18xx-firmware is not set
# CONFIG_PACKAGE_wwan is not set
# CONFIG_PACKAGE_xtables-legacy is not set
# CONFIG_PACKAGE_xtables-nft is not set
# CONFIG_WPA_MBO_SUPPORT is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Calkot, 2024-01-27:
I am running the same Fritz!Box 7520 on an line from the ISP 1&1 using netifd version 2024-01-04-c18cc79d-1 which includes commit [9cb0cb4](https://github.com/openwrt/netifd/commit/9cb0cb41830306ea9d78cc1982637d7b74b6700c), but doing a `ifdown wan && sleep 10 && ifup wan` (to simulate an ISP disconnect) still results in a killed odhcp6c without recovery. Ultimately, I need to run `/etc/init.d/network restart`.

Neustradamus, 2024-11-03:
@skyfmmf, @Calkot: Have you always the problem?

linuxtextadventurer, 2025-05-11:
Hi,

I running OpenWrt 24.10.1 with an almost identical config as @skyfmmf:
```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option packet_steering '2'

config device
        option name 'eth0'
        option eee '0'

config device
        option name 'eth1'
        option eee '0'

config device
        option type '8021q'
        option ifname 'eth1'
        option vid '7'
        option name 'eth1.7'

config device
        option name 'eth2'
        option eee '0'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth0'
        list ports 'eth2'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option device 'eth1.7'
        option proto 'pppoe'
        option username 'redacted'
        option password 'redacted'
        option ipv6 '1'

config interface 'wan6'
        option proto 'dhcpv6'
        option reqaddress 'try'
        option reqprefix 'auto'
        option device 'pppoe-wan'
        list ip6class 'wan6'

config interface 'wan4'
        option proto 'dslite'
        option peeraddr '2a00:0:cff:14a0::1'
        option encaplimit 'ignore'
        option mtu '0'

config interface 'modem'
        option proto 'static'
        option device 'eth1'
        option ipaddr '192.168.0.2'
        option netmask '255.255.255.0'
```

I'm facing the exact same issue. It's usual for a provider to disconnect its customers after 24h and assign a new IPv6 prefix upon reconnect. When this happens, the "Interfaces" says "Error: Network device is not present" next to wan4 (dslite-wan4) and wan6 (pppoe-wan). When clicking the "Restart" button next to the wan (pppoe-wan, too, but after clicking said button it then becomes eth1.7) interface, it'll say "Error: Unknown error (USER_REQUEST).". The most obvious workaround is to just schedule a reboot at night, which works, but whenever there's a disconnect in between, my network doesn't come back up without me intervening on-site. This makes having to rely on a VPN connection to my LAN at home a gamble. This is a huge issue for me.