openwrt/openwrt #13869: Fakeroot failed during compiling on macOS Sonoma

hgl, 2023-11-04:
### Describe the bug

When compiling openwrt master (75aeb7ed627ba5ea6f10f365b232bed21e40b502) on macOS Sonoma with m2 chip, fakeroot failed with:

```
/Users/hgl/openwrt/staging_dir/host/bin/fakeroot /Users/hgl/openwrt/staging_dir/host/bin/bash /Users/hgl/openwrt/scripts/ipkg-build -m "" /Users/hgl/openwrt/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc /Users/hgl/openwrt/bin/targets/x86/64/packages
dyld[26722]: tried: '/Users/hgl/openwrt/staging_dir/host/lib/libfakeroot.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need '')), '/System/Volumes/Preboot/Cryptexes/OS/Users/hgl/openwrt/staging_dir/host/lib/libfakeroot.dylib' (no such file), '/Users/hgl/openwrt/staging_dir/host/lib/libfakeroot.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need '')), '/Users/hgl/openwrt/staging_dir/host/lib/libfakeroot-0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need '')), '/System/Volumes/Preboot/Cryptexes/OS/Users/hgl/openwrt/staging_dir/host/lib/libfakeroot-0.dylib' (no such file), '/Users/hgl/openwrt/staging_dir/host/lib/libfakeroot-0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need ''))
/Users/hgl/openwrt/scripts/ipkg-build: line 182: 26721 Broken pipe: 13         $TAR -X "$tmp_dir"/tarX --format=gnu --numeric-owner --sort=name -cpf - --mtime="$TIMESTAMP" .
     26722 Abort trap: 6           | gzip -n - > "$tmp_dir"/data.tar.gz
make[4]: *** [Makefile:751: /Users/hgl/openwrt/bin/targets/x86/64/packages/libgcc1_12.3.0-4_x86_64.ipk] Error 134
make[4]: Leaving directory '/Users/hgl/openwrt/package/libs/toolchain'
time: package/libs/toolchain/compile#0.17#0.15#0.38
    ERROR: package/libs/toolchain failed to build
```

I wonder if it's because fakeroot is not of the latest version or some new patch is needed? 

cc @mpratt14 @@nbd168 @neheb @PolynomialDivision

### OpenWrt version

75aeb7ed627ba5ea6f10f365b232bed21e40b502

### OpenWrt target/subtarget

x86/64

### Device

pc

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hgl, 2023-11-04:
I tried upgrading the fakeroot package to 1.32.1, same error.

brada4, 2023-11-04:
you likely need different version of xcode that is compatible with your OSX.

hgl, 2023-11-04:
You mean I should not use Xcode 15 that corresponds to Sonoma?

brada4, 2023-11-04:
xcode with sdk 14 is compatible with Sonoma. Install brew fakeroot as in build system setup guide?

mpratt14, 2023-11-04:
this is at the point of the build where openwrt's own fakeroot is built already so it no longer matters which one the host system has

it's worth a try to use fakeroot 1.32.1

in tools/fakeroot/Makefile
adjust the version
start a build with V=sc at the end
(download check will fail)
take the correct hash from the output and adjust the hash in the Makefile
start build again 

the patches for fakeroot should apply fine...

hgl, 2023-11-05:
@mpratt14 I tried 1.32.1, same error. The patches doesn't apply cleanly, `wrapfunc.inp` contains an additional ` #include <spawn.h>`, I tried this, doesn't help:

```
--- a/wrapfunc.inp
+++ b/wrapfunc.inp
@@ -51,9 +51,11 @@ getattrlist$UNIX2003;int;(const char *pa
 #endif
 #if MAC_OS_X_VERSION_MIN_REQUIRED >= MAC_OS_X_VERSION_10_5
 #include <spawn.h>
+#if !__DARWIN_ONLY_64_BIT_INO_T
 lstat$INODE64;int;(const char *file_name, struct stat *buf);(file_name, buf)
 stat$INODE64;int;(const char *file_name, struct stat *buf);(file_name, buf)
 fstat$INODE64;int;(int fd, struct stat *buf);(fd, buf)
+#endif
 posix_spawn;int;(pid_t * __restrict pid, const char * __restrict path, const posix_spawn_file_actions_t *file_actions, const posix_spawnattr_t * __restrict attrp, char *const argv[ __restrict], char *const envp[ __restrict]);(pid, path, file_actions, attrp, argv, envp)
 posix_spawnp;int;(pid_t * __restrict pid, const char * __restrict path, const posix_spawn_file_actions_t *file_actions, const posix_spawnattr_t * __restrict attrp, char *const argv[ __restrict], char *const envp[ __restrict]);(pid, path, file_actions, attrp, argv, envp)
 #endif
```

Other chunks can apply fine.

@brada4 Sorry, what do you mean by "sdk14"? I tried installing command line tools for Xcode 14, it wouldn't let me, saying my macOS too new, and there is no option to install other macOS SDK versions in Xcode's setting's platform tab. I also can't find any mentioning of fakeroot in build system setup guide for macOS. Could you maybe elaborate?

brada4, 2023-11-05:
I mean maxOS SDK 14 as default in Xcode 15, suitable for OSX 10.14.
Please try BREW or if with plenty of time build fakeroot outside OpenWRT . The requirement to have blank architecture of dylib means something went wrong with `ld` linker down the compilation process of fakeroot.

hgl, 2023-11-05:
@brada4's suggestion helped me find out it was because I had SIP disabled which I forgot to turn on after being temporarily disabled.

With it enabled, I can compile successfully.

closed 2023-11-05 by hgl