openwrt/openwrt #13873: Logread -f -F not started at boot / Logfile not written when on mounted device

Melemir, 2023-11-04:
### Describe the bug

After upgrading from 21.02.7 via 22.03.5 to 23.05.0 logging to file on a mounted filesystem doesn't work anymore as it worked in 21.02 and as expected.
To have persistent logs, logs shall go to /opt/log/messages, where /opt is a different partition mounted via /etc/config/fstab

When /opt is not on a seperate filesystem or on a filesystem not mounted through /etc/config/fstab or log is configured to go to /var/log/messages everything works fine.
But even if the mount entry in /etc/config/fstab has option enabled '0' (and so is not mounted) and therefore /opt/log is on the  root device it will not log.

So it seems file system checking of /etc/init.d/log goes somehow wrong.

Manual testing shows that using /etc/init.d/log start everything works fine, but using /etc/init.d/log boot the logging fails.

A diff of /etc/init.d/log of 21.02 and 23.05 shows lines 55/56 having added:
       [ "$_BOOT" = "1" ] &&                                                     
                [ "$(procd_get_mountpoints "${log_file}")" ] && return 0
Removing this both lines everything works again as expected.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

x86/64

### Device

Protectli FW4B

### Image kind

Official downloaded image

### Steps to reproduce

Create device with new filesystem, e.g. sda3 in my example. Create mountpoint /opt

Add this device using /etc/config/fstab:

config mount 'sda3'
	option device 'sda3'
	option target '/opt'
	option enabled '1'

Create directory on mounted device:
mkdir /opt/log

Move Logging to this filesystem editing /etc/config/system:

config system
        option ttylogin '0'
        option urandom_seed '0'
        option timezone 'CET-1CEST,M3.5.0,M10.5.0/3'
        option zonename 'Europe/Berlin'
        option log_remote '0'
        option log_type 'file' 
        option hostname 'xxx'
        option log_file '/opt/log/messages'

Reboot router

Process logread -f -F /opt/log/messages is missing, no logfile is written in /opt/log/messages
Important: this happens only after reboot, which I assume runs /etc/init.d/log boot

Doing /etc/init.d/log stop; /etc/init.d/log start all logs are written to /opt/log/messages.

I tried to do debugging using set -x and doing diff to both logs, but sorry, thats to large to debug for someone not knowing about openwrt boot scripts.
I made sure with adding a "mount > /tmp/log" into /etc/init.d/log that /opt is already mounted.

Removing lines 55/56 in /etc/init.d/log (see above) solves the issue.

### Actual behaviour

No logs are written to /opt/log/messages after new boot

### Expected behaviour

Logs in /opt/log/messages

### Additional info

Removing lines 55 and 56  from /etc/init.d/log as it was in 21.02 it works again:

Removing:
       [ "$_BOOT" = "1" ] &&                                                     
                [ "$(procd_get_mountpoints "${log_file}")" ] && return 0

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Melemir, 2023-11-04:
Addition:
To have persistent logs, logs shall go to /opt/log/messages
I mean I want to configure it on my system this way, not as a general wish.

MrNilz, 2024-02-28:
I can confirm this behaviour. 
Calling `/etc/init.d/log boot` leads to no logs are written to a sdaX mountpoint.

Behaviour can also be checked calling `/etc/init.d/log info`
Info output from `/etc/init.d/log boot ` is:
```
{
	"log": {
		"instances": {
			"logd": {
				"running": true,
				"pid": 5466,
				"command": [
					"/sbin/logd",
					"-S",
					"64"
				],
				"term_timeout": 5,
				"respawn": {
					"threshold": 5,
					"timeout": 1,
					"retry": -1
				}
			}
		},
		"triggers": [
			[
				"config.change",
				[
					"if",
					[
						"eq",
						"package",
						"system"
					],
					[
						"run_script",
						"/etc/init.d/log",
						"reload"
					]
				],
				1000
			],
			[
				"mount.add",
				[
					"if",
					[
						"eq",
						"target",
						"/mnt/sda3"
					],
					[
						"run_script",
						"/etc/init.d/log",
						"start"
					]
				],
				1000
			]
		],
		"validate": [
			{
				"package": "system",
				"type": "system",
				"rules": {
					"log_file": "string",
					"log_hostname": "string",
					"log_ip": "host",
					"log_port": "port",
					"log_prefix": "string",
					"log_proto": "or(tcp, udp)",
					"log_remote": "bool",
					"log_size": "uinteger",
					"log_trailer_null": "bool"
				}
			}
		]
	}
}
```

Info output from `/etc/init.d/log start` is:
```
{
	"log": {
		"instances": {
			"logd": {
				"running": true,
				"pid": 5466,
				"command": [
					"/sbin/logd",
					"-S",
					"64"
				],
				"term_timeout": 5,
				"respawn": {
					"threshold": 5,
					"timeout": 1,
					"retry": -1
				}
			},
			"logfile": {
				"running": true,
				"pid": 5467,
				"command": [
					"/sbin/logread",
					"-f",
					"-F",
					"/mnt/sda3/logs/system.log",
					"-p",
					"/var/run/logread.1.pid",
					"-S",
					"64"
				],
				"term_timeout": 5
			}
		},
		"triggers": [
			[
				"config.change",
				[
					"if",
					[
						"eq",
						"package",
						"system"
					],
					[
						"run_script",
						"/etc/init.d/log",
						"reload"
					]
				],
				1000
			],
			[
				"mount.add",
				[
					"if",
					[
						"eq",
						"target",
						"/mnt/sda3"
					],
					[
						"run_script",
						"/etc/init.d/log",
						"start"
					]
				],
				1000
			]
		],
		"validate": [
			{
				"package": "system",
				"type": "system",
				"rules": {
					"log_file": "string",
					"log_hostname": "string",
					"log_ip": "host",
					"log_port": "port",
					"log_prefix": "string",
					"log_proto": "or(tcp, udp)",
					"log_remote": "bool",
					"log_size": "uinteger",
					"log_trailer_null": "bool"
				}
			}
		]
	}
}
```

Seems that the configured trigger on mount.add of /mnt/sda3 does not work as desired.

wolfgangr, 2024-11-05:
slightly different flavour of the issue:

- **syslog-ng**
- logfile is on a **btrfs**, different filesystem
- every log since boot is **written fine** to the target file, where I can read it with less
- but **logread throws "Error:** logfile not found!"
- `/cgi-bin/luci/admin/status/logs` too throws "Error: logfile not found!"

danielfdickinson, 2025-02-15:
The offender seems to be

https://github.com/openwrt/openwrt/blob/9fb44638ba9f99d295835f6bfea3b1922a012167/package/system/ubox/files/log.init#L55-L56

As /etc/init.d/log runs at 12 and /etc/init.d/fstab runs at 11 (i.e. completes before log runs) this check will cause logging to a file to an external filesystem to be skipped on boot.

This trigger:

https://github.com/openwrt/openwrt/blob/9fb44638ba9f99d295835f6bfea3b1922a012167/package/system/ubox/files/log.init#L91

does not come into play because the filesystem is already mounted, so there is no hotplug event after the trigger is active.

Simply commenting out lines 55-56, above, allows logging to file on fstab external media to work.

danielfdickinson, 2025-02-15:
Forgot to mention: in my case one device is an OpenWrt One running 24.10.0 stock OpenWrt, and the other is a Raspberry Pi Model B+, also running 24.10.0. In both cases an external f2fs filesystem is being mounted via /etc/config/fstab and logging is being sent a file on to the f2fs filesystem mounted on /data.

GraffJosh, 2025-02-17:
Is there a workaround for logging to an SSD mounted with fstab?

danielfdickinson, 2025-02-17:
@GraffJosh On the device edit /etc/init.d/log and comment out the offending lines.

E.g.
```sh
start_service_file()
{
        PIDCOUNT="$(( ${PIDCOUNT} + 1))"
        local pid_file="/var/run/logread.${PIDCOUNT}.pid"

        [ "$2" = 0 ] || {
                echo "validation failed"
                return 1
        }
        [ -z "${log_file}" ] && return

        # [ "$_BOOT" = "1" ] &&
        #       [ "$(procd_get_mountpoints "${log_file}")" ] && return 0

        mkdir -p "$(dirname "${log_file}")"

        procd_open_instance logfile
        procd_set_param command "$PROG" -f -F "$log_file" -p "$pid_file"
        [ -n "${log_size}" ] && procd_append_param command -S "$log_size"
        procd_close_instance
}
````

And add /etc/init.d/log to /etc/sysupgrade.conf so the initscript is preserved across sysupgrade.

danielfdickinson, 2025-02-17:
@GraffJosh I would also recommending setting `log_buffer_size` under `config system` in `/etc/config/system`, instead of `log_size`, otherwise your log file will be truncated to the same size as the logread buffer (which by default is 64KiB IIRC). I believe (but have not tested) that if you set log_buffer_size and log_size you can set the buffer log size and log file size independently.

GraffJosh, 2025-02-17:
Thanks. I'll do that. I also have logrotate set up.

danielfdickinson, 2025-02-17:
I'm contemplating using this as opportunity to learn ucode, by implementing the check for external mount, and is it already mounted, in a ucode script. This is only a wild idea at the moment.

danielfdickinson, 2025-02-17:
Here is a first crack at a ucode-based solution

1. Place this in `/usr/share/ucode/mountpoint.uc`
```javascript
// based on https://github.com/util-linux/sys-utils/mountpoint.c

import { access, dirname, open, realpath, stat } from 'fs';

function _is_dm_devname(canonical) {

	let dmpath = null;

	const spos = rindex(canonical, "/");
	if (spos != null) {
		dmpath = substr(canonical, spos + 1);
	} else {
		return {
			error: true,
			dmpath: null,
			error_message: "Not a device path"
		};
	}

	const can_stat = stat(canonical);

	if (can_stat == null) {
		return {
			error: true,
			dmpath: null,
			error_message: sprintf("Unable to access path %s", canonical)
		};
	}

	if (
		(length(dmpath) < 5) ||
		(substr(dmpath, 0, 4) != "/dm-") ||
		(int(substr(dmpath, 4)) == NaN) ||
		(can_stat.type != "block")
	) {
		return {
			error: false,
			dmpath: null,
			error_message: null
		};
	}

	return {
		error: false,
		dmpath: dmpath,
		error_message: null
	};
}

function _canonicalize_dm_name(path) {

	if (path == null) {
		return {
			error: true,
			dmpath: null,
			error_message: "No input path"
		};
	}

	const fpath = sprintf("/sys/block/%s/dm/name", path);
	const f = open(fpath, "r");
	if (f == null) {
		return {
			error: true,
			dmpath: null,
			error_message: sprintf("Unable to open '%s'", fpath)
[... 212 lines cut ...]
const arg1 = shift(args);

let do_message = false;
let show_errors = false;

if (arg1 == "-v") {
	do_message = true;
} else if (arg1 == "-vv") {
	do_message = true;
	show_errors = true;
} else {
	unshift(args, arg1);
}

exit(print_is_mountpoint(join(" ", args), do_message, show_errors));
```
3. Replace lines 55 and 56 in /etc/init.d/log with:
```sh
	[ "$_BOOT" = "1" ] &&
	 	[ "$(procd_get_mountpoints "${log_file}")" ] && 
		! mountpoint "$(procd_get_mountpoints "${log_file}")" &&
		return 0
```

It's a start. Comments, questions, suggestions?

danielfdickinson, 2025-02-17:
After doing all that, I remembered busybox has a `mountopoint` applet. That is probably more space efficient than (even tightened up) the code above, so the question is what will accepted be by the team.

@dangowrt Any thoughts on this?

dangowrt, 2025-02-18:
@danielfdickinson First of all thank you for taking a look at this problem and suggesting solutions. I agree that enabling the busybox applet is probably the most clean approach. On the other hand, most of the functionality of the `mountpoint` applet (or util-linux executable) isn't needed. We don't need to lookup device numbers, neither of mountpoints nor of block device nodes -- a simple `grep -q ".* $mountpoint "` would also do the trick (unless we need to validate that the filesystem mounted at a specified mountpoint is what we are expexting there). Or am I missing something?

danielfdickinson, 2025-02-18:
@dangowrt Thank you for looking at this. It is validating that the filesystem is mounted that we need. If the filesystem is not mounted, then we don't want the logging to start (and the trigger will handle it), but if it is we do want to start the logging during the boot phase because the trigger will not be triggered as the filesystem is already mounted.

danielfdickinson, 2025-02-18:
@dangowrt Sorry I didn't read that carefully enough. I guess you mean `grep -q ".* $mountpoint " /etc/mtab` or something like that?

I think for OpenWrt use that ought to be enough - following links and such to get back the actual mountpoint is probably overkill on these devices. I think saying logging to external media through a symlinked mountpoint is not supported on OpenWrt is perfectly okay.

dangowrt, 2025-02-18:
> It is validating that the filesystem is mounted that we need.

What I meant is not just that it validates that there is a filesystem mounted at a specific mountpoint, but also that what ever is mounted there is really the block device we are expecting at this place and not something else. And that should not be needed, unless people come up with very broken extroot configurations...

> I guess you mean `grep -q ".* $mountpoint " /etc/mtab` or something like that?

I'd grep over `/proc/mounts`, `/etc/mtab` feels a bit like ancient UNIX legacy to me...

GraffJosh, 2025-02-19:
I could definitely imagine a rotating log system where I ln a removable drive to a 'logging' directory, then expect logging to put its logging on whatever drive is attached to that symlink.

Perhaps overkill to enable such a functionality.

danielfdickinson, 2025-02-19:
The following snippet works for me:

```sh
start_service_file()
{
        PIDCOUNT="$(( ${PIDCOUNT} + 1))"
        local pid_file="/var/run/logread.${PIDCOUNT}.pid"

        [ "$2" = 0 ] || {
                echo "validation failed"
                return 1
        }
        [ -z "${log_file}" ] && return

        [ "$_BOOT" = "1" ] &&
                [ "$(procd_get_mountpoints "${log_file}")" ] &&
                ! grep -q ".* $(procd_get_mountpoints "${log_file}") " /proc/mounts &&
                return 0

        mkdir -p "$(dirname "${log_file}")"

        procd_open_instance logfile
        procd_set_param command "$PROG" -f -F "$log_file" -p "$pid_file"
        [ -n "${log_size}" ] && procd_append_param command -S "$log_size"
        procd_close_instance
}
```

@GraffJosh I thought about it, doing `$(readlink -f "$(procd_get_mountpoints "${log_file}")")` might be enough to handle the symlink case. Haven't tested though.

danielfdickinson, 2026-02-19:
I believe this was resolved by #18023 . At least in my case that commit resolves the issue for 25.12 and beyond.