openwrt/openwrt #13880: [23.05.0] 802.11s mesh does not use 160MHz wide channels

marcust, 2023-11-05:
### Describe the bug

From my understanding this should not be the case:
```
phy#1
	Interface phy1-mesh0
		ifindex 11
		wdev 0x100000003
		addr 02:eb:d8:4a:ce:ed
		type mesh point
		channel 48 (5240 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 23.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	733	0	0	0	4	177988		744
	Interface phy1-ap0
		ifindex 10
		wdev 0x100000002
		addr 00:eb:d8:4a:ce:ed
		type AP
		channel 48 (5240 MHz), width: 160 MHz, center1: 5250 MHz
		txpower 23.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

The mesh dev should also use 160MHz wide channels, because that is configured on the radio. 

For the given routers MR90x (MediaTek MT7986) with a WAX206 (MediaTek MT7915E) I can create a client/ap connection with 160Mhz wide channels: 
[image]

So I would expect that to also work on the Mesh? 

(This is a summary of my forum thread https://forum.openwrt.org/t/theoretical-maximum-between-wax206-and-mr90x/176057, I have seen https://github.com/openwrt/openwrt/issues/13112 and wonder if that is somewhat related)

Routers are running same version of OpenWRT. 

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

mediatek/filogic

### Device

Mercusys MR90X v1

### Image kind

Official downloaded image

### Steps to reproduce

- Configure WiFi to AX and 160MHz wide channels on both machines
- Create Mesh on the 5Ghz radio
- See Mesh connection only running on 80Mhz
[image]

### Actual behaviour

Mesh connection device is configured to use 80Mhz wide channel 

### Expected behaviour

Mesh connection device should be configured to use 160Mhz wide channel

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-05:
> option country 'DE'

Check logs to see DFS or TPC struck away 2nd half of 160MHz@ch36 `logread -e hostapd`
AP you run on one mesh point means busy channel for others' survey.

marcust, 2023-11-06:
There are no DFS related messages on `logread -f hostapd` .... actually, there are also no mesh related messages there, because that goes through wpa_supplicant if I understand correctly. 

Wouldn't anything DFS related also affect the conncection Client/AP? The output of `iw dev` is what is actually configured on the device or what is actually effective?

brada4, 2023-11-06:
Set debug level to 1 and maybe there are some hints.
https://openwrt.org/docs/guide-developer/debugging#wireless

Then restart one mesh node via web gui and check logs from all. Time sync nice to have at this moment.
Supplicant has no explicit HT settings, and seems does not inherit much from hostapd.

brada4, 2023-11-06:
? iw dev

iwinfo displays better summary, `iw dev phy1-mesh0 station dump` shows more detail on mesh communication.

gca31, 2023-11-07:
Got the same bug here with almost the same AP+Mesh config (a BPI-R3 and a WAX206).
Strangely, even if the 5GHz radio is configured with AX/HE160, iwinfo on both devices gives:
phy1-ap0  ESSID: "Ecureuil"
          Mode: Master  Channel: 100 (5.500 GHz)  HT Mode: HE160
phy1-mesh0 ESSID: "Nuts"
          Mode: Mesh Point  Channel: 100 (5.500 GHz)  HT Mode: VHT80
So the mesh network felt back to AC/VHT80
But I don't know if it is due to the BPI-R3 or to the WAX206, or both ...

brada4, 2023-11-07:
Do you get maximum speed when connecting ap to ap as normal client? Like then wds is an option.

gca31, 2023-11-19:
When connecting the WAX206 as a WDS client to the AP (BPI-R3), AX/HE160 is in use:
[image]

brada4, 2023-11-19:
Please re-test with 23.05.2

gca31, 2023-11-19:
Forgot to say I tested this with latest snapshot (r24404).

brada4, 2023-11-19:
Fix shall be in 23.05.2 if you could test ;-)

raenye, 2023-11-19:
> Fix shall be in 23.05.2

What commit fixed it? bbfb920e99672564ea24?

brada4, 2023-11-19:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=c9e8453de705818f45e306135b2ab6ce77ee8da3

gca31, 2023-11-19:
> https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=c9e8453de705818f45e306135b2ab6ce77ee8da3

This patch is already applied in latest snapshot: Version of "hostapd-common" for my APs is "2023-09-08-e5ccbfc6-6", which seems to match this patch

brada4, 2023-11-19:
Question is if release is patched up or still behind.

gca31, 2023-11-19:
Just tested both APs with 23.05.2: result is the same:
mesh not using HE160
hostapd-common version is 2023-09-08-e5ccbfc6-6

csharper2005, 2023-11-19:
Furthermore, it works in AX (not AX mode).
```
phy1-mesh0 ESSID: "5000"
          Access Point: 
          Mode: Mesh Point  Channel: 48 (5.240 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 28 dBm  Link Quality: 51/70
          Signal: -59 dBm  Noise: -92 dBm
          Bit Rate: 720.6 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: embedded [MediaTek MT7986]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

raenye, 2023-11-19:
Not sure what you mean by "ax but not ax mode".

csharper2005, 2023-11-19:
> Not sure what you mean by "ax but not ax mode".

typo

marcust, 2023-11-23:
I tested with fully package updated 23.05.2 (`hostapd-common - 2023-09-08-e5ccbfc6-6`) just now and it looks no different than before, no 160Mhz channel on the mesh, same output of `iw dev`:
[image]

As WDS client it works with 160Mhz:
[image]

brada4, 2023-11-23:
Yes, mesh is 80MHz while WDS  proves 160MHz is OK.

bluewavenet, 2024-02-19:
@ynezz @marcust @brada4 @raenye @gca31 
This seems to be caused by the `iw` utility not supporting HS modes, The mac80211.sh script uses `iw` to set up the mesh, so a mesh interface always falls back to an HT mode.....

raenye, 2024-02-19:
@bluewavenet 
Currently my mesh shows VHT80 speeds due to an 802.11ac-only node, so I cannot test, but the problem shown [here](https://github.com/openwrt/openwrt/issues/13880#issuecomment-1824072853) was that the mesh is using HE but with 80 MHz rather that 160 MHz bandwidth.

bluewavenet, 2024-02-19:
@raenye 
An interesting thread on the forum:
https://forum.openwrt.org/t/160mhz-not-working-on-802-11s-mesh-with-filogic-ax6000-ax4200/188401

filippz, 2024-03-31:
**TL;DR**
Would any of the _fellow meshers_ be so kind to try if doing `iwinfo phy1-mesh0 scan` breaks mesh for them?

I have the same issue with 2 x Asus TUF-AX4200 doing only 80MHz, but I also have a strange issue in which mesh simply stops working after a while or when either side initiates WiFi scan. After the issue appears the link is up but I'm unable to get ping to other side to work and when I restart the radio on either side the traffic restarts and continues to work for a while (couple of hours up to some days/weeks - seemingly depending on the volume of traffic) or until next WiFi scan. I have restored the exact same config to one TP-Link AX23 (I have only changed path on `wifi-device` entries in `/etc/config/wireless`) and it works fine after the AX23 initiates WiFi scan which suggests that the issue would be either chipset or device specific.

The underlying issues if rather strange to me because because `ping` fails but `arp-scan` works fine and `arping` is able to receive up to a few of replies to broadcast requests only and no reply to unicast pings:
### arp-scan
_working fine_
```
root@OpenWrt:~# arp-scan -I phy1-mesh0 --localnet
Interface: phy1-mesh0, type: EN10MB, MAC: xx:xx:xx:xx:xx:xx, IPv4: 10.20.31.1
WARNING: Cannot open MAC/Vendor file /usr/share/arp-scan/ieee-oui.txt: No such file or directory
WARNING: Cannot open MAC/Vendor file /etc/arp-scan/mac-vendor.txt: No such file or directory
Starting arp-scan 1.10.0 with 8 hosts (https://github.com/royhills/arp-scan)
10.20.31.2      yy:yy:yy:yy:yy:yy       (Unknown: locally administered)

1 packets received by filter, 0 packets dropped by kernel
Ending arp-scan 1.10.0: 8 hosts scanned in 1.476 seconds (5.42 hosts/sec). 1 responded
```
### arping
_gets only 1 reply_
```
root@OpenWrt:~# arping -I phy1-mesh0 -c 10 10.20.31.2
ARPING 10.20.31.2 from 10.20.31.1 phy1-mesh0
Unicast reply from 10.20.31.2 [XX:XX:XX:XX:XX:XX]  2.035ms
Sent 10 probes (1 broadcast(s))
Received 1 response(s)
```
### arping -b
_gets 6 replies (varies)_
```
root@OpenWrt:~# arping -I phy1-mesh0 -c 10 -b 10.20.31.2
ARPING 10.20.31.2 from 10.20.31.1 phy1-mesh0
Unicast reply from 10.20.31.2 [XX:XX:XX:XX:XX:XX]  1.636ms
Unicast reply from 10.20.31.2 [XX:XX:XX:XX:XX:XX]  2.135ms
Unicast reply from 10.20.31.2 [XX:XX:XX:XX:XX:XX]  1.534ms
Unicast reply from 10.20.31.2 [XX:XX:XX:XX:XX:XX]  1.449ms
Unicast reply from 10.20.31.2 [XX:XX:XX:XX:XX:XX]  1.893ms
Unicast reply from 10.20.31.2 [v]  2.322ms
Sent 10 probes (10 broadcast(s))
Received 6 response(s)
```

chuloplaya, 2024-04-05:
Any news about this issue?

oxavelar, 2024-10-21:
I also observe the same doing a 802.11s mesh between MT6000 and a Dynalink WRX36.

Clients in the 5 GHz AX radio do show 160 MHz.

```
root@Saturn:~# iw phy0-ap0 info
Interface phy0-ap0
        ifindex 12
        wdev 0x1
        addr xx:xx:xx:xx:xx:xx
        ssid Mars
        type AP
        wiphy 0
        channel 100 (5500 MHz), width: 160 MHz, center1:
 5570 MHz
        txpower 23.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks
overlmt hashcol tx-bytes        tx-packets
                0       0       13      0       0
0       0       580   
```

But the mesh peer link is capped @ 80 MHz.

```
root@Saturn:~# iw phy0-mesh0 info
Interface phy0-mesh0
        ifindex 14
        wdev 0x3
        addr xx:xx:xx:xx:xx:xx
        type mesh point
        wiphy 0
        channel 100 (5500 MHz), width: 80 MHz, center1:
5530 MHz
        txpower 23.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks
overlmt hashcol tx-bytes        tx-packets
                0       0       43903   0       0
0       8       5332183         43920
```

AT-StephenDetomasi, 2024-10-27:
Same issue here on my Redmi AX3000T :( if I manually set a channel and 160mhz width, the mesh simply does not connect despite the channels being allowed by my selected region

marcust, 2024-12-31:
So I played with `24.10-rc4` and understanding that the rational is that `iw` does not support HS modes the issue still seems to persist, even wit `iw` in version 6.9 which is in 24.10. Is that expected? 

(I only updated one of three in my mesh, but given the fact that it can connect as client using the HS mode I would expect one mesh node to also be able to go to HS, or is that wrong?)

marcust, 2024-12-31:
So it works when I configure `104 (5520 Mhz)` with 160 Mhz wide channels. (Why is that referred to as 114 though?) 

Why hoes wpa_supplicant have a different idea of what is allowed on a mesh than what is allowed with "normal" clients?

Neustradamus, 2025-01-01:
The problem is in main branch too?

marcust, 2025-01-06:
I fear that I do not fully understand the issue and the solution, or at least how allowing higher frequencies will make my original problem go away. 

I was running on `48 (5240 MHz)`, which as I just learened would be an f0 index of 50 as well, but this was only meshing at 80 Mhz width, I changed it to Channel 104 now, because that has the frequency that the `freq` check looks at, and it works with 160 Mhz witdh. 

My question is, should any channel configuration between 36 and 64, 100 and 128 and 147 to 177 theoretically work with 160 Mhz wide Mesh channels? Because the check in the code looks at a concrete frequency configuration, which for 50 is only 5180 Mhz and then it does not work with the 160 Mhz channel when I am actually on channel 48. Shouldn't that check the frequency range then? 

I have to check again if a concrete channel 36 config would give me also a good enough result, I was under the impression it did not work somehow and I think I have DFS issues now with the higher channel (not at home right now).

bluewavenet, 2025-01-06:
@marcust 

> My question is, should any channel configuration between 36 and 64, 100 and 128 and 147 to 177 theoretically work with 160 Mhz wide Mesh channels?

Check in the wikipedia link posted above by @romanovj:
https://github.com/openwrt/openwrt/issues/13880#issuecomment-2566709514

You will see that the only valid channels available for 160MHz are 50, 114 and 163.
You will also see that all three channels are subject to some degree to DFS, TPC or Indoor-low-power operation.

A mesh link will not work correctly on DFS or TPC  restrictions as the nodes will block each other and a partial low power band restriction will also cause similar issues.

So it is "correct" that 160MHz is blocked for mesh use on unlicensed spectrum for all regions as far as I can see.

You will have to wait for 6GHz hardware before you can go to 160MHz bandwidth, but then 320MHz will be an option (or another can of worms)....

marcust, 2025-01-06:
Yea, channel 36 does not work for me because Wifi on 5Ghz does not come up if I configure it to channel 36 on any of my Mediatek devices (MERCUSYS MR90X v1, D-Link AQUILA PRO AI M30 A1, Netgear WAX206) in HE160, I can start at channel 40 only. Is this actually how it is supposed to be?

marcust, 2025-01-06:
> You will see that the only valid channels available for 160MHz are 50, 114 and 163.

I understand that, but then OpenWrt lets me configure 
[image]

So that is at least confusing. 

> So it is "correct" that 160MHz is blocked for mesh use on unlicensed spectrum for all regions as far as I can see.

But I can run a mesh on 104 and that "works" with 160 Mhz wide channels, as I said I think have DFS issues there. I can not configure a mesh on 50, whatever I try (well, at least a selection of 40 or 48 does not seem to trigger the above code that allows 160 Mhz wide channels). OpenWrt doesn't even allow me to select the "correct" channels depending on width. 

So there is still an issue here that I can not even try that on channel 50 with a 160 Mhz wide channel.

bluewavenet, 2025-01-06:
@marcust 

> OpenWrt lets me configure

Actually, luci lets you configure the wireless hardware in a mode that can in some circumstances be used for an AP or Sta, but it is actually broken when it comes to mesh. 

> So there is still an issue here that I can not even try that on channel 50 with a 160 Mhz wide channel.

For a mesh interface, channel 50/160MHz is not useable, so it is not an issue, it is a misconception originating from a problem with luci not checking properly.

marcust, 2025-01-06:
>For a mesh interface, channel 50/160MHz is not useable, so it is not an issue, it is a misconception originating from a problem with luci not checking properly.

My misconception is coming from this code, that configures 160 MHZ on channel 50 in `wpa_supplicant`, but there seems no way for me to hit this with any configuration I do in LuCI (or `/etc/config/wireless` directly)
```
 else if (ssid->max_oper_chwidth == CONF_OPER_CHWIDTH_160MHZ) {
		if (freq->freq == 5180) {
			chwidth = CONF_OPER_CHWIDTH_160MHZ;
			vht_caps |= VHT_CAP_SUPP_CHAN_WIDTH_160MHZ;
			seg0 = 50;
		} else if (freq->freq == 5520) {
			chwidth = CONF_OPER_CHWIDTH_160MHZ;
			vht_caps |= VHT_CAP_SUPP_CHAN_WIDTH_160MHZ;
			seg0 = 114;
		}
```

marcust, 2025-01-06:
Ok, central frequency for segment it is.

But how do I satisfy the actual `freq->freq` condition in the above code with my radio config?

Apparently I can do it by setting my Channel to 104 for the lower one, because of the same frequency, but I can't set the Channel to 36 to satisfy the upper one, because then the radio won't come up.

marcust, 2025-01-07:
This is all Germany, 'DE'

gongtao0607, 2025-07-05:
I can reproduce the same issue on 24.10.2 + Qualcomm QCN5054 (5GHz) and QCN9094 (6GHz) (Device is Linksys MX8500)

5GHz: 
```
root@MX8503:~# iw dev phy0-mesh0 info
Interface phy0-mesh0
	ifindex 11
	wdev 0x2
	addr xx:xx:xx:xx:xx:xx
	type mesh point
	wiphy 0
	channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
	txpower 28.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	7	0	0	0	0	928		7
root@MX8503:~# iw dev phy0-ap0 info
Interface phy0-ap0
	ifindex 10
	wdev 0x1
	addr xx:xx:xx:xx:xx:xx
	ssid OpenWrt
	type AP
	wiphy 0
	channel 149 (5745 MHz), width: 160 MHz, center1: 5815 MHz
	txpower 28.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0
```
6GHz:
```
root@MX8503:~# iw dev phy2-mesh0 info
Interface phy2-mesh0
	ifindex 15
	wdev 0x200000004
	addr yy:yy:yy:yy:yy:yy
	type mesh point
	wiphy 2
	channel 37 (6135 MHz), width: 80 MHz, center1: 6145 MHz
	txpower 18.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	7	0	0	0	0	928		7
root@MX8503:~# iw dev phy2-ap0 info
Interface phy2-ap0
	ifindex 14
	wdev 0x200000003
	addr yy:yy:yy:yy:yy:yy
	ssid OpenWrt
	type AP
	wiphy 2
	channel 37 (6135 MHz), width: 160 MHz, center1: 6185 MHz
	txpower 18.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0
root@MX8503:~# 
```

I can verify AP+Sta 160MHz is working.