openwrt/openwrt #13881: image builder missing ./tmp/.packageinfo

ascendbeing, 2023-11-05:
### Describe the bug

feig@host ~/openwrt-imagebuilder-ath79-generic.Linux-x86_64 $ make clean; make image PROFILE="tplink_archer-c7-v2" PACKAGES="ath10k-board-qca988x attendedsysupgrade-common base-files block-mount busybox ca-bundle cgi-io dmesg dnsmasq drill dropbear ethtool-full firewall4 fstools fwtool getrandom grep hostapd-mbedtls hostapd-utils htop ip-tiny -iw iw-full iwinfo kernel kmod kmod-ath10k-ct kmod-ath9k kmod-cfg80211 kmod-gpio-button-hotplug kmod-hwmon-core kmod-mac80211 kmod-nf-conntrack kmod-nf-conntrack6 kmod-nf-flow kmod-nf-log kmod-nf-log6 kmod-nf-nat kmod-nf-reject kmod-nf-reject6 kmod-nfnetlink kmod-nft-core kmod-nft-fib kmod-nft-nat kmod-nft-offload kmod-nls-base kmod-phy-ath79-usb kmod-random-core kmod-scsi-core kmod-slhc kmod-usb-core kmod-usb-ehci kmod-usb-ledtrig-usbport kmod-usb-storage kmod-usb2 libiwinfo-data libiwinfo20230701 libkmod libustream-mbedtls20201210 logd lsof luci-app-attendedsysupgrade luci-app-commands luci-app-firewall luci-app-opkg luci-app-uhttpd luci-app-upnp luci-base luci-mod-admin-full luci-mod-network luci-mod-status luci-mod-system luci-proto-ipv6 luci-theme-bootstrap luci-theme-openwrt-2020 miniupnpd-nftables mtd nano-full netifd nftables-json odhcp6c odhcpd-ipv6only openssh-sftp-server openwrt-keyring opkg procd procd-seccomp procd-ujail procps-ng procps-ng-free psmisc px5g-mbedtls rpcd rpcd-mod-file rpcd-mod-iwinfo rpcd-mod-luci rpcd-mod-rpcsys rpcd-mod-rrdns rpcd-mod-ucode sed strace stubby swconfig terminfo uboot-envtools ubox ubus ubusd uci uclient-fetch ucode ucode-mod-fs ucode-mod-html ucode-mod-math ucode-mod-nl80211 ucode-mod-rtnl ucode-mod-ubus ucode-mod-uci ucode-mod-uloop uhttpd uhttpd-mod-ucode urandom-seed urngd usign wget-ssl wireless-regdb zlib -iw -ppp -kmod-ppp -luci-mod-ppp -kmod-lib-zstd -kmod-fs-btrfs -wpad-basic-mbedtls -ppp-mod-pppoe -ppp-mod-pppox -ppp-mod-pppoa -ath10k-firmware-qca988x-ct kmod-random-core"
rm -rf /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/tmp /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/dl /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/root-ath79 /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/bin/targets/ath79/generic
make: TMPDIR value /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/tmp: No such file or directory
make: using default temporary directory '/tmp'
make: TMPDIR value /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/tmp: No such file or directory
make: using default temporary directory '/tmp'
make: TMPDIR value /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/tmp: No such file or directory
make: using default temporary directory '/tmp'
Building images for ath79 - TP-Link Archer C7 v2
Packages: ath10k-board-qca988x attendedsysupgrade-common base-files block-mount busybox ca-bundle cgi-io dmesg dnsmasq drill dropbear ethtool-full firewall4 fstools fwtool getrandom grep hostapd-mbedtls hostapd-utils htop ip-tiny iw-full iwinfo kernel kmod kmod-ath10k-ct kmod-ath9k kmod-cfg80211 kmod-gpio-button-hotplug kmod-hwmon-core kmod-mac80211 kmod-nf-conntrack kmod-nf-conntrack6 kmod-nf-flow kmod-nf-log kmod-nf-log6 kmod-nf-nat kmod-nf-reject kmod-nf-reject6 kmod-nfnetlink kmod-nft-core kmod-nft-fib kmod-nft-nat kmod-nft-offload kmod-nls-base kmod-phy-ath79-usb kmod-random-core kmod-scsi-core kmod-slhc kmod-usb-core kmod-usb-ehci kmod-usb-ledtrig-usbport kmod-usb-storage kmod-usb2 libiwinfo-data libiwinfo20230701 libkmod libustream-mbedtls20201210 logd lsof luci-app-attendedsysupgrade luci-app-commands luci-app-firewall luci-app-opkg luci-app-uhttpd luci-app-upnp luci-base luci-mod-admin-full luci-mod-network luci-mod-status luci-mod-system luci-proto-ipv6 luci-theme-bootstrap luci-theme-openwrt-2020 miniupnpd-nftables mtd nano-full netifd nftables-json odhcp6c odhcpd-ipv6only openssh-sftp-server openwrt-keyring opkg procd procd-seccomp procd-ujail procps-ng procps-ng-free psmisc px5g-mbedtls rpcd rpcd-mod-file rpcd-mod-iwinfo rpcd-mod-luci rpcd-mod-rpcsys rpcd-mod-rrdns rpcd-mod-ucode sed strace stubby swconfig terminfo uboot-envtools ubox ubus ubusd uci uclient-fetch ucode ucode-mod-fs ucode-mod-html ucode-mod-math ucode-mod-nl80211 ucode-mod-rtnl ucode-mod-ubus ucode-mod-uci ucode-mod-uloop uhttpd uhttpd-mod-ucode urandom-seed urngd usign wget-ssl wireless-regdb zlib kmod-random-core base-files busybox ca-bundle dnsmasq dropbear firewall4 fstools kernel kmod-ath10k-ct kmod-ath9k kmod-gpio-button-hotplug kmod-nft-offload kmod-usb-ledtrig-usbport kmod-usb2 libc libgcc libustream-mbedtls logd mtd netifd nftables odhcp6c odhcpd-ipv6only opkg procd procd-seccomp procd-ujail swconfig uboot-envtools uci uclient-fetch urandom-seed urngd

Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/snapshots/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/dl/openwrt_core
Downloading https://downloads.openwrt.org/snapshots/targets/ath79/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/dl/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/targets/ath79/generic/kmods/5.15.137-1-75eb6ce3d7c44e9753368a3edec9acfd/Packages.gz
Updated list of available packages in /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/dl/openwrt_kmods
Downloading https://downloads.openwrt.org/snapshots/targets/ath79/generic/kmods/5.15.137-1-75eb6ce3d7c44e9753368a3edec9acfd/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/dl/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/dl/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/dl/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../home/feig/openwrt-imagebuilder-ath79-generic.Linux-x86_64/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.

Installing packages...
Installing libc (1.2.4-4) to root...
Installing libgcc1 (12.3.0-4) to root...
[... cut ...]
### Expected behaviour

.packageinfo error not existing

### Additional info

I used ath79 generic image builder download from latest snapshot. I extracted with tar xvJf then ran command in logs.
btw attended sysupgrade doesn't work either. so that's 2/2 of usual avenues to make this image not working and I'll try the 3rd after.
for asu, here's the log:
(I'll edit it in here but basically it flails bc no /lib/functions.sh)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ascendbeing, 2023-11-05:
ok for 3/3 firmware selector also fales:
ownloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
/builder/build_dir/target-mips_24kc_musl/root-ath79/etc/init.d/dnsmasq: line 4: /lib/functions.sh: No such file or directory
./etc/init.d/dnsmasq: line 4: /lib/functions.sh: No such file or directory
Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-mips_24kc_musl/root-ath79/dev".
Ignoring, exclude it (-e/-ef) to override.
[mktplinkfw] rootfs offset aligned to 0x2484536
[mktplinkfw] firmware file "/builder/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-735586dcbae6-ath79-generic-tplink_archer-c7-v2-squashfs-sysupgrade.bin.new" completed
[mktplinkfw] rootfs offset aligned to 0x2484536
[mktplinkfw] firmware file "/builder/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-735586dcbae6-ath79-generic-tplink_archer-c7-v2-squashfs-factory.bin.new" completed
[mktplinkfw] rootfs offset aligned to 0x2484536
[mktplinkfw] firmware file "/builder/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-735586dcbae6-ath79-generic-tplink_archer-c7-v2-squashfs-factory-us.bin.new" completed
[mktplinkfw] rootfs offset aligned to 0x2484536
[mktplinkfw] firmware file "/builder/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-735586dcbae6-ath79-generic-tplink_archer-c7-v2-squashfs-factory-eu.bin.new" completed
Cannot open '/builder/tmp/.packageinfo': No such file or directory
make[3]: *** [Makefile:104: install] Error 1
make[2]: *** [Makefile:208: build_image] Error 2
make[1]: *** [Makefile:146: _call_image] Error 2
make: *** [Makefile:262: image] Error 2

brada4, 2023-11-05:
Get it done with firmware-selector? (You can close the issue since i lost my scroll wheel half way and cannot return here)

fantom-x, 2023-11-05:
The same happens for x86-64 target: I can spin up a 23.05.0 image, but not 23.05-SNAPSHOT or SNAPSHOT.

ascendbeing, 2023-11-05:
> Get it done with firmware-selector? (You can close the issue since i lost my scroll wheel half way and cannot return here)

nope. see my follow-up reply. firmware selector is broken for 3/3 methods all broken

brada4, 2023-11-05:
I pasted your package set to image builder and it emitted images sized approx 9MB.
Please make a new polite issue attaching parseable logs. I cannot find in 150 pages which of four possible setups you used under imagebuilder or even its version.

ascendbeing, 2023-11-06:
> I pasted your package set to image builder and it emitted images sized approx 9MB. Please make a new polite issue attaching parseable logs. I cannot find in 150 pages which of four possible setups you used under imagebuilder or even its version.

snapshot for ath79/generic

brada4, 2023-11-06:
https://firmware-selector.openwrt.org/
Expand |> arrow and paste package list in first box.

[image]

ascendbeing, 2023-11-06:
I just tried ath79/generic C7 v2 archer and it didn't build

ascendbeing, 2023-11-06:
[image]

ascendbeing, 2023-11-06:
[image]

ascendbeing, 2023-11-06:
[image]

ascendbeing, 2023-11-06:
@brada4 I suspect you're a twull but in the infinitesimally microscopic chance that's not the case highlighting you to see ^ screenshot

brada4, 2023-11-06:
Copy both red sections into txt files and make a new issue.

pktpls, 2023-11-06:
Have the same issue with 23.05 x86/64 imagebuilder as of right now. First thing that's missing in output is staging_dir/target-x86_64_musl/root-x86/boot/, build fails only later.

<details>
    <summary>output of make image</summary>

```
Checking 'true'... ok.
Checking 'false'... ok.
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'perl-data-dumper'... ok.
Checking 'perl-findbin'... ok.
Checking 'perl-file-copy'... ok.
Checking 'perl-file-compare'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'xargs'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'egrep'... ok.
Checking 'getopt'... ok.
Checking 'realpath'... ok.
Checking 'stat'... ok.
Checking 'gzip'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'install'... ok.
Checking 'perl'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'python3-distutils'... ok.
Checking 'python3-stdlib'... ok.
Checking 'file'... ok.
Checking 'which'... ok.
Checking 'ldconfig-stub'... ok.
Building images for x86 - Generic x86/64
Packages: collectd-mod-interface collectd-mod-iwinfo collectd-mod-load collectd-mod-memory collectd-mod-network collectd-mod-ping collectd-mod-uptime ethtool ip6tables iperf3 iwinfo kmod-nft-bridge mtr nftables tcpdump vnstat libiwinfo-lua luci-mod-admin-full luci-proto-ipv6 luci-theme-bootstrap rpcd-mod-rrdns uhttpd uhttpd-mod-ubus collectd-mod-conntrack collectd-mod-olsrd collectd-mod-snmp6 olsrd olsrd-mod-arprefresh olsrd-mod-dyn-gw olsrd-mod-jsoninfo olsrd-mod-nameservice olsrd-mod-txtinfo olsrd-utils kmod-ipip luci-app-olsr luci-app-olsr-services luci-mod-falter falter-common luci-app-falter-owm luci-app-falter-owm-ant luci-app-falter-owm-cmd luci-app-falter-owm-gui iptables-mod-ipopt kmod-ipt-ipopt bird2 bird2c babeld luci-app-babeld ip-full gre wireguard-tools kmod-wireguard wg-installer-server wg-installer-server-hotplug-babeld wg-installer-server-hotplug-olsrd conntrackd samplicator libustream-wolfssl hostapd-wolfssl px5g-wolfssl base-files busybox ca-bundle dnsmasq dropbear e2fsprogs firewall4 fstools grub2-bios-setup kernel kmod-amazon-ena kmod-amd-xgbe kmod-bnx2 kmod-button-hotplug kmod-e1000 kmod-e1000e kmod-forcedeth kmod-fs-vfat kmod-igb kmod-igc kmod-ixgbe kmod-nft-offload kmod-r8169 kmod-tg3 libc libgcc logd mkf2fs mtd netifd nftables odhcp6c odhcpd-ipv6only opkg partx-utils procd procd-seccomp procd-ujail uci uclient-fetch urandom-seed urngd

Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/targets/x86/64/packages/Packages.gz
Updated list of available packages in /tmp/ansible-openwrt/tmp/imagebuild/l105-gw/build_dir/target-x86_64_musl/root-x86/../../../../../../../../tmp/ansible-openwrt/tmp/imagebuild/l105-gw/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/x86_64/base/Packages.gz
Updated list of available packages in /tmp/ansible-openwrt/tmp/imagebuild/l105-gw/build_dir/target-x86_64_musl/root-x86/../../../../../../../../tmp/ansible-openwrt/tmp/imagebuild/l105-gw/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/targets/x86/64/kmods/5.15.137-1-47964456485559d992fe6f536131fc64/Packages.gz
Updated list of available packages in /tmp/ansible-openwrt/tmp/imagebuild/l105-gw/build_dir/target-x86_64_musl/root-x86/../../../../../../../../tmp/ansible-openwrt/tmp/imagebuild/l105-gw/dl/openwrt_kmods
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/targets/x86/64/kmods/5.15.137-1-47964456485559d992fe6f536131fc64/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/x86_64/luci/Packages.gz
Updated list of available packages in /tmp/ansible-openwrt/tmp/imagebuild/l105-gw/build_dir/target-x86_64_musl/root-x86/../../../../../../../../tmp/ansible-openwrt/tmp/imagebuild/l105-gw/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/x86_64/packages/Packages.gz
Updated list of available packages in /tmp/ansible-openwrt/tmp/imagebuild/l105-gw/build_dir/target-x86_64_musl/root-x86/../../../../../../../../tmp/ansible-openwrt/tmp/imagebuild/l105-gw/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/x86_64/routing/Packages.gz
[... 1166 lines cut ...]
+ for entry in "$1"/*
+ '[' -f /tmp/ansible-openwrt/tmp/imagebuild/l105-gw/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-23.05-snapshot-r23595-ec54022549-x86-64-generic-ext4-combined-efi.img.gz.boot/efi/boot ']'
[... cut ...]
+ rm -f /tmp/ansible-openwrt/tmp/imagebuild/l105-gw/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-23.05-snapshot-r23595-ec54022549-x86-64-generic-ext4-combined-efi.img.gz.kernel
/tmp/ansible-openwrt/tmp/imagebuild/l105-gw/staging_dir/host/bin/grub-bios-setup: warning: Your BIOS Boot Partition is under 1 MiB, please increase its size..
Cannot open '/tmp/ansible-openwrt/tmp/imagebuild/l105-gw/tmp/.packageinfo': No such file or directory
make[3]: *** [Makefile:157: install] Error 1
make[2]: *** [Makefile:208: build_image] Error 2
make[1]: *** [Makefile:146: _call_image] Error 2
make: *** [Makefile:262: image] Error 2
```
</details>

DazzyWalkman, 2023-11-06:
Same with Image Builder r24315-ae500e62e2-x86-64
My workaround for this:
Copy .packageinfo located in the builder root dir to the builder tmp dir.

ascendbeing, 2023-11-06:
> Same with Image Builder r24315-ae500e62e2-x86-64 My workaround for this: Copy .packageinfo located in the builder root dir to the builder tmp dir.

that worked for me. thank you.

edrikk, 2023-11-06:
The same error seen when using auc and asu for a couple of devices I tried.

```
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r24238-b7c12a9bfc on qualcommax/ipq807x (xiaomi,ax3600)
Available: SNAPSHOT r24315-ae500e62e2
.....
12661660 bytes (13 MB, 12 MiB) copied, 0.100555 s, 126 MB/s
Cannot open '/builder/tmp/.packageinfo': No such file or directory
```

and also a raspberry pi ( cm27xx/bcm2708 (raspberrypi,model-b-plus ), with ASU giving error:

```
109051904 bytes (109 MB, 104 MiB) copied, 0.839944 s, 130 MB/s
Cannot open '/builder/tmp/.packageinfo': No such file or directory
```

jow-, 2023-11-06:
I guess it is caused by the CycloneDX metadata support, please check if the issue still persists after unsetting `CONFIG_JSON_CYCLONEDX_SBOM` in `.config`

DazzyWalkman, 2023-11-06:
> I guess it is caused by the CycloneDX metadata support, please check if the issue still persists after unsetting `CONFIG_JSON_CYCLONEDX_SBOM` in `.config`
Yes, it went smoothly when this line had been commented out.

jow-, 2023-11-06:
Thanks for confirming. @ynezz - I think the ConcleDX JSON handling needs to wrapped in a condition which checks that `$(SDK)` and `$(IB)` is empty.

commit c4259a65 (2023-11-06): image: fix image generation within ImageBuilder

Changes introduced in commit d604a07225c5 ("build: add CycloneDX SBOM
JSON support") broke ImageBuilder:

  Cannot open '/openwrt-imagebuilder-ath79-generic.Linux-x86_64/tmp/.packageinfo': No such file or directory

So lets fix it by wrapping the BOM generation behind condition of IB
feature check.

Fixes: #13881
Fixes: d604a07225c5 ("build: add CycloneDX SBOM JSON support")
PR #13889 (2023-11-06): image: fix image generation within ImageBuilder
Changes introduced in commit d604a07225c5 ("build: add CycloneDX SBOM JSON support") broke ImageBuilder:

```
  Cannot open '/openwrt-imagebuilder-ath79-generic.Linux-x86_64/tmp/.packageinfo': No such file or directory
```

So lets fix it by wrapping the BOM generation behind condition of IB feature check.

Fixes: #13881
Fixes: d604a07225c5 ("build: add CycloneDX SBOM JSON support")
commit cfadbc09 (2023-11-06): image: fix image generation within ImageBuilder

Changes introduced in commit d604a07225c5 ("build: add CycloneDX SBOM
JSON support") broke ImageBuilder:

  Cannot open '/openwrt-imagebuilder-ath79-generic.Linux-x86_64/tmp/.packageinfo': No such file or directory

So lets fix it by wrapping the BOM generation behind condition of IB
feature check.

Fixes: #13881
Fixes: d604a07225c5 ("build: add CycloneDX SBOM JSON support")

(cherry picked from commit c4259a658673cc1a02ed17bfa8e94de17de00ad2)
closed 2023-11-06 by openwrt-bot