openwrt/openwrt #13891: 25.02.0 image lantic-xway-arcadian_arv8539pw22 not working

h1618, 2023-11-06:
### Describe the bug

tried to install openwrt-23.05.0-lantiq-xway-arcadyan_arv8539pw22-squashfs-sysupgrade.bin via webinterface (21.02.0 openwrt running) but the device didn't come up. Then connected the serial, and flashed it via loadx / cp.b, but after a reset I saw:
`ARV8539PW22 # reset�
ROM VER: 1.0.3
CFG 01
Read EEPROM.

U-Boot 2013.10-openwrt4 (Jun 28 2015 - 10:26:00) ARV8539PW22

Board: Speedport W 504V Typ A
SoC:   Lantiq Danube-S v1.5
CPU:   333.333 MHz
IO:    166.667 MHz
BUS:   83.333 MHz
BOOT:  NOR
DRAM:  64 MiB
Flash: 8 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   ltq-eth
Hit any key to stop autoboot:  0 
## Booting kernel from Legacy Image at b0040000 ...
   Image Name:   MIPS OpenWrt Linux-5.15.134
   Created:      2023-10-09  21:45:35 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2725739 Bytes = 2.6 MiB
   Load Address: 80002000
   Entry Point:  80002000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... LZMA: uncompress or overwrite error 7 - must RESET b
ROM VER: 1.0.3
CFG 01
Read EEPROM.
`
repeating over and over. Then did a second download and compared the hashes, SHA256 being fd9fc81e0c9d2d499fe705356e706602836f4963a0cd7086cf23b6456ade1dfd identical, and then repeated the serial flash, same result. Seems the image is broken ...

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

lantiq-xway-arcadyan_arv8539pw22

### Device

T-COM Speedport W504V

### Image kind

Official downloaded image

### Steps to reproduce

see above - any attempt to flash this image produces a broken system

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-06:
rel notes> Sysupgrade from 21.02 to 23.05 is not officially supported.

Also you need to reset configuration going from 21 to 22 because of the new switch driver (hou can recover all other parts from 'config backup' tarball)
https://github.com/openwrt/openwrt/commit/b1df48caac1b3f75a65768a5abf8de84801edbf0

Should not fail at early  boot though....

h1618, 2023-11-07:
Ok, I tried sysupgrade 21->23 because it worked on a ramips-mt76x8-wavlink_wl-wn575a3, this one requiring a firmware reset afterwards, but all is working well. But flashing the 23 firmware directly via serial should reset the configuration as well, but this one isn't even able to uncompress the linux kernel, so system configuration should not be the cause.

brada4, 2023-11-07:
Does v22 work?

brada4, 2023-11-07:
> LZMA: uncompress or overwrite error 7

One has to guess which of 3 LZMA parameters to tune down. Mozt likely patching **image | build | chain** with lzma -0 (zero which still better than gzip -9) would build always bootable albeit a bit larger kernel image.

jospezial, 2023-11-08:
> Uncompressing Kernel Image ... LZMA: uncompress or overwrite error 7 - must RESET b

That was a bug in u-boot. Please update to a recent build of u-boot from current openwrt release or snapshot.
https://github.com/openwrt/openwrt/commits/main/package/boot/uboot-lantiq

brada4, 2023-11-08:
Hey @jospezial , it is 'fixed' in both current releases ??? And the pre-fix bootloader works fine....

h1618, 2023-11-09:
hello @jospezial , thank you for this hint, I updated u-boot to the 23.05.0 packages u-boot-arv8539pw22_nor version, and after reset, it unpacked the kernel and started it, but this one stopped after a few lines. I then reflashed the 23.05.0 firmware 2 times, making sure that it hasn't been altered by xmodem transfer and cp.b flashing by crc-checking (crc32 b0040000 680147), which gave the correct result. But every time after restart it showed the same stop.
`ARV8539PW22 # reset�
ROM VER: 1.0.3
CFG 01
Read EEPROM.

U-Boot 2013.10-openwrt4 (Oct 12 2023 - 00:59:15) ARV8539PW22

Board: Speedport W 504V Typ A
SoC:   Lantiq Danube-S v1.5
CPU:   333.333 MHz
IO:    166.667 MHz
BUS:   83.333 MHz
BOOT:  NOR
DRAM:  64 MiB
Flash: 8 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   ltq-eth
Hit any key to stop autoboot:  0 
## Booting kernel from Legacy Image at b0040000 ...
   Image Name:   MIPS OpenWrt Linux-5.15.134
   Created:      2023-10-09  21:45:35 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2725739 Bytes = 2.6 MiB
   Load Address: 80002000
   Entry Point:  80002000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.15.134 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23497-6637af95aa) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Oct 9 21:45:35 23
[    0.000000] SoC: Danube rev 1.5
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019641 (MIPS 24KEc)
[    0.000000] MIPS: machine is Speedport W 504V Typ A
[    0.000000] Initrd not found or empty - disabling initrd
`
so it seems this package is defect.

@brada4: yes 22.02.0 works, the start with this previous version continues after that:
`...
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 16kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 16kB, 4-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyLTQ0,115200
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00002028
[    0.000000] Readback ErrCtl register=00002028
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55484K/65536K available (6456K kernel code, 626K rwdata, 824K rodata, 1228K init, 209K bss, 10052K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 333MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 11467562657 ns
[    0.000022] sched_clock: 32 bits at 166MHz, resolution 6ns, wraps every 12884901885ns
[    0.008117] Calibrating delay loop... 221.18 BogoMIPS (lpj=1105920)
[    0.094313] pid_max: default: 32768 minimum: 301
...`

the latter is the 22.03.0 version, and I think the 23.05.0 version should have continued in a similar form.

brada4, 2023-11-09:
something broke after v22. You may use v22.03.5 in the meantime if this is your only device.

jospezial, 2023-11-09:
You could give the snapshot a try.
If you know how to setup a tftp-server  on your computer then use tftp to transfer the images to that router.
It is way more faster than xmodem.
The "bad CRC" warning goes away if you save the env.

h1618, 2023-11-10:
@jospezial: I tried the current snapshot, but the result is the same:

`U-Boot 2013.10-openwrt4 (Oct 12 2023 - 00:59:15) ARV8539PW22                                                            
                                                                                                                        
Board: Speedport W 504V Typ A                                                                                           
SoC:   Lantiq Danube-S v1.5
CPU:   333.333 MHz
IO:    166.667 MHz
BUS:   83.333 MHz
BOOT:  NOR
DRAM:  64 MiB
Flash: 8 MiB
In:    serial
Out:   serial
Err:   serial
Net:   ltq-eth
Hit any key to stop autoboot:  0 
## Booting kernel from Legacy Image at b0040000 ...
   Image Name:   MIPS OpenWrt Linux-5.15.137
   Created:      2023-11-07  23:11:05 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2894147 Bytes = 2.8 MiB
   Load Address: 80002000
   Entry Point:  80002000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.15.137 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24328-91169898ce) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Tue Nov 7 23:11:05 23
[    0.000000] SoC: Danube rev 1.5
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019641 (MIPS 24KEc)
[    0.000000] MIPS: machine is Speedport W 504V Typ A
[    0.000000] Initrd not found or empty - disabling initrd

`
... and stopping there - something seems to be broken in the generation of the lantic-xway-arcadian_arv8539pw22 firmware.

jospezial, 2023-11-19:
@ynezz The apm821xx label is wrong here. It is a lantiq device.