openwrt/openwrt #13894: Archer C60	v2	23.05.0 factory bin gives 404

uandhgroup, 2023-11-07:
### Describe the bug

[-generic-tplink_archer-c60-v2-squashfs-factory.bin](https://downloads.openwrt.org/releases/23.05.0/targets/ath79/generic/openwrt-23.05.0-ath79-generic-tplink_archer-c60-v2-squashfs-factory.bin)

gives 404.

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

23.05.0

### Device

Archer C60

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-07:
You can download v22 factory and sysupgrade to v23 or make 'frankenstien' image as in device page.
The factory image is supposed to be where  you get 404.

closed 2023-11-07 by uandhgroup

brada4, 2023-11-07:
There should be some change or admission of glitch on why the image is not there to make sure 23.05.1 has it back.