openwrt/openwrt #13902: Channels 149 and 165 in the 5G WiFi network cannot be used;

Tonkercke, 2023-11-08:
### Describe the bug

Channels 149 and 165 under 5G WiFi cannot be used. Change the channel to 149 and save it, but when used, it becomes 153. Change the channel to 165, and the wireless signal icon becomes disabled when using it.

After testing on multiple platforms, this problem exists.

### OpenWrt version

openwrt 23.05

### OpenWrt target/subtarget

openwrt 23.05.0

### Device

xiaomi R3G;redmi ac2100;R8000

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-08:
You are in EU, 80MHz 165 is not available, and 153 is super low power. Please specify exact device next time, seriously.
Please post `/etc/config/wireless` if dealing with regdom issues.

Normally:
Avoid SRD 13dBm/25mW channels
`list hostapd_options 'min_tx_power=20'`
And use automatic channel.

Shine-, 2023-11-08:
> ### Device
> xiaomi R3G;redmi ac2100;R8000

Unfortunately, you're listing three devices with **completely different** 5G chips (R3G: MT7612, AC2100: MT7615, R8000: BCM43602), without specifying which issue you have with which one of these devices.

Putting "no response" to all questions doesn't really help at all, either. If you don't give any details, or steps to reproduce, your issue will likely get ignored by anybody who could help.

So let me cut down on a few _hypothetical_ possibilities:

1. Channel 149 becoming 153: Unless you've configured your device to 20MHz, thats perfectly fine and 100% correct: ch149+153 is one 40MHz channel, or the lower half of a 80MHz channel (or the lower quarter of the US-only 160MHz ch149 - if any of the given chipset even supported 160MHz @ ch149). Therefore: not a bug, it's working just as it should.
2. Channel 165 failing: ch165 is a valid 20/40MHz channel **only** in most regdomains. Only in the US regdomain, ch165 @ 80 MHz is valid (and also part of ch149 @ 160 MHz), but likely unsupported by any of the chipsets in your devices, since the hardware precedes the regulatory change in US. You're not stating your chosen regdomain nor the channel width you selected. So again, we have to **guess** what you're trying to do. I suspect that you simply selected an invalid channel/bandwidth configuration. Therefore again, working as it should.
4. One word about Netgear R8000: I had the chance to play around with that model twice in my life (OpenWrt 19.07 times and OpenWrt 22.03 times). For me, 5G on that device never worked properly in OpenWrt. This is likely because OpenWrt enables different channels for each of the two 5G chips than the stock firmware does - probably channels that are blocked by firmware or EEPROM (may also depend on the region where the R8000 was purchased). So if you have problems with 5G on the R8000, you may want to open a separate issue, and put a **detailed and comprehensive description of what you're trying to do**. Please do also make sure that your configuration is in fact supported by chipset and by regdomain.

Since I can see quite frequently here that people are complaining about what they perceive as bugs, but which are in fact just invalid channel/bandwidth/tx-power combinations, it might be a good idea that OpenWrt sanitized such invalid configurations on its own, instead of relying on the WiFi driver to do so (most don't). That might be a valid feature request for future versions - feel free to create one, if you like.

Tonkercke, 2023-11-10:
> > ### 设备 
> > 小米R3G;红米ac2100;R8000
> 
> 不幸的是，您列出了三种具有 **完全不同的 **5G 芯片的设备（R3G：MT7612、AC2100：MT7615、R8000：BCM43602），而没有具体说明您对其中哪一款设备遇到的问题。
> 
> 对所有问题都设置“不回复” - 并且完全不提供有关您的配置的详细信息 - 也根本没有帮助。 如果您不提供任何详细信息或重现步骤，您的问题可能会被任何可以提供帮助的人忽略。
> 
> 因此，让我减少一些 _假设的 _可能性：
> 
>     1. 频道 149 变为 153：除非您已将设备配置为 20MHz，否则完全没问题且 100% 正确：ch149+153 是一个 40MHz 频道，或 80MHz 频道的下半部分（或仅美国 160MHz 的下四分之一） ch149 - 如果任何给定的芯片组甚至支持 160MHz @ ch149）。  因此：这不是一个错误，它正在正常工作。
> 
>     2. 是有效的 20/40MHz **通道 165 失败：ch165 在大多数注册域中仅 **通道。   仅在美国监管域中，ch165 @ 80 MHz 有效（也是 ch149 @ 160 MHz 的一部分），但设备中的任何芯片组可能都不支持，因为硬件先于美国的监管变更。   您没有说明您选择的注册域或您选择的通道宽度。   再说一次，我们必须 **猜测 **你想做什么。   我怀疑您只是选择了无效的通道/带宽组合。   因此，再次，按其应有的方式工作。
> 
>     3. 关于 Netgear R8000 的一句话：我一生中有两次机会使用该型号（OpenWrt 18.06 或 19.07 次和 OpenWrt 22.03 次）。   对我来说，该设备上的 5G 在 OpenWrt 中从未正常工作。   这可能是因为 OpenWrt 为两个 5G 芯片启用了与库存固件不同的通道 - 可能被 fullmac 固件或 EEPROM 阻止的通道（也可能取决于购买 R8000 的区域）。   因此，如果您在 R8000 上遇到 5G 问题，您可能需要打开一个单独的问题，并 **详细而全面地描述您正在尝试执行的操作 **。   还请确保您的配置实际上受到芯片组和注册域的支持。
> 
> 
> 由于我在这里经常看到人们抱怨他们认为的错误，但实际上只是无效的通道/带宽/发射功率组合，因此 OpenWrt（或 LuCI）清理此类无效配置可能是个好主意自行完成，而不是依赖 WiFi 驱动程序（大多数情况下不这样做）。 这可能是未来版本的有效功能请求 - 如果您愿意，请随意创建一个。

I tested it on three devices, and they all had the same problem. However, channels 149 and 165 can be used under openwrt21.02 and 22.03 systems. Also, the country code selected in my wireless network card settings is TW, while the R8000's is US. Now I change the country code to another country and still cannot use channels 149 and 165.

brada4, 2023-11-10:
Least power/most restrictions are also obeyed by regulatory domain on your client device. Sometimes you need to completely power off your AP to change second time to CORRECT regulatory domain. Dont get creative around it , please.

Shine-, 2023-11-10:
Just to stress it again: You can expect anybody, who might be able to help, to ignore your issue, since you're still not posting any detailed configuration settings or steps to reproduce your problem.

Many if not most people, including myself, are running ch149@20/40/80MHz or ch165@20/40MHz without any problems on many OpenWrt versions, including 23.05.0, with MT7612 and MT7615 5G chips (if the chosen regdomain allows it).

I just tried on a test system.
With TW regdomain, ch165 is only enabled @ 20MHz - in both 22.03 and 23,05, so no change.

I'm not familiar with TW regdomain, but `iw reg get` says:
```
country TW: DFS-FCC
[...]
        (5725 - 5850) @ 80), (N/A, 30), (N/A)
[...]
```
Which means ch169, which is ch165's partner channel for 40 MHz, is not allowed, since it spans till 5855MHz.
If the regdb is correct, which I have to presume, then OpenWrt is fully right in refusing to use ch165@40MHz in TW regdomain. Again, ch165@20MHz works fine, even in TW!

As stated, this is identical for OpenWrt 22.03 and 23.05. The limitation imposed by the regdomain setting is obeyed, therefore working as it should.

If you think the regulatory db is incorrect for TW, then you might want to request a correction upstream, ie. in the Liunx kernel.

Tonkercke, 2023-11-12:
…………
> If you think the regulatory db is incorrect for TW or US, you might want to request a correction upstream, ie. in the Linux kernel. RegDB changes usually find their way into OpenWrt rather quickly.

So that’s how it is! Thanks for helping me answer this.

Neustradamus, 2025-01-16:
@Tonkercke: Any news about it, it has been solved?