openwrt/openwrt #13903: Problem with wireless function of NETGEAR R8000 router.

Tonkercke, 2023-11-08:
### Describe the bug

Problem 1: The broadcast WiFi signal does not show the signal strength.
[image]

Question 2: The 'randomly generated' MAC address function added in openwrt 23.05 is very easy to use. It will automatically generate a new MAC address when broadcasting WiFi signals. But when I used wireless to relay another wireless signal, I found that the MAC address was always the original MAC address without any change. Even if I restarted the wireless network card or restarted the system, there was still no change. Hope this problem is fixed.
    

### OpenWrt version

openwrt 23.05

### OpenWrt target/subtarget

openwrt 23.05.0

### Device

NETGEAR R8000

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-08:
You need to associate something to get link speed. You can discuss cosmetic annoyances in forum, it is not a bug.

raenye, 2023-11-10:
> You need to associate something to get link speed. You can discuss cosmetic annoyances in forum, it is not a bug.

This is not the problem. The problem is that R8000 is Broadcom.
To get this information, the wireless driver has to support reporting the information. `brcmfmac` sadly manifests the lousy support of the firmware.

To test whether the driver can read this information, run:
`iw dev phy0-ap0 station dump` or `iwinfo phy0-ap0 a`.

Tonkercke, 2023-11-12:
> > Problem 1: The broadcast WiFi signal does not show the signal strength.
> 
> Was it working for you previously? AFAIK, this never worked on Netgear R8000 (and also Linksys EA9200, which I have).
> 
> > You need to associate something to get link speed. You can discuss cosmetic annoyances in forum, it is not a bug.
> 
> This is not the real problem. The problem is Broadcom. To get this information, the wireless driver has to support reporting the information. `brcmfmac` sadly manifests the lousy support of the firmware.
> 
> To test whether the driver can read this information, run `iw dev phy0-ap0 station dump` or `iwinfo phy0-ap0 a`.
> 
> Example output from `mt76`: root@axe:~# iwinfo phy1-ap0 a xx:xx:xx:xx:xx:xx -46 dBm / -92 dBm (SNR 46) 4200 ms ago RX: 24.0 MBit/s 239395 Pkts. TX: 300.0 MBit/s, MCS 15, 40MHz 101267 Pkts. expected throughput: unknown
> 
> yy:yy:yy:yy:yy:yy -46 dBm / -92 dBm (SNR 46) 220 ms ago RX: 6.0 MBit/s 672246 Pkts. TX: 6.0 MBit/s 34273 Pkts. expected throughput: unknown
> 
> Example output from `brcmfmac`: root@Linksys:~# iwinfo phy2-ap0 a xx:xx:xx:xx:xx:xx unknown / unknown (SNR 0) 0 ms ago RX: 866.6 MBit/s 251 Pkts. TX: 585.0 MBit/s 755339 Pkts. expected throughput: unknown
> 
> Furthermore, currently the OpenWrt `brcmfmac` implementation does not include the dump_obss patch found here: https://patchwork.kernel.org/project/linux-wireless/patch/20220929012527.4152-2-ian.lin@infineon.com/ but adding it still doesn't solve the problem, as the relevant "ivar" (whatever that means) `dump_obss` is missing from the BCM43602 firmware, so `BRCMF_FEAT_DUMP_OBSS` is disabled.

ok, I understand, thank you for helping me answer.

dotignore, 2024-03-03:
all WiF doesnt work

[   11.480610] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43602-pcie for chip BCM43602/1
[   11.489740] pci 0001:00:00.0: enabling device (0140 -> 0142)
[   11.495488] pci 0001:01:00.0: enabling device (0140 -> 0142)
[   11.501191] pci 0001:02:01.0: enabling device (0140 -> 0142)
[   11.506877] brcmfmac 0001:03:00.0: enabling device (0140 -> 0142)
[   11.517071] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.netgear,r8000.bin failed with error -2
[   11.633693] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
[   11.648508] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.clm_blob failed with error -2
[   11.650692] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43602-pcie for chip BCM43602/1
[   11.668019] pci 0001:02:02.0: enabling device (0140 -> 0142)
[   11.673822] brcmfmac 0001:04:00.0: enabling device (0140 -> 0142)
[   11.681542] brcmfmac 0001:03:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.netgear,r8000.bin failed with error -2
[   11.693348] brcmfmac 0001:03:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
[   11.703353] urngd: v1.0.2 started.
[   11.720501] brcmfmac 0001:03:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.clm_blob failed with error -2
[   11.841236] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43602-pcie for chip BCM43602/1
[   11.850257] brcmfmac 0001:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.netgear,r8000.bin failed with error -2
[   11.877986] brcmfmac 0001:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
[   11.891111] kmodloader: done loading kernel modules from /etc/modules.d/*
[   11.912212] brcmfmac 0001:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.clm_blob failed with error -2
[   13.830406] ieee80211 phy0: brcmf_msgbuf_query_dcmd: Timeout on response for query command
[   13.838716] ieee80211 phy0: brcmf_c_preinit_dcmds: Retrieving cur_etheraddr failed, -5
[   13.846691] ieee80211 phy0: brcmf_bus_started: failed: -5
[   13.852246] ieee80211 phy0: brcmf_attach: dongle is not responding: err=-5
[   13.918826] ieee80211 phy1: brcmf_msgbuf_query_dcmd: Timeout on response for query command
[   13.927276] ieee80211 phy1: brcmf_c_preinit_dcmds: Retrieving cur_etheraddr failed, -5
[   13.935289] ieee80211 phy1: brcmf_bus_started: failed: -5
[   13.940758] ieee80211 phy1: brcmf_attach: dongle is not responding: err=-5
[   14.150402] ieee80211 phy2: brcmf_msgbuf_query_dcmd: Timeout on response for query command
[   14.158747] ieee80211 phy2: brcmf_c_preinit_dcmds: Retrieving cur_etheraddr failed, -5
[   17.774030] bgmac_bcma bcma0:5 eth2: Link is Up - 1Gbps/Full - flow control off
[   17.784331] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[   17.791505] b53-srab-switch 18007000.ethernet-switch lan1: configuring for phy/gmii link mode
[   17.803141] br-lan: port 1(lan1) entered blocking state
[   17.808415] br-lan: port 1(lan1) entered disabled state
[   17.814123] device lan1 entered promiscuous mode
[   17.818780] device eth2 entered promiscuous mode
[   17.933935] b53-srab-switch 18007000.ethernet-switch lan2: configuring for phy/gmii link mode
[   17.952548] br-lan: port 2(lan2) entered blocking state
[   17.957832] br-lan: port 2(lan2) entered disabled state
[   17.963589] device lan2 entered promiscuous mode
[   18.056077] b53-srab-switch 18007000.ethernet-switch lan3: configuring for phy/gmii link mode
[   18.069289] br-lan: port 3(lan3) entered blocking state
[   18.074655] br-lan: port 3(lan3) entered disabled state
[   18.080338] device lan3 entered promiscuous mode
[   18.182476] b53-srab-switch 18007000.ethernet-switch lan4: configuring for phy/gmii link mode
[   18.196987] br-lan: port 4(lan4) entered blocking state
[   18.202314] br-lan: port 4(lan4) entered disabled state
[   18.215966] device lan4 entered promiscuous mode
[   18.321625] b53-srab-switch 18007000.ethernet-switch wan: configuring for phy/gmii link mode
[   18.524916] leds bcm53xx:white:usb2: Failed to get trigger sources for /leds/led-usb2
[   18.532961] leds bcm53xx:white:usb2: Failed to get trigger sources for /leds/led-usb2
[   18.540928] leds bcm53xx:white:usb2: Failed to get trigger sources for /leds/led-usb2
[   18.548825] leds bcm53xx:white:usb2: Failed to get trigger sources for /leds/led-usb2
[   18.572982] leds bcm53xx:white:usb3: Failed to get trigger sources for /leds/led-usb3
[   18.580987] leds bcm53xx:white:usb3: Failed to get trigger sources for /leds/led-usb3
[   18.588883] leds bcm53xx:white:usb3: Failed to get trigger sources for /leds/led-usb3
[   18.596899] leds bcm53xx:white:usb3: Failed to get trigger sources for /leds/led-usb3
[   22.401081] b53-srab-switch 18007000.ethernet-switch lan4: Link is Up - 1Gbps/Full - flow control rx/tx
[   22.410553] br-lan: port 4(lan4) entered blocking state
[   22.415800] br-lan: port 4(lan4) entered forwarding state
[   22.421792] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   22.481121] b53-srab-switch 18007000.ethernet-switch wan: Link is Up - 1Gbps/Full - flow control off
[   22.490360] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[   11.517071] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.netgear,r8000.bin failed with error -2
[   11.633693] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
[   11.648508] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.clm_blob failed with error -2
[   11.650692] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43602-pcie for chip BCM43602/1
[   11.668019] pci 0001:02:02.0: enabling device (0140 -> 0142)
[   11.673822] brcmfmac 0001:04:00.0: enabling device (0140 -> 0142)
[   11.681542] brcmfmac 0001:03:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.netgear,r8000.bin failed wit

raenye, 2024-03-03:
Post your /etc/config/wireless (without passwords).

dotignore, 2024-03-03:
No in web interface is absents

[image]

raenye, 2024-03-03:
GUI doesn't have direct access to config files.
Either connect via `ssh` or download a backup to your computer and find the file within the `tar.gz` backup file.

dotignore, 2024-03-04:
[backup-OpenWrt-2023-11-14.tar.gz](https://github.com/openwrt/openwrt/files/14486977/backup-OpenWrt-2023-11-14.tar.gz)

raenye, 2024-03-04:
You should delete it from GitHub, it contains private information.
Just post the relevant information here, as text.
Make it easy for people to help you.

dotignore, 2024-03-05:
[image]

[image]

freshtomato is work good any wifi is visible and full function instead
https://freshtomato.org/downloads/freshtomato-arm/2024/2024.1/Netgear%20initial%20files/

brada4, 2024-03-05:
it is well documented. 
https://openwrt.org/docs/techref/hardware/soc/soc.broadcom.bcm53xx

raenye, 2024-03-05:
#14685 might be related, btw.

Abdoymani, 2025-10-23:
Despite having `brcmfmac-firmware-43602a1-pcie` installed, the driver reports missing `.txt`, `.clm_blob`, and `.txcap_blob` files.

---

### What I have tried

- Installed all related packages: `kmod-brcmfmac`, `brcmfmac-firmware-43602a1-pcie`.
- Extracted multiple Netgear stock firmwares (`R8000-V1.0.4.xx.chk`) using `binwalk` — **no CLM or TXCAP blobs were found**.
- Tested firmwares from Asus RT-AC88U and Linksys EA8500 (same BCM43602 chipset) — only partial success.
- Attempted to download blobs from `linux-firmware.git` and several mirrors, but these specific files are missing for BCM43602 PCIe.
- Created a custom NVRAM `.txt` file to extend channel support; partial improvement, but TX queue timeouts persist.

---

### What I need

Could anyone please:
1. Share working versions of the following files (for BCM43602 PCIe):
   - `brcmfmac43602-pcie.txt`
   - `brcmfmac43602-pcie.clm_blob`
   - `brcmfmac43602-pcie.txcap_blob`
2. Or confirm whether these blobs are embedded in the proprietary Broadcom firmware and not distributed publicly?

Any help or pointers would be greatly appreciated.
This issue is reproducible on OpenWrt (latest snapshot, kernel 6.6.x) and affects Wi-Fi channel availability and TX stability.

---

### References
- Similar discussions on OpenWrt and linux-firmware mailing lists mention the same missing blobs for BCM43602.
- Device tree and dmesg logs available upon request.