openwrt/openwrt #13906: iptables-nft v1.8.8: incorrect port translation

arturioxas, 2023-11-08:
### Describe the bug

Not sure if I should be filing this straight to `netfilter` developers, so feel free to disregard if needed.
Short summary is:

- iptables version is: iptables v1.8.8 (nf_tables), the build is OpenWrt 23.05 for x86-64 VM
- I'm trying to add this rule (that should be translated info nft​ form): `/usr/sbin/iptables -t filter -A INPUT -p udp -m udp --sport 443 -m connmark ! --mark  0x2/0x2 -m mark ! --mark 0x1/0x1`
- It gets translated into: `“A INPUT -p udp -m udp ! --sport 256 ! --dport 0 -m connmark ! --mark 0x2/0x2 -m mark ! --mark 0x1/0x1`

`nft` translation for my untrained eye looks fine​. As you can see, `--sport​` and `--dport`​ have changed. If I remove `-m -mark`​​ part, port is inserted correctly. `iptables-translate`​ does the translation correctly like I'd expect.

### OpenWrt version

OpenWrt 23.05.0, r23497-6637af95aa

### OpenWrt target/subtarget

x86_64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Self-built image

### Steps to reproduce

Add a rule like this:
```
/usr/sbin/iptables -t filter -A INPUT -p udp -m udp --sport 443 -m connmark ! --mark  0x2/0x2 -m mark ! --mark 0x1/0x1
```

### Actual behaviour

Ports translation in iptables form is incorrect:
```
 -A INPUT -p udp -m udp ! --sport 256 ! --dport 0 -m connmark ! --mark 0x2/0x2 -m mark ! --mark 0x1/0x1
```

### Expected behaviour

All of items translated properly into both nft and iptables forms

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

arturioxas, 2023-11-08:
I think I was misunderstood. I am using `iptables`, not `iptables-translate`. I'm inserting a rule with `iptables` and then I list all the rules via `iptables-save` and that's the rule I get

brada4, 2023-11-08:
Please attach `opkg list-installed` , i get perfect readback having all iptables/ipt/xt kmods installed.

arturioxas, 2023-11-08:
Thank you for the swift answer!
If you can tell me what the list should be, I'd then add the missing ones and we could close this ticket.
```
$ opkg list-installed | grep -e 'nft' -e 'ipt' -e 'xt'
ip6tables-nft - 1.8.8-1
iptables-mod-conntrack-extra - 1.8.8-1
iptables-mod-filter - 1.8.8-1
iptables-mod-nfqueue - 1.8.8-1
iptables-nft - 1.8.8-1
kmod-ipt-conntrack - 5.15.134-1
kmod-ipt-conntrack-extra - 5.15.134-1
kmod-ipt-core - 5.15.134-1
kmod-ipt-filter - 5.15.134-1
kmod-ipt-ipset - 5.15.134-1
kmod-ipt-nat - 5.15.134-1
kmod-ipt-nat6 - 5.15.134-1
kmod-ipt-nfqueue - 5.15.134-1
kmod-lib-textsearch - 5.15.134-1
kmod-nf-ipt - 5.15.134-1
kmod-nf-ipt6 - 5.15.134-1
kmod-nft-compat - 5.15.134-1
kmod-nft-core - 5.15.134-1
kmod-nft-fib - 5.15.134-1
kmod-nft-nat - 5.15.134-1
kmod-nft-offload - 5.15.134-1
libext2fs2 - 1.47.0-2
libiptext-nft0 - 1.8.8-1
libiptext0 - 1.8.8-1
libiptext6-0 - 1.8.8-1
libjson-script20230523 - 2023-05-23-75a3b870-1
libnftnl11 - 1.2.6-1
libxtables12 - 1.8.8-1
nftables-json - 1.0.8-1
xtables-nft - 1.8.8-1
```

brada4, 2023-11-08:
(Stop asking USAGE questions every day, there are forums for that)

arturioxas, 2023-11-08:
I think I can close the ticket, it clearly a problem of module loading. Thank you for the input, it was helpful

closed 2023-11-08 by arturioxas