openwrt/openwrt #13907: 在Ubuntu 22.04.3系统上用23.05.0版代码本地编译运行make defconfig出现以下问题

Jewell1024, 2023-11-08:
### Describe the bug

tmp/.config-package.in:33662:error: recursive dependency detected!
tmp/.config-package.in:33662:	symbol PACKAGE_python3-pymysql depends on PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT
/home/amy/openwrt23/feeds/packages/lang/python/pymysql/Config.in:4:	symbol PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT depends on PACKAGE_python3-pymysql
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

### OpenWrt version

. /etc/openwrt_release && echo $DISTRIB_REVISION

### OpenWrt target/subtarget

. /etc/openwrt_release && echo $DISTRIB_TARGET

### Device

cat /tmp/sysinfo/model

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-08:
Can you add diffconfig? mysql is not build by default anywhere.

Jewell1024, 2023-11-09:
> 可以添加 diffconfig 嗎？ mysql 不是在任何地方預設建置的。

Is it possible to add the above?

brada4, 2023-11-09:
You are supposed to run diffconfig script in your build tree to show changed configuration options.

Jewell1024, 2023-11-09:
I haven't changed anything, but the problem occurred when running make defconfig. Does it affect compilation?

brada4, 2023-11-09:
either you did not update/install feeds or missed out default build config seed.
https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem#using_official_build_config

Jewell1024, 2023-11-09:
> 您沒有更新/安裝提要或錯過了預設建置配置種子。 [https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem#using_official_build_config ](https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem#using_official_build_config)
tmp/.config-package.in:33673:error: recursive dependency detected!
tmp/.config-package.in:33673:	symbol PACKAGE_python3-pymysql depends on PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT
/home/amy/openwrt23/feeds/packages/lang/python/pymysql/Config.in:4:	symbol PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT depends on PACKAGE_python3-pymysql
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:33673:error: recursive dependency detected!
tmp/.config-package.in:33673:	symbol PACKAGE_python3-pymysql depends on PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT
/home/amy/openwrt23/feeds/packages/lang/python/pymysql/Config.in:4:	symbol PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT depends on PACKAGE_python3-pymysql
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

Jewell1024, 2023-11-10:
I followed the standard operating procedures. When using the code from the 23.05.0 branch, the above problems will occur as long as I run make defconfig. However, there will be no problems when using the code from the 22.03.5 branch. Could you please take a look at what's going on?

gaoderby, 2023-11-10:
After a certain update, this issue has occurred for almost a month. Replacing feeds/packages/lang/Python/pymysql/Makefile with the following content can solve the problem. However, we hope that the official can address this issue:
（某次更新后出现此问题，差不多有一个多月了，把feeds/packages/lang/python/pymysql//Makefile替换为如下内容可解决。但，希望官方能解决此问题：）
#
# Copyright (C) Alexandru Ardelean <ardeleanalex@gmail.com>
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=pymysql
PKG_VERSION:=1.1.0
PKG_RELEASE:=2

PYPI_NAME:=PyMySQL
PKG_HASH:=4f13a7df8bf36a51e81dd9f3605fede45a4878fe02f9236349fd82a3f0612f96

PKG_MAINTAINER:=Alexandru Ardelean <ardeleanalex@gmail.com>
PKG_LICENSE:=MIT
PKG_LICENSE_FILES:=LICENSE

PKG_CONFIG_DEPENDS:=CONFIG_PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT

include ../pypi.mk
include $(INCLUDE_DIR)/package.mk
include ../python3-package.mk

define Package/python3-pymysql/Default
  SUBMENU:=Python
  SECTION:=lang
  CATEGORY:=Languages
  TITLE:=Pure Python MySQL Client
  URL:=https://pymysql.readthedocs.io/
endef

define Package/python3-pymysql
$(call Package/python3-pymysql/Default)
  DEPENDS:=+python3
endef

define Package/python3-pymysql/description
  This package contains a pure-Python MySQL client library, based on PEP 249.
endef

define Package/python3-pymysql-sha-pwd
$(call Package/python3-pymysql/Default)
  TITLE+=w/ SHA256 password auth
  DEPENDS:=+python3-pymysql $(RUST_ARCH_DEPENDS) +PACKAGE_python3-pymysql-sha-pwd:python3-cryptography
endef

define Package/python3-pymysql-sha-pwd/description
  This is a meta-package installing python3-pymysql and python3-cryptography
  packages to be able to use pymysql with “sha256_password” or
  “caching_sha2_password” for authentication.
endef

define Package/python3-pymysql-sha-pwd/install
  true
endef

$(eval $(call Py3Package,python3-pymysql))
$(eval $(call BuildPackage,python3-pymysql))
$(eval $(call BuildPackage,python3-pymysql-src))
$(eval $(call BuildPackage,python3-pymysql-sha-pwd))

gaoderby, 2023-11-10:
After a certain update, this issue has occurred for almost a month. Replacing feeds/packages/lang/Python/pymysql/Makefile with the following content can solve the problem. However, we hope that the official can address this issue:
某次更新后出现此问题，差不多有一个多月了，把feeds/packages/lang/python/pymysql//Makefile替换为如下内容可解决。但，希望官方能解决此问题：

```
#
# Copyright (C) Alexandru Ardelean <ardeleanalex@gmail.com>
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=pymysql
PKG_VERSION:=1.1.0
PKG_RELEASE:=2

PYPI_NAME:=PyMySQL
PKG_HASH:=4f13a7df8bf36a51e81dd9f3605fede45a4878fe02f9236349fd82a3f0612f96

PKG_MAINTAINER:=Alexandru Ardelean <ardeleanalex@gmail.com>
PKG_LICENSE:=MIT
PKG_LICENSE_FILES:=LICENSE

PKG_CONFIG_DEPENDS:=CONFIG_PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT

include ../pypi.mk
include $(INCLUDE_DIR)/package.mk
include ../python3-package.mk

define Package/python3-pymysql/Default
  SUBMENU:=Python
  SECTION:=lang
  CATEGORY:=Languages
  TITLE:=Pure Python MySQL Client
  URL:=https://pymysql.readthedocs.io/
endef

define Package/python3-pymysql
$(call Package/python3-pymysql/Default)
  DEPENDS:=+python3
endef

define Package/python3-pymysql/description
  This package contains a pure-Python MySQL client library, based on PEP 249.
endef

define Package/python3-pymysql-sha-pwd
$(call Package/python3-pymysql/Default)
  TITLE+=w/ SHA256 password auth
  DEPENDS:=+python3-pymysql $(RUST_ARCH_DEPENDS) +PACKAGE_python3-pymysql-sha-pwd:python3-cryptography
endef

define Package/python3-pymysql-sha-pwd/description
  This is a meta-package installing python3-pymysql and python3-cryptography
  packages to be able to use pymysql with “sha256_password” or
  “caching_sha2_password” for authentication.
endef

define Package/python3-pymysql-sha-pwd/install
  true
endef

$(eval $(call Py3Package,python3-pymysql))
$(eval $(call BuildPackage,python3-pymysql))
$(eval $(call BuildPackage,python3-pymysql-src))
$(eval $(call BuildPackage,python3-pymysql-sha-pwd))
```

Jewell1024, 2023-11-10:
Thank you so much ！

brada4, 2023-11-15:
https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem
2 lines  feed sčripts

obsy, 2023-11-15:
What do you suggest? Just download clean openwrt sources, install feeds and there will be recursive dependency error.

brada4, 2023-11-15:
RTFM is a good start.....
```
 # Update the feeds
./scripts/feeds update -a
./scripts/feeds install -a
```

obsy, 2023-11-15:
Did you know that packge/symlinks does exactly what you wrote? RTFM. See https://github.com/openwrt/openwrt/blob/main/include/toplevel.mk#L246

Once again: try it yourself and see it.