openwrt/openwrt #13908: D-Team Newifi D2 23.05.0 can not manage 434M memory

liaoya, 2023-11-08:
### Describe the bug

D-Team Newifi D2 22.03 can manage almost 500M memory. But when I update to 23.05.0, it can only manage 434M memory. I use imagebuilder docker image build firmware and I disable wireless also

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ramips/mt7621

### Device

D-Team Newifi D2

### Image kind

Official downloaded image

### Steps to reproduce

Flash the device with 23.05.0 image.

### Actual behaviour

The device can not find 434M memory.

[image]

### Expected behaviour

The device can manage about 500M memory

### Additional info

I use imagebuilder for firmware.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-08:
now it is 2 ethernet adapters with 12 queues each, see eg tc -s qdisc ?

Shine-, 2023-11-08:
That is on purpose, since enabling highmem support (which is needed to address the full 512 MB memory) was causing a heavy performance hit on MT7621. The change leaves 64MB unusable, which was considered the lesser evil compared to the performance degradation.

See #13151, fixed with cd2b74e01e8d and a110de8152df46a1e2adf7010ba75fb3b1236cd8

liaoya, 2023-11-09:
@Shine- 
Thank you for the information. I will close the ticket.

closed 2023-11-09 by liaoya