openwrt/openwrt #13913: Non-Updated firmware-links for Fritz!Box 4040?

useross, 2023-11-08:
### Describe the bug

Could it be that the links inside the installation and upgrade infobox have been forgotten to update?
https://openwrt.org/toh/avm/avm_fritz_box_4040?datasrt=firmware%20openwrt%20upgrade%20url#installation

As seen in the figure, the versions are different.

[image]

By modifying the URL to `23.05.0`, the Fritz!Box 4040 is listed for the firmware upgrade
https://downloads.openwrt.org/releases/23.05.0/targets/ipq40xx/generic/

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

Webpage

### Device

Fritz!Box 4040

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2023-11-08:
It's a wiki page, you can always propose a change.

brada4, 2023-11-08:
It is a db filter in wiki page, something is missing in wiki toh database.

Its been some time since fw selector comes before static mirror link in release announce:

https://firmware-selector.openwrt.org/?version=23.05.0&target=ipq40xx%2Fgeneric&id=avm_fritzbox-4040

useross, 2024-01-13:
The mismatch is still there.
Shows `23.05.0` (may was updated manually after the initial message here), but current version is `23.05.2`, what can be downloaded by modifying the given link manually.

brada4, 2024-01-13:
It has been like that for years, you dont have to do url hacks for latest varsion, the "latest" images are always in firmware-selector, and browsing directories.

useross, 2024-01-14:
But exactly this was working for the past years. I was always checking this section for updates.
Now, it is no more up to date

[image]

https://openwrt.org/toh/avm/avm_fritz_box_4040?datasrt=firmware%20openwrt%20upgrade%20url#installation

brada4, 2024-01-14:
Not sure how release process is supposed to update database.... Down below in CSV no .2 version anywhere.
https://openwrt.org/supported_devices

useross, 2024-01-14:
The linked database shows for FritzBox4040 version `23.05.0`
[image]

But here, i can find version `23.05.2` for exactly this same model
https://downloads.openwrt.org/releases/23.05.2/targets/ipq40xx/generic/

i am confused. 
So, `23.05.2` is available for `Fritzbox 4040`. 
Does it mean, there is a newer firmware `23.05.2` available for Fritzbox 4040, but not supported? So the latest supported version is `23.05.0` even there is a newer firmware `23.05.2` for exactly this device?

Or is there any manual approval necessary to have updated versions in the database?

brada4, 2024-01-14:
[Release checklist](https://openwrt.org/docs/guide-developer/releases/release-process)
Probably reminder in forums would help, without surplus dramatics