openwrt/openwrt #13916: Device unusable when enabling 5 GHz band

castillofrancodamian, 2023-11-08:
### Describe the bug

After this commit, the device becomes unusable. After [this commit](https://github.com/openwrt/openwrt/commit/7d3651f1b9be522e3849412d0aba010e81e15a51) and [this one](https://github.com/openwrt/openwrt/commit/8de3ee2e79e324fd238491be1e102e7c043a6633)., the device becomes unusable. When the 5 GHz radio is disabled, the device can be used. I also see that it runs out of memory after a few minutes.

### OpenWrt version

r24328-91169898ce

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C60 v1

### Image kind

Self-built image

### Actual behaviour

The device becomes very slow when using the 5 GHz band and clients disconnect. It also cannot be accessed via SSH or LuCI.

### Additional info

ath10k-board-qca9888 - 20230804-1
ath10k-firmware-qca9888-ct - 2020-11-08-1
base-files - 1545-r24328-91169898ce
busybox - 1.36.1-1
ca-bundle - 20230311-1
cgi-io - 2022-08-10-901b0f04-21
curl - 8.4.0-2
ddns-scripts - 2.8.2-41
ddns-scripts-services - 2.8.2-41
dnsmasq - 2.89-6
dropbear - 2022.82-5
firewall4 - 2023-11-03-698a5335-1
fstools - 2023-02-28-bfe882d5-1
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2022-08-13-4c7b720b-2
hostapd-common - 2023-09-08-e5ccbfc6-5
htop - 3.2.2-1
iperf3 - 3.15-1
iw - 5.19-1
iwinfo - 2023-07-01-ca79f641-1
jansson4 - 2.14-3
jshn - 2023-05-23-75a3b870-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 6.1.61-1-92b8717ac3653cfb6f3c8798c6fcff09
kmod-ath - 6.1.61+6.5-1
kmod-ath10k-ct-smallbuffers - 6.1.61+2023-06-05-fadd0768-1
kmod-ath9k - 6.1.61+6.5-1
kmod-ath9k-common - 6.1.61+6.5-1
kmod-cfg80211 - 6.1.61+6.5-1
kmod-crypto-acompress - 6.1.61-1
kmod-crypto-aead - 6.1.61-1
kmod-crypto-ccm - 6.1.61-1
kmod-crypto-cmac - 6.1.61-1
kmod-crypto-crc32c - 6.1.61-1
kmod-crypto-ctr - 6.1.61-1
kmod-crypto-gcm - 6.1.61-1
kmod-crypto-gf128 - 6.1.61-1
kmod-crypto-ghash - 6.1.61-1
kmod-crypto-hash - 6.1.61-1
kmod-crypto-hmac - 6.1.61-1
kmod-crypto-lib-chacha20 - 6.1.61-1
kmod-crypto-lib-chacha20poly1305 - 6.1.61-1
kmod-crypto-lib-curve25519 - 6.1.61-1
kmod-crypto-lib-poly1305 - 6.1.61-1
[... 201 lines cut ...]
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_terminfo=y
# CONFIG_PACKAGE_uclient-fetch is not set
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mini=y
CONFIG_PACKAGE_zram-swap=y
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
CONFIG_TESTING_KERNEL=y
CONFIG_ZRAM_DEF_COMP_LZ4=y
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-09:
You have iperf, wireguard and relayd installed that consume some memory when activated.
ddns-scripts should not be heavy even at the moments thay get activated.
zram-swap to sort of ease that?

Can you check half-way to OOM if zram swap is being used (free , swapon -l)  just to lean to kernel or to user applications more...
For usermode - no rebuilds necessary - enable external syslog if possible and collect it, run htop to see if any process grows too large (first will capture biggest process on crash either way)
For kernel mode - youll need to rebuild kernel with slabinfo and probably kallsyms options, lets leave that for later.

raenye, 2023-11-10:
For OOM you could try the small-buffers version of the ath10k firmware.

brada4, 2023-11-10:
>  kmod-ath10k-ct-smallbuffers - 6.1.61+2023-06-05-fadd0768-1

It is there.... If it crashes predictably one may try various firmwares from candela. On the other hand on a 128MB device default ath10k driver + firmware works perfectly. 80211k measurement leaks above 64MB in a populated place.

raenye, 2023-11-10:
@castillofrancodamian anyway a log would be useful (run `logread -f` in a terminal/console window with 5GHz off, then turn on 5GHz).

brada4, 2023-11-10:
When I enable 11k (just run usteer) some 40MB gets eaten up by kernel in 10 minutes, then usage gets fixed around 80m (30M in running processes as before) i really dont need anything extra, "i am fine". I would project that 64MB device goes down at that point. Here 11k is explicitly not supported.

castillofrancodamian, 2023-11-11:
New cleaner build

```
root@OpenWrt:~# . /etc/openwrt_release && echo $DISTRIB_REVISION
r24364-79ee0d2cee
```

```
[franco@pc openwrt]$ ./scripts/diffconfig.sh
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_tplink_archer-c60-v1=y
CONFIG_LINUX_6_1=y
CONFIG_LUCI_LANG_es=y
# CONFIG_PACKAGE_ca-bundle is not set
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libmbedtls=m
CONFIG_PACKAGE_libustream-mbedtls=m
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-i18n-base-es=y
CONFIG_PACKAGE_luci-i18n-firewall-es=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-theme-openwrt-2020=y
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
# CONFIG_PACKAGE_opkg is not set
# CONFIG_PACKAGE_ppp is not set
# CONFIG_PACKAGE_procd-ujail is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mini=y
CONFIG_TESTING_KERNEL=y
CONFIG_PACKAGE_libpthread=y
```
```
[franco@pc generic]$ cat openwrt-ath79-generic-tplink_archer-c60-v1.manifest 
ath10k-board-qca9888 - 20230804-1
ath10k-firmware-qca9888-ct - 2020-11-08-1
base-files - 1547-r24364-79ee0d2cee
busybox - 1.36.1-1
cgi-io - 2022-08-10-901b0f04-21
dnsmasq - 2.89-6
dropbear - 2022.82-5
firewall4 - 2023-11-03-698a5335-1
fstools - 2023-05-28-3a07943e-1
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2022-08-13-4c7b720b-2
hostapd-common - 2023-09-08-e5ccbfc6-6
iw - 5.19-1
iwinfo - 2023-07-01-ca79f641-1
jansson4 - 2.14-3
jshn - 2023-05-23-75a3b870-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 6.1.62-1-1fcd16cdf00951d869170daa56e4e2ac
[... 661 lines cut ...]
Sat Nov 11 14:39:05 2023 daemon.notice hostapd: 5g_test: interface state COUNTRY_UPDATE->ACS
Sat Nov 11 14:39:05 2023 daemon.notice hostapd: 5g_test: ACS-STARTED
Sat Nov 11 14:39:05 2023 daemon.notice netifd: Wireless device 'radio1' is now up
Sat Nov 11 14:39:05 2023 daemon.notice netifd: Wireless device 'radio0' is now up
Sat Nov 11 14:39:18 2023 daemon.notice netifd: wan (2070): udhcpc: sending renew to server 192.168.50.1
Sat Nov 11 14:39:18 2023 daemon.notice netifd: wan (2070): udhcpc: lease of 192.168.50.234 obtained from 192.168.50.1, lease time 300
Sat Nov 11 14:39:24 2023 daemon.notice hostapd: 5g_test: ACS-COMPLETED freq=5680 channel=136
Sat Nov 11 14:39:24 2023 daemon.notice hostapd: 5g_test: interface state ACS->HT_SCAN
Sat Nov 11 14:39:24 2023 daemon.notice hostapd: 5g_test: interface state HT_SCAN->DFS
Sat Nov 11 14:39:24 2023 daemon.notice hostapd: 5g_test: DFS-CAC-START freq=5680 chan=136 sec_chan=-1, width=1, seg0=138, seg1=0, cac_time=60s
Sat Nov 11 14:40:29 2023 daemon.notice hostapd: 5g_test: DFS-CAC-COMPLETED success=1 freq=5680 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5690 cf2=0
Sat Nov 11 14:40:29 2023 kern.info kernel: [ 2245.611718] ath10k_pci 0000:00:00.0: mac flush null vif, drop 0 queues 0xffff
Sat Nov 11 14:40:29 2023 kern.info kernel: [ 2245.990532] IPv6: ADDRCONF(NETDEV_CHANGE): 5g_test: link becomes ready
Sat Nov 11 14:40:29 2023 kern.info kernel: [ 2245.997594] br-lan: port 3(5g_test) entered blocking state
Sat Nov 11 14:40:29 2023 kern.info kernel: [ 2246.003319] br-lan: port 3(5g_test) entered forwarding state
Sat Nov 11 14:40:29 2023 daemon.notice netifd: Network device '5g_test' link is up
Sat Nov 11 14:40:29 2023 daemon.notice hostapd: 5g_test: interface state DFS->ENABLED
Sat Nov 11 14:40:29 2023 daemon.notice hostapd: 5g_test: AP-ENABLED
Sat Nov 11 14:41:48 2023 daemon.err uhttpd[1489]: [info] luci: accepted login on /admin/status/overview for root from 192.168.1.110
Sat Nov 11 14:41:48 2023 daemon.notice netifd: wan (2070): udhcpc: sending renew to server 192.168.50.1
Sat Nov 11 14:41:48 2023 daemon.notice netifd: wan (2070): udhcpc: lease of 192.168.50.234 obtained from 192.168.50.1, lease time 300
Sat Nov 11 14:42:19 2023 authpriv.info dropbear[3526]: Child connection from 192.168.1.110:46918
Sat Nov 11 14:42:22 2023 authpriv.notice dropbear[3526]: Password auth succeeded for 'root' from 192.168.1.110:46918

```

brada4, 2023-11-11:
16 mb is reserved by 
sysctl vm.min_free_kbytes
Setting that to 1mb (1024) in sysctl.conf should not hurt a lot
I wonder if upstream somehow dropped lowmem heuristics at some point.
REF: some kernel sysctl doc saying to have it >=1MB
REF: https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/6/html/performance_tuning_guide/s-memory-tunables
Rest looks quite sane.

castillofrancodamian, 2023-11-11:
I tried the non-CT version and it no longer runs out of memory or reboots like the CT version.

raenye, 2023-11-12:
> I tried the non-CT version and it no longer runs out of memory or reboots like the CT version.

Plus you get 802.11s supported.

castillofrancodamian, 2023-11-12:
@raenye I have noticed that with ath10k I have about 12 db less signal than with ath10k-ct. I still went back to ath10k-ct and I think it has already been solved with this [commit](https://github.com/openwrt/openwrt/commit/36b1dd75fd9da1ea13f1ce1ee679c6b3c9c402cc). I have no reboots or crashes.