openwrt/openwrt #13919: rtl8812au-ct get hang or dead lock with latest code base openwrt

ptpt52, 2023-11-09:
### Describe the bug

```
[   76.724257] usb 1-4: new high-speed USB device number 4 using xhci_hcd
[   77.041102] 
[   77.041890] ============================================
[   77.043188] WARNING: possible recursive locking detected
[   77.044603] 5.15.137 #0 Not tainted
[   77.045765] --------------------------------------------
[   77.046920] kworker/1:3/381 is trying to acquire lock:
[   77.048031] ffff888116011358 (plock){+...}-{2:2}, at: rtw_alloc_macid+0x106/0x28d [rtl8812au]
[   77.049512] 
[   77.049512] but task is already holding lock:
[   77.051902] ffffc90002eb7ce8 (plock){+...}-{2:2}, at: rtw_alloc_stainfo+0x2d/0x1c5 [rtl8812au]
[   77.053267] 
[   77.053267] other info that might help us debug this:
[   77.055491]  Possible unsafe locking scenario:
[   77.055491] 
[   77.057106]        CPU0
[   77.057964]        ----
[   77.058905]   lock(plock);
[   77.059667]   lock(plock);
[   77.060687] 
[   77.060687]  *** DEADLOCK ***
[   77.060687] 
[   77.062793]  May be due to missing lock nesting notation
[   77.062793] 
[   77.064308] 6 locks held by kworker/1:3/381:
[   77.065104]  #0: ffff888101c79b38 ((wq_completion)usb_hub_wq){+.+.}-{0:0}, at: process_one_work+0x1f8/0x530
[   77.066212]  #1: ffffc90000cf7e68 ((work_completion)(&hub->events)){+.+.}-{0:0}, at: process_one_work+0x1f8/0x530
[   77.067376]  #2: ffff888103d62220 (&dev->mutex){....}-{3:3}, at: hub_event+0x5a/0x18b0
[   77.068395]  #3: ffff888115b35220 (&dev->mutex){....}-{3:3}, at: __device_attach+0x26/0x1e0
[   77.069428]  #4: ffff8881158f09a8 (&dev->mutex){....}-{3:3}, at: __device_attach+0x26/0x1e0
[   77.070473]  #5: ffffc90002eb7ce8 (plock){+...}-{2:2}, at: rtw_alloc_stainfo+0x2d/0x1c5 [rtl8812au]
[   77.071578] 
[   77.071578] stack backtrace:
[   77.072984] CPU: 1 PID: 381 Comm: kworker/1:3 Not tainted 5.15.137 #0
[   77.073905] Hardware name: LENOVO 80U9/Lenovo XiaoXin 5, BIOS 0VCN22WW(V1.06) 09/26/2016
[   77.074906] Workqueue: usb_hub_wq hub_event
[   77.075710] Call Trace:
[   77.076389]  <TASK>
[   77.077040]  dump_stack_lvl+0x61/0x7e
[   77.077775]  dump_stack+0x10/0x16
[   77.078469]  __lock_acquire.cold+0xc3/0x29b
[   77.079212]  lock_acquire+0xbd/0x270
[   77.079894]  ? rtw_alloc_macid+0x106/0x28d [rtl8812au]
[   77.080702]  ? module_assert_mutex_or_preempt+0x13/0x40
[   77.081476]  ? __module_address+0x27/0x110
[   77.082206]  _raw_spin_lock_bh+0x31/0x50
[   77.082894]  ? rtw_alloc_macid+0x106/0x28d [rtl8812au]
[   77.083675]  rtw_alloc_macid+0x106/0x28d [rtl8812au]
[   77.084410]  rtw_alloc_stainfo+0x195/0x1c5 [rtl8812au]
[   77.085173]  rtw_init_bcmc_stainfo+0x25/0x30 [rtl8812au]
[   77.085984]  rtw_init_drv_sw+0x140/0x179 [rtl8812au]
[   77.086741]  rtw_usb_if1_init+0x169/0x204 [rtl8812au]
[   77.087498]  rtw_drv_init+0x227/0x2c1 [rtl8812au]
[   77.088214]  usb_probe_interface+0x101/0x2b0
[   77.088893]  really_probe+0x1c8/0x3f0
[   77.089517]  __driver_probe_device+0x80/0x130
[   77.090168]  driver_probe_device+0x1f/0x90
[   77.090833]  __device_attach_driver+0x93/0x100
[   77.091479]  ? driver_allows_async_probing+0x60/0x60
[   77.092149]  bus_for_each_drv+0x6f/0xa0
[   77.092759]  __device_attach+0xb3/0x1e0
[   77.093366]  device_initial_probe+0xe/0x20
[   77.094004]  bus_probe_device+0x9f/0xb0
[   77.094613]  device_add+0x3bd/0x930
[   77.095207]  usb_set_configuration+0x5e7/0x860
[   77.095875]  usb_generic_driver_probe+0x50/0x70
[   77.096535]  usb_probe_device+0x42/0x110
[   77.097182]  really_probe+0x1c8/0x3f0
[   77.097813]  __driver_probe_device+0x80/0x130
[   77.098471]  driver_probe_device+0x1f/0x90
[   77.099112]  __device_attach_driver+0x93/0x100
[   77.099770]  ? driver_allows_async_probing+0x60/0x60
[   77.100474]  bus_for_each_drv+0x6f/0xa0
[   77.101091]  __device_attach+0xb3/0x1e0
[   77.101709]  device_initial_probe+0xe/0x20
[   77.102320]  bus_probe_device+0x9f/0xb0
[   77.102925]  device_add+0x3bd/0x930
[   77.103522]  ? _raw_spin_unlock_irqrestore+0x40/0x50
[   77.104174]  usb_new_device+0x1cc/0x3f0
[   77.104755]  hub_event+0xd96/0x18b0
[   77.105347]  ? trace_hardirqs_on+0x3b/0xe0
[   77.105942]  process_one_work+0x284/0x530
[   77.106525]  worker_thread+0x50/0x400
[   77.107077]  ? process_one_work+0x530/0x530
[   77.107673]  kthread+0x147/0x170
[   77.108206]  ? set_kthread_struct+0x50/0x50
[   77.108792]  ret_from_fork+0x22/0x30
[   77.109348]  </TASK>
[   77.109935] RTL871X: rtw_ndev_init(wlan0)
[  136.525433] warning: catch unregister event for dev=wlan0
```

### OpenWrt version

xxx

### OpenWrt target/subtarget

xxx

### Device

xx

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-11-09:
ping @greearb @hauke @nbd168

ptpt52, 2023-11-09:
more log:
```
[  366.969506] INFO: task ucode:13862 blocked for more than 123 seconds.
[  366.973243]       Not tainted 5.15.137 #0
[  366.976754] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  366.980670] task:ucode           state:D stack:    0 pid:13862 ppid: 13834 flags:0x00004000
[  366.984690] Call Trace:
[  366.988363]  <TASK>
[  366.991844]  __schedule+0x2a1/0x850
[  366.995380]  schedule+0x6d/0xc0
[  366.998903]  schedule_preempt_disabled+0x10/0x20
[  367.002527]  __mutex_lock+0x3d3/0x760
[  367.005921]  ? lock_acquire+0x1d2/0x270
[  367.009353]  ? cfg80211_netdev_notifier_call+0x1ed/0x540 [cfg80211]
[  367.013054]  ? desc_read_finalized_seq+0x20/0x70
[  367.016517]  mutex_lock_nested+0x16/0x20
[  367.020038]  ? mutex_lock_nested+0x16/0x20
[  367.023486]  cfg80211_netdev_notifier_call+0x1ed/0x540 [cfg80211]
[  367.027327]  ? lock_acquire+0x1d2/0x270
[  367.030761]  ? prb_read_valid+0x17/0x20
[  367.034166]  ? console_unlock+0x501/0x560
[  367.037643]  ? lock_acquire+0x1d2/0x270
[  367.041061]  ? lock_acquire+0x1d2/0x270
[  367.044556]  ? __mutex_lock+0x8d/0x760
[  367.047979]  ? nft_offload_netdev_event+0x3d/0xf0 [nf_tables]
[  367.051567]  ? lock_release+0x162/0x260
[  367.054996]  ? nft_offload_netdev_event+0x64/0xf0 [nf_tables]
[  367.058708]  ? nft_offload_netdev_event+0x3d/0xf0 [nf_tables]
[  367.062390]  ? __mutex_unlock_slowpath+0x2d/0x270
[  367.065880]  ? mutex_unlock+0xd/0x20
[  367.069363]  raw_notifier_call_chain+0x44/0x60
[  367.073092]  call_netdevice_notifiers_info+0x4b/0x90
[  367.076671]  unregister_netdevice_many+0x399/0x750
[  367.080358]  unregister_netdevice_queue+0x6d/0xb0
[  367.083919]  cfg80211_rtw_del_virtual_intf+0x24/0x5a [rtl8812au]
[  367.087982]  cfg80211_remove_virtual_intf+0x4d/0x130 [cfg80211]
[  367.091971]  nl80211_del_interface+0x54/0x80 [cfg80211]
[  367.095873]  genl_family_rcv_msg_doit+0xc9/0x110
[  367.099623]  genl_rcv_msg+0xcc/0x1c0
[  367.103222]  ? nl80211_stop_ap+0x40/0x40 [cfg80211]
[  367.107181]  ? genl_get_cmd+0xe0/0xe0
[  367.110817]  netlink_rcv_skb+0x52/0x100
[  367.112781]  genl_rcv+0x23/0x40
[  367.113710]  netlink_unicast+0x174/0x240
[  367.114655]  netlink_sendmsg+0x22d/0x470
[  367.115579]  ? netlink_unicast+0x240/0x240
[  367.116406]  ____sys_sendmsg+0x235/0x260
[  367.117235]  ? __import_iovec+0x4b/0x190
[  367.118037]  ___sys_sendmsg+0xb3/0xf0
[  367.118813]  ? __slab_free+0x2d8/0x440
[  367.119602]  ? trace_hardirqs_on+0x3b/0xe0
[  367.120385]  ? lock_acquire+0x1d2/0x270
[  367.121138]  ? lock_acquire+0x1d2/0x270
[  367.121892]  ? mntput_no_expire+0x71/0x4c0
[  367.122632]  ? call_rcu+0x13f/0x2b0
[  367.123357]  ? __fput+0x131/0x280
[  367.124062]  ? trace_hardirqs_on+0x3b/0xe0
[  367.124764]  ? __alloc_file+0xf0/0xf0
[  367.125406]  ? call_rcu+0x13f/0x2b0
[  367.126036]  __sys_sendmsg+0x40/0x90
[  367.126674]  ? bts_event_start+0x20/0xd0
[  367.127349]  __x64_sys_sendmsg+0x18/0x20
[  367.128081]  do_syscall_64+0x42/0x90
[  367.128780]  entry_SYSCALL_64_after_hwframe+0x62/0xcc
[  367.129513] RIP: 0033:0x7f383e160467
[  367.130160] RSP: 002b:00007ffcaa5814d8 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
[  367.131003] RAX: ffffffffffffffda RBX: 000000000000002e RCX: 00007f383e160467
[  367.131852] RDX: 0000000000000000 RSI: 00007ffcaa581528 RDI: 000000000000000a
[  367.132688] RBP: 00007f383e181b48 R08: 0000000000000000 R09: 0000000000000000
[  367.133537] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000000
[  367.134319] R13: 000000000000001d R14: 0000000000000000 R15: 00007f383e09c080
[  367.135126]  </TASK>
[  367.135655] INFO: task iwinfo:14126 blocked for more than 123 seconds.
[  367.136426]       Not tainted 5.15.137 #0
[  367.137056] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  367.137916] task:iwinfo          state:D stack:    0 pid:14126 ppid: 12504 flags:0x00000000
[  367.138796] Call Trace:
[  367.139359]  <TASK>

```

ptpt52, 2023-11-09:
build with kernel debug options:
```
+CONFIG_KERNEL_PROVE_LOCKING=y
+CONFIG_KERNEL_SOFTLOCKUP_DETECTOR=y
+CONFIG_KERNEL_HARDLOCKUP_DETECTOR=y
+CONFIG_KERNEL_DETECT_HUNG_TASK=y
+CONFIG_KERNEL_WQ_WATCHDOG=y
+CONFIG_KERNEL_DEBUG_ATOMIC_SLEEP=y
```

greearb, 2023-11-09:
Hello,

I am no longer using rtl8812au, so I cannot help you with this.

I am curious what backports version you are using though, 6.7 kernel has huge locking changes
and probably some regressions...

Thanks,
Ben