openwrt/openwrt #13922: 23.05.0 WD My Book Live Duo has no static address set by default

Emedequi, 2023-11-09:
### Describe the bug

23.05.0 (for WD My Book Live Duo / apm821xx/sata / factory .img / powerpc_464fp) has no static IP address set by default (192.168.1.1). The NAS becomes unreachable (under Linux and Windows 10). However, the leds show the booting process and, in the end, a solid green light.

Once, I was able to find the NAS (which is directly attached to the computer) with Angry IP Scanner and set a static IP address (192.168.1.1) in LuCI. It didn't work. Later, during a new setup, I modified the network configuration file (etc/conf/network), as shown below, without success.

Last working version with static IP is 17.09.10. A sysupgrade from 17.09.10 to 23.05.0 changes the original static IP address to DHCP, as well, and therefore the problem persists.

### OpenWrt version

23.05.0-r23497-6637af95aa

### OpenWrt target/subtarget

apm821xx/sata

### Device

WD My Book Live Duo

### Image kind

Official downloaded image

### Steps to reproduce

Clean install. NAS directly connected to the computer. No router. Changing SDDs' order (A/B) or booting with only one disk makes no difference.

# 23.05.0 (etc/conf/network) [modified]

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd9f:fb21:89e8::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'

config interface 'lan6'
        option device 'br-lan'
        option proto 'dhcpv6'

# 23.05.0 (etc/conf/network) [original]

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd9f:fb21:89e8::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'

config interface 'lan'
	option device 'br-lan'
	option proto 'dhcp'

config interface 'lan6'
	option device 'br-lan'
	option proto 'dhcpv6'

### Actual behaviour

The NAS boots. No static IP address (192.168.1.1).

### Additional info

All tests done with the same SSDs (KingDian 2TB). Linux Mint 21,2 and Windows 10. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-09:
Not reflected in device page but yes, since v22....
https://github.com/openwrt/openwrt/commit/abb4083fa06841ef5b094aee24a3b1f58403516b

Emedequi, 2023-11-09:
Is it still possible to set a static IP by editing a configuration file? Many thanks.

brada4, 2023-11-09:
It is possible to spin up dhcp server somewhere on the network (like linux you mentioned) and configure from freshly issued IP address... (only that documentation wiki does not state that)

brada4, 2023-11-10:
Please add 'wiki' to title. Rhis has to be documented for all affected nas devices. Not trivial to figure out, though practical bc NA in NAS and not being routers in normal sense

raenye, 2023-11-10:
Besides updating the wiki to document the current behaviour, it makes sense also to say how to force static IP for new installations via `uci-default`  (without a DHCP server).

brada4, 2023-11-10:
Or failsafe mode that boots on 1.1 besides normal conf getting DHCP address for instant integration.

Emedequi, 2023-11-10:
Do you mean in the title? Keep in mind, I'm no expert. Besides, I'm new on GitHub. Tell me and I rewrite the post.

raenye, 2023-11-10:
@Emedequi you don't have wiki write permissions, so someone else would have to do it. Everything is documented in this issue.

Emedequi, 2023-11-10:
Thanks raenye for clearing that up.