openwrt/openwrt #13927: Issues with user/group IDs 65536 and up

tew42, 2023-11-10:
### Describe the bug

User/group IDs of 65536 and greater have significant issues in usage, but are implemented as default in package installation when new users are created.

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

ath79, x86_64

### Device

Archer A7 v5, Proxmox LXC

### Image kind

Official downloaded image

### Steps to reproduce

Create user with UID/GID of 65536 or greater. _Note that this is the default case for package installations that specify username:groupname without specific IDs as per `group_add_next()`, called by `default_postinst()`, in `functions.sh`_

Attempt to set permissions on any file with `chown username filename`

### Actual behaviour

`chown: filename: Invalid argument`

```
ls -l
-rw-r--r--    1 root     root             4 Nov 10 11:02 filename
```

### Expected behaviour

```
ls -l
-rw-r--r--    1 username     root             4 Nov 10 11:02 filename
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

tew42, 2023-11-10:
Thanks - indeed it works (or doesn't) independent of packages, I just wanted to highlight that it becomes an actual issue due to the package installer defaults.

Anyhow:
```
BusyBox v1.36.1 (2023-10-09 21:45:35 UTC) multi-call binary.
```
and, unsurprisingly / in line with the original report (with "tang" being just one of the generic examples of out-of-bound uids already present)
```
root@OpenWrt-PVE:~# chown 99999 `mktemp -d`
chown: /tmp/tmp.PaHaFI: Invalid argument
root@OpenWrt-PVE:~# chown 65536 `mktemp -d`
chown: /tmp/tmp.NegDnN: Invalid argument
root@OpenWrt-PVE:~# id tang
uid=65536(tang) gid=65536(tang) groups=65536(tang),65536(tang)
```

tew42, 2023-11-10:
update: I just spun up a snapshot LXC and a 22.03 LXC, same (faulty) behavior

tew42, 2023-11-10:
Alright, retested with one more architecture now - which works fine?!

So, the chown mktemp only gives insight when there's an error message (LXC case). But in ath79 case, it's insufficient...

on x64 LXC:
```
root@OpenWrt-PVE:/var/run# chown 99999 `mktemp -d`
chown: /tmp/tmp.ckjMfP: Invalid argument
root@OpenWrt-PVE:/var/run# chown 9999 `mktemp -d`
root@OpenWrt-PVE:/var/run#
```

on ath79:
```
root@OpenWrt-ArcherA7:~# chown 99999 `mktemp -d`
root@OpenWrt-ArcherA7:~# mkdir test
root@OpenWrt-ArcherA7:~# chown testuser test
root@OpenWrt-ArcherA7:~# ls -l
drwxr-xr-x    2 root     root             0 Nov 10 16:57 test
root@OpenWrt-ArcherA7:~# id testuser
uid=65536(testuser) gid=65536(testuser) groups=65536(testuser),65536(testuser)
```

ramips/mt7621
```
root@OpenWrt-DAPx1860:~# chown 99999 `mktemp -d`
root@OpenWrt-DAPx1860:~# mkdir test
root@OpenWrt-DAPx1860:~# chown testuser test
root@OpenWrt-DAPx1860:~# ls -l
drwxr-xr-x    2 testuser root           160 Nov 10 16:57 test
root@OpenWrt-DAPx1860:~# id testuser
uid=65536(testuser) gid=65536(testuser) groups=65536(testuser),65536(testuser)
```

brada4, 2023-11-10:
It is lxc configuration issue. Just map few uid-s above 64k-1 to whatever and your containers will run well.

tew42, 2023-11-10:
On LXC, sure, and that's not a typical use case anyway. But ath79 is the real issue here as I see it

brada4, 2023-11-10:
hmm, looks like jffs overlay on squashfs then, no problem in tmpfs. but ramips is more or less same cpu as ath79.

brada4, 2023-11-10:
Userland chmod does OK, just that filesystem combination does not work out.

is there a difference in `mount` in places containing overlays (modern ramips with lots of flash and ram):
```
mount | grep -e /rom -e /overlay
/dev/root on /rom type squashfs (ro,relatime,errors=continue)
/dev/ubi0_1 on /overlay type ubifs (rw,noatime,assert=read-only,ubi=0,vol=1)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
```

tew42, 2023-11-10:
not familiar enough with this to understand it, but there does seem to be a distinction between platforms on that front:

ath79
```
root@OpenWrt-ArcherA7:~# mount | grep -e /rom -e /overlay
/dev/root on /rom type squashfs (ro,relatime,errors=continue)
/dev/mtdblock5 on /overlay type jffs2 (rw,noatime)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
```

ramips
```
root@OpenWrt-DAPx1860:~# mount | grep -e /rom -e /overlay
/dev/root on /rom type squashfs (ro,relatime,errors=continue)
/dev/ubi0_1 on /overlay type ubifs (rw,noatime,assert=read-only,ubi=0,vol=1)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)

```

brada4, 2023-11-10:
jffs2 supports only 16bit uid/gid :-( probably reason to bring generated upper users half way down....

tew42, 2023-11-10:
So potentially like this? #13937

brada4, 2023-11-10:
I cant grep other 64k on archer c7 ;-) looks like it.

commit 140b48a9 (2024-05-10): base-files: fix uid/gid auto-enumeration to avoid 16-bit limit

uid/gid range should be limited to 16bit unsigned integer range to
avoid "wraparound" issues with permissions where jffs2
is employed for storage and chown 65536 (first auto-created user)
becomes equivalent to chown 0

Fixes: #13927
commit ccac75a3 (2024-07-08): base-files: fix uid/gid auto-enumeration to avoid 16-bit limit

uid/gid range should be limited to 16bit unsigned integer range to
avoid "wraparound" issues with permissions where jffs2
is employed for storage and chown 65536 (first auto-created user)
becomes equivalent to chown 0

Fixes: #13927

(cherry picked from commit 140b48a9e93909873bee777ca5bdf49fe4f01271)
Link: https://github.com/openwrt/openwrt/pull/15898
closed 2024-05-10 by openwrt-bot