openwrt/openwrt #13934: 23.05.0 bricks Easybox 803A

bodop, 2023-11-10:
### Describe the bug

sysupgrade from 22.03.3 leads to the following output on the serial console

```ROM VER: 1.0.3
CFG 01
Read EEPROMX
 X

U-Boot 2013.10-openwrt4 (May 28 2014 - 21:46:54) ARV752DPW22

Board: Arcadyan ARV752DPW22
SoC:   Lantiq Danube-S v1.5
CPU:   333.333 MHz
IO:    166.667 MHz
BUS:   83.333 MHz
BOOT:  NOR
DRAM:  64 MiB
Flash: 8 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   ltq-eth
Hit any key to stop autoboot:  0
## Booting kernel from Legacy Image at b0040000 ...
   Image Name:   MIPS OpenWrt Linux-5.15.134
   Created:      2023-10-09  21:45:35 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2726016 Bytes = 2.6 MiB
   Load Address: 80002000
   Entry Point:  80002000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... LZMA: uncompress or overwrite error 7 - must RESET b
ROM VER: 1.0.3
.
.
.
```

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

lantiq/xway

### Device

Arcor 803

### Image kind

Official downloaded image

### Steps to reproduce

sysupgrade -v openwrt-23.05.0-lantiq-xway-arcadyan_arv752dpw22-squashfs-sysupgrade.bin

### Actual behaviour

Device to not boot into 23.05.0

### Expected behaviour

Device should boot into 23.05.0

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-10:
Seems bootloader is defective due to multiple devices with same issues.
https://github.com/openwrt/openwrt/issues/13891

bodop, 2023-11-11:
Current bootloader fixed the uncompression error. The [device page](https://openwrt.org/toh/astoria/arv752dpw22#u-boot) should be updated, it recommends to use a very old u-boot version.

Like in #13891 the boot process stops a few lines later

```   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.15.134 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23497-6637af95aa) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Oct 9 21:45:35 2023
[    0.000000] SoC: Danube rev 1.5
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019641 (MIPS 24KEc)
[    0.000000] MIPS: machine is Arcor 803
[    0.000000] Initrd not found or empty - disabling initrd
```

brada4, 2023-11-11:
cpuid should follow after that line...
from device page regarding unrelated hang - disconnect power for more than a minute and try again...

hauke, 2023-11-12:
Please try to modify the `loadaddr`in u-boot. It is probably set to `0xb0040000, try setting it to `0xb0060000`.

The boot loader loads the kernel to `loadaddr` and then the kernel loader uncompresses the kernel to `80002000`. If the kernel is too big it will overwite itself while uncompressing.

bodop, 2023-11-14:
The first error message came from a flashed kernel: Uncompressing from b0040000 to 0x80020000. I am no bootloader expert, but for me this sound safe.

u-boot's default loadaddr is 81000000 I retried again with 23.05.1 and bootm:

```
ARV752DPW22 # tftpboot openwrt-23.05.1-lantiq-xway-arcadyan_arv752dpw22-initramfs-kernel.bin
Using ltq-eth device
TFTP from server 192.168.1.2; our IP address is 192.168.1.1
Filename 'openwrt-23.05.1-lantiq-xway-arcadyan_arv752dpw22-initramfs-kernel.bin'.
Load address: 0x81000000
Loading: #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #############################
done
Bytes transferred = 5196310 (4f4a16 hex)
ARV752DPW22 # bootm
## Booting kernel from Legacy Image at 81000000 ...
   Image Name:   MIPS OpenWrt Linux-5.15.137
   Created:      2023-11-12  15:19:39 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    5196246 Bytes = 5 MiB
   Load Address: 80002000
   Entry Point:  80002000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.15.137 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23619-101988c61a) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sun Nov 12 15:19:39 2023
[    0.000000] SoC: Danube rev 1.5
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019641 (MIPS 24KEc)
[    0.000000] MIPS: machine is Arcor 803
[    0.000000] Initrd not found or empty - disabling initrd
```

bodop, 2023-12-31:
[This hint](https://github.com/openwrt/openwrt/issues/14283#issuecomment-1872572959) makes my device boot again with the current snapshot. (OK, the WLAN chip is not recognized, but this seems to be different topic).

commit 64d8c277 (2024-01-16): Revert "lantiq: xway: kernel: enable SMP support"

Enabling SMP on the xway target results in two issues:

* some danube chipset-based devices fail on boot,
* on devices based on the arx100 chipset, enabling smp
results in a degradation of NAT performance.

After these two issues are fixed, SMP can be re-enabled.

This reverts commit 084c20f6c54180f2dc6b3efc1c5ba9b62afbaf26.

Fixes: #13934
Fixes: #14283
commit 65fb97b4 (2024-01-22): lantiq: fix boot isues on danube boards

Enabling SMP on Danube[1] is incompatible with a patch that
adds support for interrupt handling on all cores on other
platforms[2]. This patch fixes the mentioned issue.

1. 084c20f6c54180f2dc6b3efc1c5ba9b62afbaf26 ("lantiq: xway: kernel: enable SMP support ")
2. fbd33d61648ae8982fbada7ad3b6d8222b367ab5 ("lantiq: enable interrupts on second VPEs")

Fixes: #13934
Fixes: #14283
PR #14388 (2024-01-22): lantiq: fix boot isues on danube boards
Enabling SMP on Danube[1] is incompatible with a patch that adds support for interrupt handling on all cores on other  platforms[2]. This patch fixes the mentioned issue.

1. 084c20f6c54180f2dc6b3efc1c5ba9b62afbaf26 ("lantiq: xway: kernel: enable SMP support ")
2. fbd33d61648ae8982fbada7ad3b6d8222b367ab5 ("lantiq: enable interrupts on second VPEs")

Fixes: #13934
Fixes: #14283
closed 2024-01-22 by openwrt-bot