openwrt/openwrt #13936: odhcpd doesn't relay a second ipv6 address from same device(mac address?)

thomasschaeferm, 2023-11-10:
### Describe the bug

in relay mode odhcpd doesn't add all necessary ndp proxies:
[logread-clat.txt](https://github.com/openwrt/openwrt/files/13323135/logread-clat.txt)
`
grep  '4e6b\|6378' logread-clat.txt 
Fri Nov 10 20:20:24 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:20:24 2023 daemon.debug odhcpd[9298]: Added proxy neighbour entry 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on wan6
Fri Nov 10 20:20:24 2023 daemon.debug odhcpd[9298]: Learning about proxy routing for 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:20:24 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:20:24 2023 daemon.debug odhcpd[9298]: Added proxy neighbour entry 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on wan6
Fri Nov 10 20:20:24 2023 daemon.debug odhcpd[9298]: Learning about proxy routing for 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:20:24 2023 daemon.debug odhcpd[9298]: Got a NS for 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on wan6
Fri Nov 10 20:20:24 2023 daemon.debug odhcpd[9298]: Pinging for 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:20:24 2023 daemon.debug odhcpd[9298]: Sent 8 bytes to 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b%lan@br-lan
Fri Nov 10 20:20:53 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:20:53 2023 daemon.debug odhcpd[9298]: Added proxy neighbour entry 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on wan6
Fri Nov 10 20:20:53 2023 daemon.debug odhcpd[9298]: Learning about proxy routing for 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:21:06 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:21:06 2023 daemon.debug odhcpd[9298]: Added proxy neighbour entry 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on wan6
Fri Nov 10 20:21:06 2023 daemon.debug odhcpd[9298]: Learning about proxy routing for 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:21:06 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:21:06 2023 daemon.debug odhcpd[9298]: Added proxy neighbour entry 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on wan6
Fri Nov 10 20:21:06 2023 daemon.debug odhcpd[9298]: Learning about proxy routing for 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:21:06 2023 daemon.debug odhcpd[9298]: Got a NS for 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on wan6
Fri Nov 10 20:21:06 2023 daemon.debug odhcpd[9298]: Pinging for 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:21:06 2023 daemon.debug odhcpd[9298]: Sent 8 bytes to 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b%lan@br-lan
Fri Nov 10 20:21:34 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:21:34 2023 daemon.debug odhcpd[9298]: Added proxy neighbour entry 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on wan6
Fri Nov 10 20:21:34 2023 daemon.debug odhcpd[9298]: Learning about proxy routing for 2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b on lan
Fri Nov 10 20:22:55 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:9c64:9735:6378:0 on wan
Fri Nov 10 20:22:55 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:9c64:9735:6378:0 on wan6
Fri Nov 10 20:22:59 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:9c64:9735:6378:0 on wan
Fri Nov 10 20:22:59 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:9c64:9735:6378:0 on wan6
Fri Nov 10 20:23:03 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:9c64:9735:6378:0 on wan
Fri Nov 10 20:23:03 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:9c64:9735:6378:0 on wan6
Fri Nov 10 20:23:07 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:9c64:9735:6378:0 on wan
Fri Nov 10 20:23:07 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:9c64:9735:6378:0 on wan6
Fri Nov 10 20:23:11 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:9c64:9735:6378:0 on wan
Fri Nov 10 20:23:11 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:9c64:9735:6378:0 on wan6
Fri Nov 10 20:23:16 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:9c64:9735:6378:0 on wan
Fri Nov 10 20:23:16 2023 daemon.debug odhcpd[9298]: Netlink newneigh 2001:a61:4b1:8502:9c64:9735:6378:0 on wan6
`
as you see, the first address is recognized and added for proxying and routing
the second address is just recognized on the wrong interface (despite it is a lan address) and not added for proxying and routing 

The effect is, it breaks clat on linux and android devices, since the usually use a second IPv6 address for clat.
I use the relay mode together with jool as NAT64-Gateay (PLAT part of 464xlat). During ping ipv4 destinations the packet goes out, comes back and drops with openwrt because of missing last ndp-proxy and route to the host.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ipq40xx/generic

### Device

GL.iNet GL-A1300

### Image kind

Official downloaded image

### Steps to reproduce

configure [jool](https://openwrt.org/docs/guide-user/network/ipv6/nat64)
configure ipv6 relay mode for ipv6

### Actual behaviour

the first address of a device works as expected (routing ipv6 works, nat64 works)
the second address of a device doesn't work (routing and nat64 works only one way) 

### Expected behaviour

both address of a host should be routed and translated (NAT64) correctly through ndp proxying and the (host) routes

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-11:
Whats in `ip -6 ro sh`

thomasschaeferm, 2023-11-11:
`root@OpenWrt:~# /bin/busybox ip  -6 ro sh
default from 2001:a61:4b1:8502::/64 via fe80::9a9b:cbff:fe05:4242 dev wan  metric 512 
default from fd00:0:0:1::/64 via fe80::9a9b:cbff:fe05:4242 dev wan  metric 512 
2001:a61:4b1:8500::/56 from 2001:a61:4b1:8502::/64 via fe80::9a9b:cbff:fe05:4242 dev wan  metric 512 
2001:a61:4b1:8500::/56 from fd00:0:0:1::/64 via fe80::9a9b:cbff:fe05:4242 dev wan  metric 512 
2001:a61:4b1:8502:204e:1ecc:d0e4:fe11 dev br-lan  metric 1024 
2001:a61:4b1:8502:35d2:ab2b:fb2f:2f0 dev br-lan  metric 1024 
2001:a61:4b1:8502:3fc6:9c50:9588:c2ef dev br-lan  metric 1024 
2001:a61:4b1:8502:5b13:7770:74fd:d702 dev br-lan  metric 1024 
2001:a61:4b1:8502:ca01:3ef3:84b4:3d26 dev br-lan  metric 1024 
2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b dev br-lan  metric 1024 
2001:a61:4b1:8502::/64 dev wan  metric 256 
unreachable 2001:a61:4b1:8502::/64 dev lo  metric 2147483647 
fd00:0:0:1::/64 dev wan  metric 256 
fd00:0:0:1::/64 via fe80::9a9b:cbff:fe05:4242 dev wan  metric 512 
unreachable fd00:0:0:1::/64 dev lo  metric 2147483647 
fd8b:779d:6bbe::/64 dev br-lan  metric 1024 
unreachable fd8b:779d:6bbe::/48 dev lo  metric 2147483647 
fe80::/64 dev eth0  metric 256 
fe80::/64 dev wan  metric 256 
fe80::/64 dev phy1-ap0  metric 256 
fe80::/64 dev br-lan  metric 256 
anycast 2001:a61:4b1:8502:: dev wan  metric 0 
anycast fd00:0:0:1:: dev wan  metric 0 
anycast fd8b:779d:6bbe:: dev br-lan  metric 0 
anycast fe80:: dev eth0  metric 0 
anycast fe80:: dev wan  metric 0 
anycast fe80:: dev phy1-ap0  metric 0 
anycast fe80:: dev br-lan  metric 0 
multicast ff00::/8 dev eth0  metric 256 
multicast ff00::/8 dev br-lan  metric 256 
multicast ff00::/8 dev wan  metric 256 
multicast ff00::/8 dev phy1-ap0  metric 256 `

ip from ip package (iproute2)
`ip -6 r s 
default from 2001:a61:4b1:8502::/64 via fe80::9a9b:cbff:fe05:4242 dev wan proto static metric 512 pref medium
default from fd00:0:0:1::/64 via fe80::9a9b:cbff:fe05:4242 dev wan proto static metric 512 pref medium
2001:a61:4b1:8500::/56 from 2001:a61:4b1:8502::/64 via fe80::9a9b:cbff:fe05:4242 dev wan proto static metric 512 pref medium
2001:a61:4b1:8500::/56 from fd00:0:0:1::/64 via fe80::9a9b:cbff:fe05:4242 dev wan proto static metric 512 pref medium
2001:a61:4b1:8502:204e:1ecc:d0e4:fe11 dev br-lan proto static metric 1024 pref medium
2001:a61:4b1:8502:35d2:ab2b:fb2f:2f0 dev br-lan proto static metric 1024 pref medium
2001:a61:4b1:8502:3fc6:9c50:9588:c2ef dev br-lan proto static metric 1024 pref medium
2001:a61:4b1:8502:5b13:7770:74fd:d702 dev br-lan proto static metric 1024 pref medium
2001:a61:4b1:8502:ca01:3ef3:84b4:3d26 dev br-lan proto static metric 1024 pref medium
2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b dev br-lan proto static metric 1024 pref medium
2001:a61:4b1:8502::/64 dev wan proto static metric 256 pref medium
unreachable 2001:a61:4b1:8502::/64 dev lo proto static metric 2147483647 pref medium
fd00:0:0:1::/64 dev wan proto static metric 256 pref medium
fd00:0:0:1::/64 via fe80::9a9b:cbff:fe05:4242 dev wan proto static metric 512 pref medium
unreachable fd00:0:0:1::/64 dev lo proto static metric 2147483647 pref medium
fd8b:779d:6bbe::/64 dev br-lan proto static metric 1024 pref medium
unreachable fd8b:779d:6bbe::/48 dev lo proto static metric 2147483647 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev wan proto kernel metric 256 pref medium
fe80::/64 dev phy1-ap0 proto kernel metric 256 pref medium
fe80::/64 dev br-lan proto kernel metric 256 pref medium
`

`ip n show proxy
2001:a61:4b1:8502:3fc6:9c50:9588:c2ef dev wan proxy 
2001:a61:4b1:8502:35d2:ab2b:fb2f:2f0 dev wan proxy 
2001:a61:4b1:8502:204e:1ecc:d0e4:fe11 dev wan proxy 
2001:a61:4b1:8502:5b13:7770:74fd:d702 dev wan proxy 
2001:a61:4b1:8502:ca01:3ef3:84b4:3d26 dev wan proxy 
fd00::1:9683:c4ff:fe33:a81e dev br-lan proxy 
2001:a61:4b1:8502:9683:c4ff:fe33:a81e dev br-lan proxy 
2001:a61:4b1:8502:ee7e:3abd:a49f:4e6b dev wan proxy 
root@OpenWrt:~# 
`

thomasschaeferm, 2023-11-11:
tcpdump -i wan 
during pinging 8.8.8.8 from host via clat

`21:55:03.978572 IP 192.168.179.4 > dns.google: ICMP echo request, id 63861, seq 65, length 64
21:55:03.984395 IP dns.google > 192.168.179.4: ICMP echo reply, id 63861, seq 65, length 64
21:55:03.985250 IP6 fe80::9683:c4ff:fe33:a81e > ff02::1:ff78:0: ICMP6, neighbor solicitation, who has 2001:a61:4b1:8502:2cbc:d477:6378:0, length 32
21:55:04.992036 IP 192.168.179.4 > dns.google: ICMP echo request, id 63861, seq 66, length 64
21:55:04.996251 IP dns.google > 192.168.179.4: ICMP echo reply, id 63861, seq 66, length 64
21:55:05.020166 IP6 fe80::9683:c4ff:fe33:a81e > ff02::1:ff78:0: ICMP6, neighbor solicitation, who has 2001:a61:4b1:8502:2cbc:d477:6378:0, length 32
21:55:06.005517 IP 192.168.179.4 > dns.google: ICMP echo request, id 63861, seq 67, length 64
21:55:06.011375 IP dns.google > 192.168.179.4: ICMP echo reply, id 63861, seq 67, length 64
21:55:06.060856 IP6 fe80::9683:c4ff:fe33:a81e > ff02::1:ff78:0: ICMP6, neighbor solicitation, who has 2001:a61:4b1:8502:2cbc:d477:6378:0, length 32
21:55:07.018622 IP 192.168.179.4 > dns.google: ICMP echo request, id 63861, seq 68, length 64
21:55:07.023463 IP dns.google > 192.168.179.4: ICMP echo reply, id 63861, seq 68, length 64
21:55:08.032183 IP 192.168.179.4 > dns.google: ICMP echo request, id 63861, seq 69, length 64
21:55:08.037106 IP dns.google > 192.168.179.4: ICMP echo reply, id 63861, seq 69, length 64
21:55:08.037364 IP6 fe80::9683:c4ff:fe33:a81e > ff02::1:ff78:0: ICMP6, neighbor solicitation, who has 2001:a61:4b1:8502:2cbc:d477:6378:0, length 32
21:55:08.177067 IP6 fe80::9a9b:cbff:fe05:4242 > fe80::9683:c4ff:fe33:a81e: ICMP6, neighbor solicitation, who has fe80::9683:c4ff:fe33:a81e, length 32
21:55:08.177290 IP6 fe80::9683:c4ff:fe33:a81e > fe80::9a9b:cbff:fe05:4242: ICMP6, neighbor advertisement, tgt is fe80::9683:c4ff:fe33:a81e, length 24
21:55:09.045629 IP 192.168.179.4 > dns.google: ICMP echo request, id 63861, seq 70, length 64
21:55:09.049700 IP dns.google > 192.168.179.4: ICMP echo reply, id 63861, seq 70, length 64
21:55:09.100147 IP6 fe80::9683:c4ff:fe33:a81e > ff02::1:ff78:0: ICMP6, neighbor solicitation, who has 2001:a61:4b1:8502:2cbc:d477:6378:0, length 32
`
[br-lan.tcpdump.zip](https://github.com/openwrt/openwrt/files/13326603/br-lan.tcpdump.zip)

brada4, 2023-11-11:
the routable address on loopback also looks weird.

brada4, 2023-11-12:
Can you check on WAN if ping6 gets out and response arrives to router? I trust your interpretation.
It seems though that new proxied IPs are routed via wrong interface somehow.
Is there any problem having local dhcp/radv service as by default or you need to acquire individual IP6 addresses from provider in absence of delegations?

brada4, 2023-11-12:
Compare firewall rules made by jool (nft list ruleset) in both modes. Something gets re-ordered in an evil way.

thomasschaeferm, 2023-11-12:
'nft list ruleset'

is empty.

thomasschaeferm, 2023-11-12:
`ping 2600::` works from host as well from openwrt router.

Yes there is a reason for using relay mode. The "normal" mode with prefix delegation works fine. But I want to be flexible with that router. Most of my uplinks don't provide prefix delegation, they only provide SLAAC with/64. (all my mobile uplinks (Telekom, Vodafone, Telefonica, except Telekom when already running in IPv6only mode), Freifunk offloader, LAN at my working place)

brada4, 2023-11-12:
I mean router running jool obviously.

thomasschaeferm, 2023-11-13:
It was the openwrt router with jool. I am not sure if the firewall was disabled by me or by enabling relay mode.

brada4, 2023-11-13:
Maybe iptables-save then.. jool is x-tables module, it is supposed to inject encapsulated v4 from single v6 address in theory not touching the rest, similar to flowtable offload.

thomasschaeferm, 2023-11-13:
At the moment I am not at home for direct access to my openwrt router, but based on experience with jool on my raspberry pi - jool is not visible, not in nft, not in the old ip(6)tables. 
In the meantime I tried to understand, what ndp.c in relay modes does, but I am failed. Is somewhere a explanation of the code? How is the IPv6-traffic monitored, why is `ping` integrated? The problem with more ip addresses seems only to exist, if the second address of host is made via ndp-proxy too. A other address (e.g. made by privacy extensions) is no problem.

brada4, 2023-11-13:
Ping works for duplicate elimunation , it is part of spec.
hmm... does it work without jool in relay mode?

thomasschaeferm, 2023-11-13:
I will try to a test later on a natural ipv6-only uplink with openwrt in ipv6-relaymode - otherwise I can't test the special situation of clat (android/linux).
Could you tell me which spec you are referring to? 
((possibly wrong)duplicate elimination sounds dangerous to me - maybe ndp proxied addresses don't answer to pings)

brada4, 2023-11-13:
Since they are routed wrong they don get a chance to get an answer. 
Frankly no spec for ping, just that it is always sent and populates neighbour mapping (cf arp table)

thomasschaeferm, 2023-11-13:
I started with a similar device (GL-inet AR750) with a configuration reset. This time I used the nat64 gateway, that already exists in the uplink. IPv6-relay-mode works as expected for all IPs. After disabling dhcp4 also clatd worked (android as well the clatd/tayga combination on linux) 
So odhcpd is probably not the reason for my problems.  
Sorry for blaming.
Maybe I start the jool setup again from the beginning.

brada4, 2023-11-13:
Well, backup config to quickly undo jool.