openwrt/openwrt #13945: [ipq807x] Redmi AX6/Xiaomi AX3600 wan interface dhcp not working

XiaoliChan, 2023-11-11:
### Describe the bug

DHCP not working anymore with the latest build

The kernel log can't see any exception
[image]

### OpenWrt version

Main branch

### OpenWrt target/subtarget

IPQ807x

### Device

Redmi AX6/xiaomi ax3600

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-11-11:
Please add the exact snapshot version

XiaoliChan, 2023-11-11:
> Please add the exact snapshot version

`SNAPSHOT r24364-79ee0d2cee`

XiaoliChan, 2023-11-11:
I have three of this router, two routers have no issues with the wan interface(added into br-lan), and one router lost network connection after sysupgrade(even I reflash the firmware)

brada4, 2023-11-11:
Is the uplink cable 100Mbps? Please show `ip link` and ethtool lan@eth0?
Is it migration or restoring config back? From which version?

XiaoliChan, 2023-11-11:
> Is the uplink cable 100Mbps? 

Yes, only 100Mbps here

XiaoliChan, 2023-11-11:
> Please show `ip link` and ethtool lan@eth0? 

- `ip link`
[image]

- `ethtool`
[image]
[image]

> Is it migration or restoring config back? From which version?

Even I did clean install from sysupgrade

XiaoliChan, 2023-11-11:
I can confirm the network cable is working, because I plugin it into my notebook and get dhcp ipv4 address

robimarko, 2023-11-11:
Any chance you can revert commit:
https://github.com/openwrt/openwrt/commit/947b44d9ae17b4d56578c10a5dd675e79e4c621b

To make sure its not the culprit

XiaoliChan, 2023-11-11:
> To make sure its not the culprit

Sure, I'm going to recompile it

robimarko, 2023-11-11:
Can I also ask that you post output of:
`cat /sys/kernel/debug/clk/clk_summary`

XiaoliChan, 2023-11-11:
> Can I also ask that you post output of: `cat /sys/kernel/debug/clk/clk_summary`
[bug-firmware-clk.txt](https://github.com/openwrt/openwrt/files/13325203/bug-firmware-clk.txt)

XiaoliChan, 2023-11-11:
@robimarko I can confirm that commit cause the issues

hnyman, 2023-11-11:
Interesting, also ipq806x is suffering from the same symptom: no DHCP on wan.
I compiled r24365-b7cf706539 for ipq806x/R7800, but was left without wan connectivity (unanswered dhcp  broadcast).

The commit 947b44d9ae only affects ipq807x, so the reason for ipq806x might instead be 4fdd5530a7 by @Ansuel 

But interesting that similar symptoms are surfacing for both ipq806x and ipq807x.

Ansuel, 2023-11-11:
Feeling we are jumping to conclusions here. Think we need to investigate this further with actually reverting the commits

Ansuel, 2023-11-11:
@XiaoliChan can you do one test for me? Can you verify if each lan port have connection and you can reach the router?

XiaoliChan, 2023-11-11:
> @XiaoliChan can you do one test for me? Can you verify if each lan port have connection and you can reach the router?

I only can do the test in tomorrow

XiaoliChan, 2023-11-11:
> Can you verify if each lan port have connection and you can reach the router?

Oh, I can't reach the router, I remember I have a route has plugins each LAN port with devices

brada4, 2023-11-11:
> network cable is working 

Indeed UP+LOWER_UP says that and mii status confirms 100mbps is negotiated. 

Do you have all ports in a br-lan without VLAN-s? like /etc/config/network , you can keep static public IPs private if you wish.

XiaoliChan, 2023-11-11:
> Indeed UP+LOWER_UP says that and mii status confirms 100mbps is negotiated.
> 
> Do you have all ports in a br-lan without VLAN-s? like /etc/config/network , you can keep static public IPs private if you wish.

I think this is not the issue here, because when I revert the commit https://github.com/openwrt/openwrt/commit/947b44d9ae17b4d56578c10a5dd675e79e4c621b as @robimarko  said, it works now

hnyman, 2023-11-11:
Regarding ipq806x/R7800:
> > also ipq806x is suffering from the same symptom: no DHCP on wan.
> > ...
> > so the reason for ipq806x might instead be https://github.com/openwrt/openwrt/commit/4fdd5530a7ffba356ea9768769ed1713beb7445a

> investigate this further with actually reverting the commits

I reverted 4fdd5530a and connectivity of my R7800 is back  with r24365-b7cf706539.

I wonder if it is about the actual source code changes or about the removal of ip-tiny rom device packages. 
(however, ip-tiny was added due to DSA, so it makes sense that it should not be needed for anything else)

brada4, 2023-11-11:
If you loop br-lan dhcp server with wan dhcp client in a software bridge the result in semi-technical terms becomes "dhcp not working"

Ansuel, 2023-11-11:
> Regarding ipq806x/R7800:
>
>
>
> I reverted 4fdd5530a and connectivity of my R7800 is back  with r24365-b7cf706539.
>
> I wonder if it is about the actual source code changes or about the removal of ip-tiny [...]

Please open a new issue so we don't mix the 2 thing

XiaoliChan, 2023-11-11:
Looks like we need to revert that PR before find the bug.

Ansuel, 2023-11-11:
Problem debugged... preparing patch to fix this.

Ansuel, 2023-11-11:
@XiaoliChan please test https://github.com/openwrt/openwrt/pull/13950

XiaoliChan, 2023-11-11:
Ah sure, let me recompile the firmware with your patch

XiaoliChan, 2023-11-11:
> @XiaoliChan please test #13950

Ah sure, let me recompile the firmware with your patch

XiaoliChan, 2023-11-11:
@Ansuel That patch fixed the weird issues in WAN port on my AX3600

I still need to test the AX6 with full connection, maybe tomorrow

closed 2023-11-11 by openwrt-bot