openwrt/openwrt #13957: 100M Lan keine Verbindung zum Router (AX3600) 1GB works

stixif, 2023-11-12:
### Describe the bug

100M LAN keine Verbindung zum Router (AX3600) 1GB works
seit einem Patch im "MAIN" in den letzten 6 Tagen
habe ich Probleme mit LAN Teilnehmern die nur 100Mbit können
bei 1GB treten keine Probleme auf

laufender SNAPSHOT r0-a15eb1c
problem  hier erkannt SNAPSHOT r0-79ee0d2

https://github.com/stixif/ipq807x-ax3600-openwrt-builder

### OpenWrt version

r0-79ee0d2

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-12:
Try again in few days till fix lands in snapshots.
https://github.com/openwrt/openwrt/issues/13945

stixif, 2023-11-12:
Ok... but my Problem is not only on WAN... Port2 have the same Issu...

brada4, 2023-11-12:
Wait for snapshot, the fixed thing is more accurate active port enumeration.

stixif, 2023-11-16:
now it works! Thanks

closed 2023-11-16 by stixif