openwrt/openwrt #13965: connecting wan and lan on the same switch will break connection

NPCuse, 2023-11-13:
### Describe the bug

configuring two simple interfaces on my wallys wan and lan (assigning different ips) branching them both on the same switch and the board is unreachable

### OpenWrt version

r23607-5106f554bb

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Wallystech DR40X9

### Image kind

Official downloaded image

### Steps to reproduce

Scenario 1:
    Configuration:
        WAN and LAN interfaces are configured with different IP addresses.

    Testing Separately:
        WAN interface connected to a switch and pinged (successful).
        LAN interface connected to a switch and pinged (successful).

    Combined Connection:
        While pinging LAN, WAN interface is connected to the same switch.
        Result: Ping to LAN stops.

Scenario 2:

    Initial Ping on LAN:
        LAN interface is pinged (successful).

    Connecting WAN without Disconnecting LAN:
        WAN interface is connected without unplugging LAN.
        Result: Ping to LAN still works.

    Unplugging and Reconnecting LAN:
        LAN interface is unplugged and then reconnected.
        Result: Ping to LAN doesn't work anymore.

    Waiting Period:
        Wait for approximately 5 minutes.

    Observation after Waiting:
        After waiting, ping to LAN interface becomes successful again.
this is my /etc/config/network:

`
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd91:cf7a:18e1::/48'

config interface 'lan'
	option device 'lan'
	option proto 'static'
	option ipaddr '192.168.25.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option device 'wan'
	option proto 'static'
	option ipaddr '192.168.1.49'
	option netmask '255.255.255.0'
	option gateway '192.168.1.1'

config interface 'wlan1'
	option proto 'none'

`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

xback, 2023-11-13:
Could you please post the full output of command `ifconfig` ?

brada4, 2023-11-13:
How is your measurement device connected? Under normal conditions no swotch is predictably handling same mac on different ports. Either uae random mac on openwrt or check your switches documentation on how to disable all kinds of loop detection.

NPCuse, 2023-11-13:
this is my ifconfig result:

eth0      Link encap:Ethernet  HWaddr C4:4B:D1:00:5D:34  
          inet6 addr: fe80::c64b:d1ff:fe00:5d34/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:14971 errors:0 dropped:0 overruns:0 frame:0
          TX packets:232 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1805185 (1.7 MiB)  TX bytes:25642 (25.0 KiB)
          Base address:0x8000 

lan       Link encap:Ethernet  HWaddr C4:4B:D1:00:5D:35  
          inet addr:192.168.25.1  Bcast:192.168.25.255  Mask:255.255.255.0
          inet6 addr: fe80::c64b:d1ff:fe00:5d35/64 Scope:Link
          inet6 addr: fd91:cf7a:18e1::1/60 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:14972 errors:0 dropped:17 overruns:0 frame:0
          TX packets:209 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1595671 (1.5 MiB)  TX bytes:23064 (22.5 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:160 errors:0 dropped:0 overruns:0 frame:0
          TX packets:160 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:12480 (12.1 KiB)  TX bytes:12480 (12.1 KiB)

wan       Link encap:Ethernet  HWaddr C4:4B:D1:00:5D:34  
          inet addr:192.168.1.49  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

i'm not supposed to have any loop since i have different interfaces with differenct mac and ips
this is a basic scenario that works fine on openwrt22
the major difference i can see is the use of DSA architecture on openwrt23 with wallys board

xback, 2023-11-13:
Ok, i'll check this on my board.
hold on

brada4, 2023-11-13:
Please install supported release
https://firmware-selector.openwrt.org/?version=23.05.0&target=ipq40xx%2Fgeneric&id=wallys_dr40x9
Wan and eth0 mac-s are identical.

xback, 2023-11-13:
Yeah, I can simulate abnormal behaviour

koen@koen-nuc11:~$ ping 10.0.253.5
```
PING 10.0.253.5 (10.0.253.5) 56(84) bytes of data.
64 bytes from 10.0.253.5: icmp_seq=1 ttl=64 time=0.648 ms
64 bytes from 10.0.253.5: icmp_seq=1 ttl=64 time=0.648 ms (DUP!)
64 bytes from 10.0.253.5: icmp_seq=2 ttl=64 time=0.540 ms
64 bytes from 10.0.253.5: icmp_seq=2 ttl=64 time=0.541 ms (DUP!)
```

NPCuse, 2023-11-13:
> Please install supported release https://firmware-selector.openwrt.org/?version=23.05.0&target=ipq40xx%2Fgeneric&id=wallys_dr40x9 Wan and eth0 mac-s are identical.

so ive downloaded the ubi for the link u sent and flashed my board but still have the same exact behaviour

ive changed manually the wan MAC address and still get same results ....

NPCuse, 2023-11-13:
> Yeah, I can simulate abnormal behaviour
>
> koen@koen-nuc11:~$ ping 10.0.253.5
>
> ```
> PING 10.0.253.5 (10.0.253.5) 56(84) bytes of data.
> 64 bytes from 10.0.253.5: icmp_seq=1 ttl=64 time=0.648 ms
> 64 bytes [...]

can you please confirm the ping interruption when connecting both interfaces to the same switch?
or it just happen in my setup

xback, 2023-11-13:
@NPCuse 
I can confirm that pings interrupts on the lan port.

The wan keep pinging but with duplicate's
The link is also extremely jittery at this point.

brada4, 2023-11-13:
normally there is no disruption connecting cable.
Check with `ip link` that mac addresses are different. `option macaddr 'random'` is certain to change wan mac. Only problem may occur with identical macs and traffic balanced or duplicated at random at swotches' discretion.

NPCuse, 2023-11-13:
> normally there is no disruption connecting cable. Check with `ip link` that mac addresses are different. `option macaddr 'random'` is certain to change wan mac. Only problem may occur with identical macs and traffic balanced or duplicated at random at swotches' discretion.

ive changed the /etc/config/network : 
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd91:cf7a:18e1::/48'

config interface 'lan'
        option device 'lan'
        option proto 'static'
        option ipaddr '192.168.25.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option device 'wan'
        option proto 'static'
        option ipaddr '192.168.1.49'
        option netmask '255.255.255.0'
        option gateway '192.168.1.1'
        option macaddr '8A:F1:D6:0D:2B:37'
config interface 'wlan1'
        option proto 'none'

and rebooted the board ; this is the ifconfig:
eth0      Link encap:Ethernet  HWaddr C4:4B:D1:00:5D:34  
          inet6 addr: fe80::c64b:d1ff:fe00:5d34/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:109053 errors:0 dropped:0 overruns:0 frame:0
          TX packets:152 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:12400325 (11.8 MiB)  TX bytes:15734 (15.3 KiB)
          Base address:0x8000 

lan       Link encap:Ethernet  HWaddr C4:4B:D1:00:5D:35  
          inet addr:192.168.25.1  Bcast:192.168.25.255  Mask:255.255.255.0
          inet6 addr: fd91:cf7a:18e1::1/60 Scope:Global
          inet6 addr: fe80::c64b:d1ff:fe00:5d35/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:53410 errors:0 dropped:645 overruns:0 frame:0
          TX packets:105 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5316117 (5.0 MiB)  TX bytes:10958 (10.7 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:337 errors:0 dropped:0 overruns:0 frame:0
          TX packets:337 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:28240 (27.5 KiB)  TX bytes:28240 (27.5 KiB)

wan       Link encap:Ethernet  HWaddr 8A:F1:D6:0D:2B:37  
          inet addr:192.168.1.49  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::88f1:d6ff:fe0d:2b37/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:55714 errors:0 dropped:669 overruns:0 frame:0
          TX packets:32 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5565530 (5.3 MiB)  TX bytes:3064 (2.9 KiB)

brada4, 2023-11-13:
Please show 'brctl show' and you have 2 proto lines on wan.
Another thing that may interfere - if you run dhcp on lan interface (likely bridge) - it may plainly kick out dhcp server onm that network and move your workstation somewhere in other net config.

NPCuse, 2023-11-13:
> Please show 'brctl show' and you have 2 proto lines on wan. Another thing that may interfere - if you run dhcp on lan interface (likely bridge) - it may plainly kick out dhcp server onm that network and move your workstation somewhere in other net config.

the point that i removed the default bridge that come with openwrt "br-lan"; you can see it in my /etc/config/network file
it shows that i have now two interfaces wan and lan with different ip addresses 
i didnt configure any other service even my wireless interface is down

brada4, 2023-11-13:
Where do you measure from? Does problem persist if you replace switch with loop cable?
You need to walk from there with tcpdump to find damage spot generating double/ duplicating ping req or response.