openwrt/openwrt #13968: WR902AC Hardware watchdog causes repeatable router hardware freezes and compatibility issues with new V4 rev.

Edmondena, 2023-11-13:
### Describe the bug

The WR902AC routers integrated hardware watchdog as initialized during power up by the OpenWrt Linux Kernel causes highly reproducable hardware freezes and thus reoccurring filesystem damages in Extroot e.g. on micro SD cards (explored with btrfs and ext3/4 and swap).
Anytime the watchdog could not be properly triggered because (presumably) of its too high procd/ubusd nice level the TL-WR902AC hardware watchdog freezes the router.

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link TL-WR902AC v3/v4

### Image kind

Official downloaded image

### Steps to reproduce

1. Configure the Watchdog to something higher than >>60s "delay" and >>30s trigger "frequency" and just wait for the router to freeze.
2. Reproduction by intense usb load when creating a swapfile e.g. by
# Install Extroot Environment on new Sandisk SD card Extreme Pro and Micro Mate Adapter
      /dev/sda1 'Extroot' ext3/4, /dev/sda2 'Swap'
      configure /proc/sys/vm/swappiness -> 100 and vfs_cache_pressure -> 200
# tar -cvzf <to anyfile> /overlay/upper/ &
# dd if=/dev/zero of=/overlay/swap bs=4k count=32k
# maybe also Install heavy memory processes like adguardhome and config /proc/sys/vm/swappiness = 100 and vfs_cache_pressure = 200
# use plenty of space behind swap partition (>=4GB) for solid wear leveling

### Actual behaviour

Root Bug:
The hardware watchdog is leading to a hardware freeze when triggered instead of a hardware initiated rebooting. Instead of increasing the availability by a hard reboot it violates the reliability by a hardware freeze and according filesystem damages especially with SDcard memory extensions that require stable power to properly finish their wear leveling before any "normal" shutdown.

Sympthomatic Problems:
- The WDG triggering is not reliable as the procd/ubusd watchdog nice level is '0' and thus can be pushed away from triggering the watchod by most processes
- Any user, testing or maintenance process together with and enforced by heavy usb swapping or extroot access causes the WR902AC to freeze 

### Expected behaviour

Safe Watchdog behaviour:
1. As default behaviour only a reboot may be acceptable but only a soft reboot in advance of a following hardware reboot would be acceptable to not damage filesystems and micro sd cards
2. Never freeze any embedded hardware per SW default but it should be a users self responsibly configurable safe state (for very rare and special safety requirements) 
3. Non maskable watchdog triggers are required at least at lower (negative) system nice level's priority. Very efficient but very short trigger events to not harm systems performance
4. The Watchdog timeouts should have the longest possible and most robust default. I found out the Timeouts=60s and duty cycles (freuency=30s) seems to have the best possible effect. Longer (>60/30) lead directly to hardware freeze.

### Additional info

Problems with Openwrt 23.05.0
- persistent hard hardware watchdog reset not acceptable for IT equipment leads to availability lack, filesystem corruption and hardware (SDXC cards Extroot fs) damage.
     -> much better to never enable watchdog by default it, or configure it to 1. try a proper SW-shutdown, 2. try to reboot, 3. do a hw reset if nothing helps
- Procd init of ubusd has std. process prio of "0" which enables most and processes to suspend the watchdog from properly triggering the hw watchdog
    -> change ubusd and watchdog prio to e.g. -10 or similar

Easy reproduction by high usb load when creating a swapfile e.g. by
# tar -cvzf <to anyfile> /overlay/upper/ &
# dd if=/dev/zero of=/overlay/swap bs=4k count=32k

Work Arounds:
1. Disable it by using magicclose and watchdog stop
2. Double the timeout to 60s (helps a bit but not reliably with extroot and havy swap loads
3. Use nice priorities e.g. as offered by samba configuration in /etc/init.d/samba4 to renice all heavy swapping memory processes (in /etc/init.d/...) getting higher nice levels than procd/ubusd's "0" level including dropbear, so that users can't prevent the watchdog from triggering even when making backups or tests.

Environment:
- Openwrt 23.05.0
- WR902AC Router old V4 and probably also V3
- Extroot with BTRFS or EXT3/4
- Swap-Parition oder Swap-File
- Adguardhome, Dnscrypt-Proxy, Samba4....
- New drivers firmware vor Mediatek MT7615E installed
- Kmod-usb-storage, block-mount, kmod-fs-ext4 / kmod-fs-btrfs
- Micro-SDXCards with up to date Sandisc-Micromate SDXC/USB Adapter

Thank you for your great Work and thank you for improvements even of the great TP-Links WR902AC Tiny Travel Router 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-14:
Please check your other issue and try to participate at least with testing port to V4.

IO timeout is not specific to hardware.
You need to add some IO scheduler like kmod-iosched-bfq to make IO finish in .... deadline time, or make random access possible.
Disabling IO scheduler will make any USB storage dead slow, just run some iobench/ioping while having steady writes and reads of varied ionice classes. No need to crash mini system. It is actually just one IO request taking longer than watchdog (youll get hung_task soft watchdog on any full linux if you try hard enough)

Edmondena, 2023-11-14:
I'm sorry to mention that my only problem is the buggy hardware watchdog configuration which is presumably a configuration problem and not a hardware problem. Unintended freezing the router in case of a watchdog timeout is the big elephant in this room. Really sorry to mention this but such a configuration is a real defect on base of more the 15 years experience as a functional HW/SW safety engineer - Please don't do that, better fully disable this hardware watchdog completely instead of (surely unwanted) harming or wearing out everybodies SD or USB storage devices. A hardware watchdog is a critical safety feature and if not properly designed on a system level causing much more trouble then it anyone helps.

I guarantee not wanting to blame some one here as I love Openwrt this community works for years now. But I expect the many of this work is naturally done on base off trial and error and thus maybe not exactly knowing how to properly configure the hardware watchdog to avoid any hard freezing the system.

One simple example:
You are on holidays and expect your router to provide you a VPN tunnel for your private home access. Once logged in and transferring data the weak routers performance lead to a high loadavg. Procd, respectively ubusd cannot trigger the hardware watchdog anymore in time. The hardware watchdog of your router freezes you router and - cuts your connection, your offline. No chance to ever regain VPN access until weeks later you are back home to manually repower your router - Do you thing this behaviour is good? I don't know anyone that would support such a behaviour as default watchdog setup configuration that, today only can be switched off by programming the /etc/rc.local and required to deeply digging into watchdog setups and process prirotiy/nice level effect.
Really hope you could help me/us to improve all of this to hopefully gain much better OpenWrt stability than today?

brada4, 2023-11-14:
Your device is not yet supported and you use sub-par storage concoction together with intentional misconfiguration, and plainly refuse to work towards your ideal world with cheap things serving at enterprise quality level.
Please adjust your expectations to your hardware. Choose the service you run alongside wifi AP function. JUST ONE OF THEM.

Edmondena, 2023-11-14:
Sorry, I forgot to mention that this watchdog behavior ASFAIK is identical to the the long ago released WR902AC v3 that I'm even running today. This is the reason why I expect this bug to be a systematic one. But to ensure this, I can gladly support you and prove it again using one of my older v3 devices again.
If you don't agree with my above research work and conclusions you're free of choice to ignore my hints and arguments as its my choice for what application and how I wan't to use my router.

brada4, 2023-11-14:
Install IO scheduler for concurrent access. Disable watchdog if still no good, but it is just consequence of rest being sub-standard. IO request taking 60+s i would not call working well. Dont expect watchdog do reboot on unsupported device.

Edmondena, 2023-11-14:
Thank you much for fast reply!
I edited my above text for v3 revision in between, maybe you want to re read?
60s+ delay is directly leading into repeatable and sudden hw-freeze that I can confirm to that it even not working at all.

Do you know if someone knows howto configure the hw watchdig it to properly reboot instead of freezing? E.g. by triggering it witch "echo 0 or 1 or 2 or 3 >/dev/watchdog"?

brada4, 2023-11-14:
Check with the person talking in your other issue. They must know better if it works at all, or you can contribute burnout test.

Did it help your USB disk (has to be V60+ to saturate USB 2.0) to attach IO scheduler?

Edmondena, 2023-11-14:
Sorry but I don't understand your question? What is V60+? Maybe you mix me with some other person as I can't follow you?

In between I used the following test to confirm the watchdog freeze behavior for WR902AC v3 which I can't confirm:
- alias watchcrash="ubus call system watchdog '{\"frequency\":35}'; ubus call system watchdog '{\"timeout\":70}'"

HomeWrt ~ # watchcrash
{
	"status": "running",
	"timeout": 70,
	"frequency": 35,
	"magicclose": false
}
{
	"status": "running",
	"timeout": 70,
	"frequency": 35,
	"magicclose": false
}

Some second after this the "old" WR902AC **v3** rebooted as expected and so I **cannot confirm** the same v4 experienced hardware freeze for the v3 version.

brada4, 2023-11-14:
V60 class SD card means 60MB/s (480Mbps) write, like 8k video and better.
Please continue on porting effort. You will not get here further than "disable watchdog"

DragonBluep, 2023-12-11:
> 1. Configure the Watchdog to something higher than >>60s "delay" and >>30s trigger "frequency" and just wait for the router to freeze.

You can't do that. The max watchdog timeout value supported by the ramips target is 2^16 ms ≈ 65 seconds. This repport is not bug and should be closed.