openwrt/openwrt #13969: TL-WR902AC v4 routers 2nd 5G Wifi changed from mt76x0e (v3 5GWifi)  to mt7615e (v4 5GWifi)

Edmondena, 2023-11-13:
### Describe the bug

For the new v4 HW Version the obsolete old 76x0e driver is now obsolete, though the 2nd Wifi supported by the "old" mt7603 (2G4) would be a sufficient common base for both.
The new mt7615 drivers are too big to be included into a standard image as they'd bloat the base image for the new v4 and prevent other needed tools to be installed.

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link TL-WR902AC v3/v4

### Image kind

Official downloaded image

### Steps to reproduce

Install Openwrt V3 Image on V4 Hardware and see the following things to fail:
1. 5G Wifi Network don't work anymore
2. Dmesg show errors about missing mt7615 firmware images
3. mt7615 drivers don't fit into 8MB Flash together with kmod-usb-storage and kmod-fs-ext4
4. Hardware-Watchdog (see todays bug report) hard-freezes the router after beeing erroneously initialized during OpenWrt boot process instead of rebooting it

### Actual behaviour

WR902AC V4 cannot be properly setup using the luci interface by removing unneeded mt76x0e packages as they are leaving memory holes after deletion.
There are bloating kmods like e.g. pppox, pppoe that could be removed or reinstalled later as optiona for the base image as they are maybe not needed for most application.
5G Wifi cannot be run with the base installation as the wifi hardware was changed by TP-Link at HW-Revision v4.
BTRFS cannot be comfortably used together with extroot needed driver here as it includes somehow bloating zstd compression or other mods and libs included in this package that could maybe easily separated from it as unneeded for smaller routers with 8GBFlash / 64GB RAM

### Expected behaviour

- Have a minimum WR902AC base image including "old" simple mt7603 2G4 only as drivers sufficient to have both wifi and lan ready for access
- Easy install addon drivers e.g. mt76x0e for "old" v3 or mt7615 for "new" v4 hardware
- Easy postinstall pppoe/x kmods and libs only if required to reduce luci footprint not leaving memory holes after removal
- have smallest possible base flash memory footprint for comfortable luci setup and for extroot
setup maybe enabling to include btrfs-progs for filesystem checks outside of extroot
- have a bugfixed stable watchdog or disabled watchdog in the base image instead of a hardware freezing kernel initialized wadtchdog with too high wdg-trigger nice level from procd/ubusd.

### Additional info

It maybe is a good idea to more easy downsize the base WR902AC base image with smaller
memory footprint to support both v3 and new v4 hardware versions properly.
- reduce base image memory footprint be removing obsolete mt76x0e 5G Wifi drivers and libs
- reduce luci footprint by removing mostly not required kmod-ppp-xxx dependency
- heavily reduce btrfs-package footprint e.g. by zstd compression and libs to gain flash space for btrfs-progs

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-13:
You lucky new platform has similar hardware. You can open a PR with new hardware. Wifi drivers are kmod-s, old out, new in. If it is just 2 kmod-s it is likely good for next minor rel.
You cannot imagine how many providers use pppoe, focus on basic hw changes, not general components.

brada4, 2023-11-13:
Please revisit PR with your research in hand. Seems little teamwork can make it happen ;-)
https://github.com/openwrt/openwrt/pull/13845

kjonca, 2023-11-14:
> Please revisit PR with your research in hand. Seems little teamwork can make it happen ;-) #13845

Better: https://github.com/openwrt/openwrt/pull/13877

DragonBluep, 2023-12-11:
Fixed by https://github.com/openwrt/openwrt/commit/ffa40fcc7fe1aee041775e097bccecb9be83cd21. This issue can be closed.

cgm999, 2024-05-30:
I think the wmac eeprom offset is wrong  ,Right now we have 

&wmac {
        status = "okay";
        nvmem-cells = <&eeprom_factory_28000>, <&macaddr_factory_f100 0>;
        nvmem-cell-names = "eeprom", "mac-address";
};

However 2.4Ghz is not working properly on this Target wr902ac-v4 ..  so I think we should have just like on wr902-v3:

&wmac {
        status = "okay";
        nvmem-cells = <&eeprom_factory_20000>, <&macaddr_factory_f100 0>;
        nvmem-cell-names = "eeprom", "mac-address";
};

dgiese, 2024-12-12:
I think you want to create a PR for that.