openwrt/openwrt #13975: rtw88_8822bu module crashes on enable or disable access point

mirobiala, 2023-11-14:
### Describe the bug

When Enable or Disable access point for rtw88_8822bu device, rtw88_8822bu module crashes.

### OpenWrt version

r24396-09b7e1955c

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Self-built image

### Steps to reproduce

Enable or Disable access point via Luci->Network->Wireless.

### Actual behaviour

Module crashes

### Expected behaviour

Not to crash

### Additional info

root@OpenWRT:~# dmesg
[    0.000000] Linux version 5.15.138 (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24384-fbe48e99bd) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Nov 13 22:33:40 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=dfe968b2-291b-1ba0-741b-694a295ff302 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003ffd7fff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ffd8000-0x000000003fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b0000000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 1.16.2-debian-1.16.2-1 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 26e3001, primary cpu clock
[    0.000000] kvm-clock: using sched offset of 385843196364 cycles
[    0.000002] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000004] tsc: Detected 3799.996 MHz processor
[    0.000489] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000491] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000494] last_pfn = 0x3ffd8 max_arch_pfn = 0x400000000
[    0.000531] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001960] found SMP MP-table at [mem 0x000f5b70-0x000f5b7f]
[    0.001968] Using GB pages for direct mapping
[    0.002071] ACPI: Early table checksum verification disabled
[    0.002074] ACPI: RSDP 0x00000000000F59B0 000014 (v00 BOCHS )
[    0.002076] ACPI: RSDT 0x000000003FFE2154 000034 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.002080] ACPI: FACP 0x000000003FFE1F84 0000F4 (v03 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.002083] ACPI: DSDT 0x000000003FFE0040 001F44 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.002085] ACPI: FACS 0x000000003FFE0000 000040
[    0.002086] ACPI: APIC 0x000000003FFE2078 000078 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.002088] ACPI: MCFG 0x000000003FFE20F0 00003C (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[... 660 lines cut ...]
CONFIG_PACKAGE_px5g-mbedtls=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_rtl8822be-firmware=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_ucode-mod-nl80211=y
CONFIG_PACKAGE_ucode-mod-rtnl=y
CONFIG_PACKAGE_ucode-mod-uloop=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wireless-regdb=y
CONFIG_PACKAGE_wpad-mesh-mbedtls=y
# CONFIG_TARGET_IMAGES_GZIP is not set
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_WPA_MSG_MIN_PRIORITY=3
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-11-14:
mark

brada4, 2023-11-14:
> QEMU Standard PC (Q35 + ICH9, 2009)

Please verify function on clean hardware and seek guidance with your virtualisation vendor.

mirobiala, 2023-11-14:
> Please verify function on clean hardware and seek guidance with your virtualisation vendor.

Can you guide me how to do that? It's local QEMU virtual machine with USB Host Device "TP-Link Archer T3U Nano" on my desktop PC.

bluecmd, 2023-11-17:
I have the same issue with building from master using a [ALFA AWUS036ACU](https://gist.github.com/bluecmd/32dbdd5ceb65b430fb4bd5f69010bd60) on a [Cisco vEdge 1000](https://openwrt.org/inbox/toh/cisco/vedge_1000)

Snippet:
```
root@OpenWrt:/# [  113.150025] ------------[ cut here ]------------
[  113.154667] WARNING: CPU: 1 PID: 4862 at backports-6.5/drivers/net/wireless/realtek/rtw88/rtw8822b.c:823 rtw8822b_set_antenna+0x50/0xa8 [rtw88_8822b]
[  113.168119] write RF mode table fail
[  113.171701] Modules linked in: rtw88_8822bu rtw88_8822b pppoe ppp_async nft_fib_inet nf_flow_table_inet wireguard rtw88_usb rtw88_core rtl8xxxu pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nfv
[  113.171984]  ledtrig_usbport cryptodev ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 tunnel6 tunnel4 ip_tunnel tun xfrm_user xfrm_ipcomp af_key xfrm_algo marvell vxlan udp_tunnel ip6_udp_tunnel crypto_user algif_skcipi
[  113.308751] CPU: 1 PID: 4862 Comm: iw Tainted: G         C         6.1.62 #0
[  113.315818] Stack : 0000000000000000 0000000000000008 8000000005fe9100 80000000f009aa00
[  113.323849]         0000000000000000 80000000f5ba3650 ffffffff81a3dcb0 80000000f5ba35f0
[  113.331879]         0000000000000000 c0000000ffffefff 0000000000000001 ffffffffffffffea
[  113.339909]         0000000000000010 ffffffff81935000 fffffffffffe0000 80000000f5ba3535
[  113.347937]         ffffffff81a3dcb0 0000000000000001 0000000000000009 0000000000000009
[  113.355967]         ffffffffc05f02f0 0000000000000337 ffffffffc05f0330 80000000f58e03c0
[  113.363997]         8000000175ba3527 ffffffff81581ba8 0000000000000000 0000000000000000
[  113.372025]         80000000f5ba0000 80000000f5ba3650 80000000f5ba3dd8 ffffffff8190d6c4
[  113.380055]         0000000000000000 f948303aec9d40de 0000000000000001 ffffffff81a3dcb0
[  113.388084]         ffffffff81c31183 ffffffff8111a940 80000000f03b0e00 ffffffff8190d624
[  113.396113]         ...
[  113.398566] Call Trace:
[  113.401015] [<ffffffff8111a940>] show_stack+0x28/0xf0
[  113.406092] [<ffffffff8190d6c4>] dump_stack_lvl+0x5c/0x7c
[  113.411513] [<ffffffff8113aad8>] __warn+0x98/0xe0
[  113.416229] [<ffffffff8113ace0>] warn_slowpath_fmt+0x1c0/0x1c8
[  113.422076] [<ffffffffc05c0c10>] rtw8822b_set_antenna+0x50/0xa8 [rtw88_8822b]
[  113.429243] [<ffffffffc055a79c>] rtw_ops_set_antenna+0x64/0xa8 [rtw88_core]
[  113.436271] [<ffffffffc02db65c>] nl80211_set_wiphy+0x324/0x7a0 [cfg80211]
[  113.443171] [<ffffffff81771a48>] genl_family_rcv_msg_doit+0xd8/0x138
[  113.449542] [<ffffffff81771da4>] genl_rcv_msg+0x114/0x268
[  113.454953] [<ffffffff81770668>] netlink_rcv_skb+0xa8/0x148
[  113.460538] [<ffffffff817711c0>] genl_rcv+0x30/0x48
[  113.465427] [<ffffffff8176fa10>] netlink_unicast+0x238/0x380
[  113.471100] [<ffffffff8176fe40>] netlink_sendmsg+0x2e8/0x4f8
[  113.476772] [<ffffffff816dfd50>] ____sys_sendmsg+0xd0/0x2b8
[  113.482360] [<ffffffff816e1a88>] ___sys_sendmsg+0x80/0xd8
[  113.487773] [<ffffffff816e1f04>] __sys_sendmsg+0x4c/0xc0
[  113.493099] [<ffffffff81123b48>] syscall_common+0x34/0x58
[  113.498517] 
[  113.500034] ---[ end trace 0000000000000000 ]---
```

bluecmd, 2023-11-17:
FWIW, this also happens when I put the interface in client mode, so I do not believe this to be limited to being an access point.

hauke, 2023-11-26:
This looks like a problem in the upstream driver, please report this to the upstream Linux wireless mailing list.

mirobiala, 2023-11-29:
It turns out that these are two different problems:

1. For the call trace there is a patch :[https://patchwork.kernel.org/project/linux-wireless/patch/20231012140120.891411-1-amadeus@jmu.edu.cn/](https://patchwork.kernel.org/project/linux-wireless/patch/20231012140120.891411-1-amadeus@jmu.edu.cn/)
2. For the disconnection issue I found another discussion : [https://github.com/lwfinger/rtw88/issues/151](https://github.com/lwfinger/rtw88/issues/151), but looks like there is no fix for now.