openwrt/openwrt #13978: Xiaomi AX3200

MMAA7, 2023-11-14:
### Describe the bug

Connection stability problems

### OpenWrt version

23.05.0 (r23497-6637af95aa).

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi AX3200

### Image kind

Official downloaded image

### Steps to reproduce

Welcome,

I report to you a problem with my Xiaomi AX3200 router, mediatek/mt7622, 23.05.0 (r23497-6637af95aa).
The problem is an unstable internet connection. My fiber optic cable is symmetrical with a speed of 1 Gbps.
On the old manufacturer's software I had download speed of 947 Mbps, upload speed of 949 Mbps.
After installing OpenWrt 23.05.0, the download speed is 881 Mbps, upload is 193 Mbps.

I would like to add that I have Offloading hardware and software support enabled, as well as the function from network - interfaces - global network options - packet control. NOTHING HELPED. 

Please help or make corrections in the new edition. because he doesn't have the knowledge to fix the problem.

Best Regards

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2023-11-14:
Similar issue as: https://github.com/openwrt/openwrt/issues/13771 and https://github.com/openwrt/openwrt/issues/13739
Please add missing information there and close this issue.

danpawlik, 2023-11-14:
So why don't you update other issue but you create new one?

MMAA7, 2023-11-14:
This is my first problem. Perhaps another one about AX3200. Other problems regarding this router are already closed.

MMAA7, 2023-11-14:
What information do you expect and what commands should I enter to get it?

danpawlik, 2023-11-14:
If the issue is in the LAN/WAN connection, for example:
- `cat /etc/config/network` (please check how to add block code in markdown)
- details if you enable HW offload, SW offload, packet stering
- did you try to use snapshot builds and the issue still apears?
- did you add some custom Firewall rules?
- are you using PPPoE auth for your ISP?
- do you use irqbalance?
- is something strange in `dmesg`?
- is something strange in `logread` ?

Still, I think you should ask on the openWRT forum (https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/) instead of creating a bug.