openwrt/openwrt #13989: 23.05.2 wrong routes emit from uci

brada4, 2023-11-16:
### Describe the bug

Consider following route config:
```
config route
        option type 'blackhole'
        option target '224.0.0.0/3'
        option metric '12345'
        option interface 'loopback'
        option gateway '0.0.0.0'
        option table 'main'
```
Emits 0/3 route in place of intended

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

Add fragment to config/network

### Actual behaviour

Wrong route is applied as observable by inability to connest great part of internet.

### Expected behaviour

Configured multicast forwarding happens

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-16:
@Ansuel plese undo your nag script and its consequences ASAP

Ansuel, 2023-11-16:
@brada4 WOW THAT IS EXTEREMLY TOXIC MY GOD -.-

Ansuel, 2023-11-16:
@brada4 thanks for the tag I fixed the bug with the labeller :D (you could have been more friendly with saying, "I think there is something wrong with the applied tag" but whatever)

brada4, 2023-11-16:
Now that bot garbage is deleted everybody is superpluspolite.

Ansuel, 2023-11-16:
@brada4 sure lets delete everything... please for the next time be more kind... :(

brada4, 2023-11-16:
Also 172.16/12 gets replaced by 172.0/12
No problem there with 23.05.0 (used to work starting from 18.06.1), problem appears only on 23.05.2 , no matter ip is iproute2 or busybox.

Shine-, 2023-11-16:
This was, to my knowledge, an issue with 23.05.**1**, which was pulled just because of that bug. 23.05.**2** doesn't have this bug anymore. Also see release notes, openwrt/netifd#17 and 842932a63d8993150ff339f11f0cd0d5dc45e6cc.

Since I have just upgraded my devices to 23.05.2 this morning, I tested the above `/etc/config/network` snippet. Working fine in 23.05.2 r23630-842932a63d.

jow-, 2023-11-16:
Was about to write the same, this looks like issue https://github.com/openwrt/netifd/issues/17 which was the reason for skipping 23.05.1. I just tested in a x86/64 VM and it works here as well.

```
root@OpenWrt:/# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS=''
root@OpenWrt:/# uci show network
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.globals.ula_prefix='fd1e:caa0:755f::/48'
network.@device[0]=device
network.@device[0].name='br-lan'
network.@device[0].type='bridge'
network.@device[0].ports='eth0'
network.lan=interface
network.lan.device='br-lan'
network.lan.proto='static'
network.lan.ipaddr='192.168.1.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.@route[0]=route
network.@route[0].type='blackhole'
network.@route[0].target='224.0.0.0/3'
network.@route[0].metric='12345'
network.@route[0].interface='loopback'
network.@route[0].gateway='0.0.0.0'
network.@route[0].table='main'
root@OpenWrt:/# ip r
192.168.1.0/24 dev br-lan scope link  src 192.168.1.1 
blackhole 224.0.0.0/3  metric 12345 
root@OpenWrt:/# 
```

brada4, 2023-11-16:
Could be bitness issue, i can enter route manually via ip route -> 172.16 -> wg0 224/3-> stb vlan , but not via web or uci. making aarch64 vm....

jow-, 2023-11-16:
I suppose you didn't use a vanilla image but a self-built one from the ImageBuilder or AUC? Release images contain fixed netifd version 2023-11-10-35facc83-1.1 while many target repos still ship the broken netifd version 2023-11-10-35facc83-1, e.g. here https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/base/

This should sort itself once the repos are updated (should happen today or tomorrow latest)

brada4, 2023-11-16:
```
opkg list netifd
netifd - 2023-11-10-35facc83-1 - OpenWrt Network Interface Configuration Daemon
```

jow-, 2023-11-16:
Yep, need to upgrade to `...-1.1` to fix this. The update should appear in the repo soon.

brada4, 2023-11-16:
firmware selector stands prominently on first page.

brada4, 2023-11-16:
auc wants to downgrade right away.
```
root@OpenWRT:~# auc -n
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on ramips/mt7621 (asus,rt-ax53u)
Available: 23.05.2 r23630-842932a63d
Requesting package lists...
 netifd: 2023-11-10-35facc83-1.1 -> 2023-11-10-35facc83-1
Are you sure you want to continue the upgrade process? [N/y] 
```

ynezz, 2023-11-16:
> auc wants to downgrade right away.

This should sort itself once the package repos are updated, when [the build](https://buildbot.staging.openwrt.org/openwrt-23.05/packages/#/builders/20/builds/63) finishes.

closed 2023-11-16 by ynezz