openwrt/openwrt #13991: Configuration not persisted.

jmranger, 2023-11-17:
### Describe the bug

Just upgraded an Ubiquity Nanostation AC Loco from 22.03.5 to 23.05.2. I asked to keep the configuration, but upon reboot, while the update had been done, the configuration hadn't been restored.

At this point, the device appears to be fully read-only:
- no change is persisted following a reboot
- reflashing with the "keep settings" checkbox unchecked doesn't help
- reflashing (through LuCI) with 22.03.5 does nothing - upon reboot, device's still running 23.05.2

Device was updated to pretty much all 22.03.x releases without issue, so I don't really suspect a hardware issue.

I haven't tried flashing from TFTP yet. I plan to try this over the weekend.
I don't have a second one, so no idea if this is reproducible.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ath79/generic

### Device

Ubiquiti Nanostation AC loco (WA)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

voxelcat, 2023-11-18:
It's a known issue with some older Ubiquity devices, see https://github.com/openwrt/openwrt/issues/13750 or https://forum.openwrt.org/t/powerbeam-m5-xw-configuration-loss-after-reboot/141925.

I would also like to see this addressed but so far I can't find out where and how this bug was introduced. 
Flashing back to openwrt 22 does however work for me on my Nanostation Loco M XW.

jmranger, 2023-11-18:
@voxelcat: Thanks. I'll close this one as a duplicate.

closed 2023-11-18 by jmranger