openwrt/openwrt #13992: Latest OpenWRT version can't read MAC on Meraki MR42

nickshanks347, 2023-11-17:
### Describe the bug

On OpenWRT versions 23.05.1 and 23.05.2 the MAC address is randomised on each reboot presumably because it can't read it from the hardware correctly. This issue is not there on versions 23.05.0 and below including the release candidates.

### OpenWrt version

N/A

### OpenWrt release

23.05.1 - 23.05.2

### OpenWrt target/subtarget

ipq806x/generic

### Device

Meraki MR42

### Image kind

Official downloaded image

### Expected behaviour

MAC address should be read correctly from the hardware.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-11-17:
Could you please provide a diff between the logs from 23.05.0 and 23.05.2?
(redact personally identifying information)

neheb, 2023-11-18:
https://github.com/openwrt/openwrt/compare/v23.05.0...v23.05.1

maybe @Ansuel has an idea...

Ansuel, 2023-11-18:
Maybe this? 0da199f60c47a4b861562ffec3842535eb56a67b

Have to check the device

nickshanks347, 2023-11-19:
> Could you please provide a diff between the logs from 23.05.0 and 23.05.2? (redact personally identifying information)

What specific logs?

nickshanks347, 2023-11-19:
> Maybe this? [0da199f](https://github.com/openwrt/openwrt/commit/0da199f60c47a4b861562ffec3842535eb56a67b)
> 
> Have to check the device

Maybe it is this. What am I checking the device for?

brada4, 2023-11-22:
worth installing openwrt to check

nasbdh9, 2024-05-05:
Same error here https://github.com/openwrt/openwrt/issues/15393

nasbdh9, 2024-05-05:
> Maybe this? [0da199f](https://github.com/openwrt/openwrt/commit/0da199f60c47a4b861562ffec3842535eb56a67b)
> 
> Have to check the device

Can confirm that MAC address cannot be read correctly starting from b649b0bf7100bdc6adb7e857c74738cab7c47b4c

appleeimac, 2024-05-21:
Can also confirm that the MR42 does this, on 22.05.3, and the 6.6 Kernel SNAPSHOT fixes it, but, it breaks the Wireless Drivers with a "Generic Unknown", on all three radios.

rkboni, 2024-11-04:
Should be fixed in https://github.com/rkboni/openwrt/commit/cb8b632f1ad8bee438139aaf8c947879380fbdc2 (verified on a newly-installed MR42).