openwrt/openwrt #13993: No Netgear WAX620 target in 23.05.X

dw4rren, 2023-11-17:
### Describe the bug

The 23.05.0 [release notes](https://forum.openwrt.org/t/openwrt-23-05-0-first-stable-release/174239) state that [Netgear WAX620 (IPQ8072A)](https://forum.openwrt.org/t/openwrt-support-for-netgear-wax620/156087) is supported, but I don't see it as a firmware download or listed when I try to build it with the image builder for 23.05.0 or 23.05.2.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ipq807x/generic

### Device

Netgear WAX620

### Image kind

Self-built image

### Steps to reproduce

1. Download and extract 23.05.X image builder.
2. `make info | grep netgear_wax620`

### Actual behaviour

No `netgear_wax620` target image

### Expected behaviour

I expected netgear_wax620 to be a target image

### Additional info

netgear_wax620 doesn't exist in the image builder's `target/linux/ipq807x/image/generic.mk`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ansuel, 2023-11-17:
@dw4rren Indeed it's a typo... dts is not present in 23.05. Only way is to use SNAPSHOT :(

brada4, 2023-11-17:
23.05.2 kind of mentions wifi support for it (not device)
https://openwrt.org/releases/23.05/changelog-23.05.2

dw4rren, 2023-11-20:
@Ansuel Thanks for the response. I can use SNAPSHOT for now. As far as stable support goes: do I need to wait until the next stable series or will the next 23.05 version include support?

hauke, 2023-11-26:
I updated the release notes.

closed 2023-11-26 by hauke