openwrt/openwrt #13994: Add STMMAC for x86_64 build - Compulab Fitlet 3 Mini PC

dekashio, 2023-11-17:
### Describe the bug

In order to get openwrt to recognize the ethernet adapter it's needed to add STMMAC to kernel_menuconfig.
It doe's work aftet the change but changes the hash of the kernel so in order to install any other kmod-* package there is a mismatch and i have to recompile every thing myself.
Maybe adding a special Variant specific to Fitlet 3? 

### OpenWrt version

23

### OpenWrt release

23.05

### OpenWrt target/subtarget

X86_64

### Device

Compulab Fitlet 3

### Image kind

Self-built image

### Steps to reproduce

Install pure x86_64 image on the pc

### Actual behaviour

Doesn't recognize the ethernet adapters.

### Expected behaviour

Working ethernet adapters.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dekashio, 2023-11-17:
> Invalid Target/Subtarget reported. `X86_64` Is this from a supported device?

X86/64

brada4, 2023-11-17:
They write ubuntu supported. Can you check with that lspci and ethtool -i to cross-check IF driver in v23.05.X kernel will be able to drive it?

dekashio, 2023-11-17:
When i modified the kernel using make kernel_menuconfig and checked STMMAC,
it works fine.

dekashio, 2023-11-17:
It is however a default part of ipx806x kernel build profile.

brada4, 2023-11-17:
Should be made as a PR towards (noting that it will work with 23.05 series based on your tests)
package/kernel/linux/modules/netdevices.mk

Likely it will initially land as kmod-???? package selectable from imagebuilder/firmwareselector unless it becomes as prolific as intel or realtek or broadcom netcards.

dekashio, 2023-11-17:
Thanks for your reply, so i'll make a pull request and see hoe it goes from
there?

brada4, 2023-11-17:
Down the road you will get kmod-stmmac built with 23.05.2 SDK that is installable on your machine.

dekashio, 2023-11-17:
Sorry for my ignorance, should i make anything else (PR?) or just wait?

brada4, 2023-11-17:
If you change file and are able to test new kmod it is best way to proceed.

hauke, 2023-11-26:
Please try #14079 and report back in the pull request.

commit 4b4c940f (2023-12-01): x86: Add kmod-dwmac-intel

This adds the Intel Ethernet driver for the Intel Quark/EHL/TGL chips.

Fixes: #13994
PR #14079 (2023-12-01): X86: add kmod dwmac intel
 * armsr: Move kmod-stmmac-core to common place
    
    Move the kmod-stmmac-core package to the common place to share it with
    x86 later.
    
* x86: Add kmod-dwmac-intel
    
    This adds the Intel Ethernet driver for the Intel Quark/EHL/TGL chips.
    
    Fixes: #13994