openwrt/openwrt #13995: TP-Link Archer C5 v1: 2.4GHz txpower over 17 dBm isn't applied (even when available)

julienrobin28, 2023-11-17:
### Describe the bug

On the Archer C5 V1, even when available (depending on the currently applied reg domain), setting the txpower of the 2.4 GHz (phy1 / QCA9550) to available values above 17 dBm, seems to results in 17 dBm of applied txpower at each time.  

Here is the output of `iwinfo phy1 txpower` after having tried to set 20, 24 and 25 dBm with "AU" country code, which allows up to 25 dBm for this device:  
```
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
* 17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
  24 dBm ( 251 mW)
  25 dBm ( 316 mW)
```

I believe the original firmware wasn't capped at 17 dBm/50 mW, so it seems that failing to set higher values is a bug.  

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C5 v1

### Image kind

Official downloaded image

### Steps to reproduce

Set a combination of reg domain country code and channel, which should allow more than 17 dBm on 2.4 GHz channels, and apply (and enable if not already done) the WiFi access point.  

Then select through the refreshed list of available txpower above 17 dBm from WebUI for example, or even from `/etc/config/wireless`, for example:  
```
config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/ahb/18100000.wmac'
        option band '2g'
        option country 'AU'
        option htmode 'HT20'
        option channel '9'
        option txpower '25'
```
You can even use `iw dev phy1-ap0 set txpower fixed 2500` to ask for 25 dBm for example.  

### Actual behaviour

Looking at `iwinfo phy1 txpower`, `iwinfo` or even the WebUI, will tell you that the current txpower is 17 dBm.  

### Expected behaviour

Available values from `iwinfo phy1 txpower` output, and from the list of available txpower values from the WebUI, should be applied successfully, as long as the reg domain country code has been set and applied.  

### Additional info

Setting the value from 15 dBm to 20 dBm sets the tx power to 17 dBm so the command isn't completely ignored (like it could have been be for an incorrect / out of range value).  

No problem for the 5 GHz QCA9880 phy0 however. Depending on the country code, value as high as 30 dBm can be set successfully on some 5 GHz channels (up to 26 dBm with my actual country code).  

I noticed this issue some years ago already (on older versions of OpenWrt) so this problem isn't new. Tonight I finally tried to understand it by Googling quite a while about the "17 dBm" limit, but found nothing detailed about it

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-17:
17dBm is 20dBm EIRP adjusted by 3dBm fixed antenna.

julienrobin28, 2023-11-18:
Thanks for the answer, however this approach is more than confusing, and if this is on purpose, there are 3 important mistakes doing so.

### 1 - For the 3 dB difference: wrong chip

Then I believe this 3 dB difference is applied to the wrong chip.  
If I'm right the 2.4 GHz antennas are those 3 internal kind of [copper strange shaped things](https://www.smallnetbuilder.com/wp-content/uploads/2016/09/tplink_archerc5v1_board.jpg), which are supposed to emit more equally in more directions.

The 3 external cylindrical / rod shaped antennas which are supposed to emit only to the perpendicular directions (but not above and not under them), and which are generally sold described as 3 dBi, are for the 5 GHz.  

### 2 - Wrong value displayed and used by reg domains

I believe I didn't see such a confusing behavior on other devices. With device and reg domains working on wrong value then having another kinda hidden feature unexpectedly applying another one to compensate an error (but not always) by creating an error in the other direction (but then the displayed value is still wrong). 
And currently for example, for a 10 dBm Tx-Power selection, if the 3 dB correction is indeed required, then it isn't applied.

### 3 - About ignored country codes limits

I believe that if it is fine to ignore some country code features because it is considered as an impossible use case, then one approach would be to have them removed from available country codes, to be much less confusing. 

For the country code indeed AU was just an example, from where I live I'm using FR country code. Which is indeed limited to 20 dBm for 2.4 GHz (I tried AU just to check the behavior about even more available but unreachable options). I could have used AS BM BR BZ CA CU DM DO EC FM GD GT GU GY HK HT ID IN KR MH MO MP NI NZ PA PR PW SG TW US VE VI...

Or more another cleaner approach, if banned values are on purpose, instead of deleting their associated country code, reduced maximum Tx-Power and frequency ranges should be set and reported by the driver.

### ART Partition
Is this 3 dB correction really required, or aren't the driver and TP-Link provided ART partition already taking into account the 3 dB antennas gain?

brada4, 2023-11-18:
Looks like 2.4g takes EU caldata where typical or fixed antenna has to be counted towards EIRP

julienrobin28, 2023-11-18:
Oh, then it would mean, this confusing limitation above available values isn't coming from a manually / badly added quirk into the "tx-power set" functions of the QCA9550 used driver, but rather, this is how the driver currently applies the content of calibration data from ART Partition 😨 but this looks so wrong

But anyway I may have to rely on the true experts for untangling this, because apart from being able to make good use of OpenWrt, and noticing such troubles, tracking and correcting the root cause of this behavior is a little bit outside of my scope of skills (at least for today), in case it wasn't already obvious 😅 

[C46E1FEAF5C6-mtd5-art.bin](https://pix-server-sorel.luoss.fr/Manual/openwrt-debug/archer-c5/C46E1FEAF5C6-mtd5-art.bin)

Please find above the content of the ART Partition I'm using here (extracted using OpenWrt WebUI), in case it can be used or analyzed to confirm hypothesis or track this issue further.

[image]

If needed, I may have access to another same device (but not from where I am today).

Many thanks for the work

brada4, 2023-11-18:
The restricted versions have C5 (EU) or C5 (US) written in top line. Behaviour shows EU restriction, like you had UK firmware flashed at one point like versionitis in the shop?
You need to revert to factory image and flash v1.20 from outside ETSI and US/CA.

julienrobin28, 2023-11-18:
Looking at my archived files about my Archer C5 V1.2 devices, I found back this information about what were the TP-Link factory firmware versions I was using before installing OpenWrt.

It turns out I did that on one of my 2 devices. However the issue is the same on my 2 devices. Confirmed by re-installing the full firmware before going back to OpenWrt just in case.

### 1st device: Archer_C5_V1_151111

For my 1st Archer C5 V1.2 device, before installing OpenWrt I previously did flash [Archer_C5_V1_151111](https://static.tp-link.com/res/down/soft/Archer_C5_V1_151111.zip) while running the TP-Link factory firmware (so that everything got updated including u-boot).  

### 2nd device: Archer C5_V1.20_150428

However for my 2nd Archer C5 V1.2 device, which is experiencing the same issue, before installing OpenWrt I previously did flash [Archer C5_V1.20_150428](https://static.tp-link.com/res/down/soft/Archer_C5_V1.20_150428.zip).

### Double check method

To double check all this, I compared the 1st one's `mtd0.bin` file extracted from WebUI and the u-boot portion of the Archer C5 V1 2015-11-11 firmware (extracted using `dd "if=ArcherC5v1_en_eu_3_14_3_up_boot(151111).bin" bs=512 skip=1 count=256 of=151111-uboot-only.bin`. By converting my [mtd0.bin](https://pix-server-sorel.luoss.fr/Manual/openwrt-debug/archer-c5/30B5C27F4616-mtd0-u-boot.bin) to [mtd0.hex](https://pix-server-sorel.luoss.fr/Manual/openwrt-debug/archer-c5/30B5C27F4616-mtd0-u-boot.hex) with hexdump, and doing the same for [151111-uboot-only.bin](https://pix-server-sorel.luoss.fr/Manual/openwrt-debug/archer-c5/151111-uboot-only.bin) to [151111-uboot-only.hex](https://pix-server-sorel.luoss.fr/Manual/openwrt-debug/archer-c5/151111-uboot-only.hex), using Meld show that apart from an additional blob at the end, this 1st device is still running OpenWrt above parts of the `Archer_C5_V1_151111` firmware.  

The 2nd one however is still running OpenWrt above `Archer C5_V1.20_150428` firmware, confirmed by comparing [150428-uboot-only.img](https://pix-server-sorel.luoss.fr/Manual/openwrt-debug/archer-c5/150428-uboot-only.img) and [C46E1FEAF5C6-mtd0-u-boot.bin](https://pix-server-sorel.luoss.fr/Manual/openwrt-debug/archer-c5/C46E1FEAF5C6-mtd0-u-boot.bin) into their [150428-uboot-only.hex](https://pix-server-sorel.luoss.fr/Manual/openwrt-debug/archer-c5/150428-uboot-only.hex) and [C46E1FEAF5C6-mtd0-u-boot.hex](https://pix-server-sorel.luoss.fr/Manual/openwrt-debug/archer-c5/C46E1FEAF5C6-mtd0-u-boot.hex) form.

### Full reflash to v1.20 TP-Link factory firmware had no effect

I still managed to reinstall the factory image `Archer C5_V1.20_150428` (cropping the 257 first 512 bytes sectors before flashing by OpenWrt sysupgrade), and using the original firmware, re-flashing again the full factory firmware file, before going back to OpenWrt but unfortunately, no difference.

brada4, 2023-11-18:
Ouch. Hardware-wise device should be able to do full AU power. 
Look at the [device page](https://openwrt.org/toh/tp-link/archer-c5-c7-wdr7500) - seems in US they drop by 3dBm which should not happen normally (and hints 00 regdom may permit at least 20dBm not looking into antenna)
https://github.com/openwrt/openwrt/blob/9062e5faaedc03823ee419fe34de1de73f48babc/target/linux/ath79/dts/qca9558_tplink_archer-c5-v1.dts#L63

julienrobin28, 2023-11-18:
Interestingly enough, I just noticed `iw reg get` tells me `phy0` and `phy1` are stuck to the US regdom, despite having mine (FR) expected to be enabled for both phy0 and phy1 (even after a reboot).

Same happens to both routers (the one that has been flashed to `Archer_C5_V1_151111` previously (I have now access to it too) and the one that has been flashed to `Archer C5_V1.20_150428` previously.

```
# Using FR regulatory domain country code
root@OpenWrt:~# iw reg get
global
country FR: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
	(57000 - 71000 @ 2160), (N/A, 40), (N/A)

phy#1
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)
```

I was also curious about trying out 00 regdom, so I tried it. But when trying to enable 00 - World, both Hotspots are keeping disabled. A reboot makes no difference, even after having switched to compatible channels of course.
```
# Using 00 - World regulatory domain country code
root@OpenWrt:~# iw reg get
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#1
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
[... 46 lines cut ...]
[   21.702000] ath: country maps to regdmn code: 0x3a
[   21.702007] ath: Country alpha2 being used: US
[   21.702014] ath: Regpair used: 0x3a
[...]
[   44.856203] ath: EEPROM regdomain: 0x80fa
[   44.856225] ath: EEPROM indicates we should expect a country code
[   44.856232] ath: doing EEPROM country->regdmn map search
[   44.856238] ath: country maps to regdmn code: 0x37
[   44.856245] ath: Country alpha2 being used: FR
[   44.856252] ath: Regpair used: 0x37
[   44.856258] ath: regdomain 0x80fa dynamically updated by user
[   44.856343] ath: EEPROM regdomain: 0x80fa
[   44.856350] ath: EEPROM indicates we should expect a country code
[   44.856357] ath: doing EEPROM country->regdmn map search
[   44.856362] ath: country maps to regdmn code: 0x37
[   44.856368] ath: Country alpha2 being used: FR
[   44.856373] ath: Regpair used: 0x37
[   44.856379] ath: regdomain 0x80fa dynamically updated by user

```

Anyway, here is the full dmesg log, may be I should have provided it sooner:
[dmesg-archer-c5-v1.2-openwrt23.05.2-started-fr-switched-to-00.txt](https://github.com/openwrt/openwrt/files/13401415/dmesg-archer-c5-v1.2-openwrt23.05.2-started-fr-switched-to-00.txt)

By the way, do we know if others Archer C5 V1.2 users are also affected by the issue? Yesterday I was assuming I was experiencing the same thing as all OpenWrt "Archer C5 V1" users, having just took a closer look at it after a long while ignoring this issue. I would feel sorry if this bug report turns out to be an individual troubleshooting case (but I still hope it is not).

brada4, 2023-11-18:
ath10k shows correct phy power when brought up and the card cc stays US even though it supports only etsi dfs... 17dBm is correct for France. You might want to power device off to get initial domain change though.

julienrobin28, 2023-11-18:
I just tried to power off the device after having applied FR country code (because indeed most of the time I'm just rebooting), to check what happens at the following start-up.  

However the new `dmesg` output is showing no difference.

[dmesg-archer-c5-v1.2-openwrt23.05.2-regdomain-fr-then-power-off-then-new-start.txt](https://github.com/openwrt/openwrt/files/13401472/dmesg-archer-c5-v1.2-openwrt23.05.2-regdomain-fr-then-power-off-then-new-start.txt)

But this attempt makes me thinking of another one:  

Few hours ago, when fully reflashing `Archer C5_V1.20_150428` just to be sure, I believe "US" was the default regdomain into TP-Link original firmware configuration (then I didn't even change this settings before flashing OpenWrt again - however, I'm sure others country codes were available into the list).

But wouldn't it be a good idea if I try, just in case, to change the settings from the original TP-Link firmware before flashing OpenWrt? Just in case it writes something somewhere that would be kept fixed after OpenWrt installation, and affecting the behavior then. I have backups of all current mtd partitions so comparison after this would be easy if it turns out to be a win.

brada4, 2023-11-18:
17dBm is legit conf in your location. Older version/production date caldata may permit 20dBm on removable antenna connectors, obeying your local laws is not a bug.

julienrobin28, 2023-11-18:
To finish about my last chance attempt, I just tried to go back the the original factory firmware to change the setting from US to FR before flashing back to OpenWrt. But it made no difference.

### To summarize about this:

#### About the 2.4 GHz phy (with internal antennas):

 * you believe that the 17 dBm Tx-Power value that the driver is failing to overcome (despite showing higher values as available), is supposed to give a 20 dBm EIRP in some directions, which (if it is right) would give a correct maximum value of 20 dBm in a lot of countries
 * the country from which I use the device, has a 20 dBm Tx-Power limit on the 2400-2483 MHz range
 * meaning that if the +3 dB on EIRP value is right, failing to set any Tx-Power value above 17 dBm is not supposed to be an issue, at least not for me
 * selecting 20 dBm and having 17 dBm of obtained Tx-Power is then supposed to give 20 dBm EIRP
 * selecting 17 dBm and having 17 dBm of obtained Tx-Power is then (not?) supposed to give 20 dBm EIRP
 * values which are shown as available, depending on selected country codes are not actually available because this device is used in France and/or contains calibration data which are specific to this zone
 *  this behavior is likely to be due to what the driver is doing with the calibration data, into the ART partition, which is provided on factory by TP-Link

#### About the 5 GHz phy (with external antennas):

Nothing about the points listed above about the 2.4 GHz phy applies to the 5 GHz phy and its external antennas. Also, 5 GHz rod shaped antennas are less prone to be isotropic than the internal ones, leading me to think that if the 17 dBm limitation because it is 3 dB under 20 dBm really was intended, it is not even applied to the right phy. 

### To conclude:

As I believe only the 5 GHz phy is behaving normally, and having never seen anything as much confusing as this 2.4 GHz phy behavior into any other device, this leads me to still think that this behavior is (at least) an issue, if not simply a bug, as it seems way too weird and not consistent to me. But I may be wrong and I'm not the expert here.

If @brada4 (you are more an expert than me!) you think that there is no bug, there is nothing more I'm able to do anyway for now. 

Digging more would need me to have a better understanding of calibration data and how the driver is using it, which may be interesting some day, but for now I also have others devices that aren't affected by this issue, having a much stronger reception probably because they are newer, and also successfully showing running on 20 dBm Tx-Power on 2.4 GHz range with the FR country code.

Many thanks @brada4 anyway for the time you spent for me on this.

closed 2023-11-18 by julienrobin28

brada4, 2023-11-18:
it is about interference. If you start 400x overpowered transmission as from Australia in a city in France whole block suffers.

julienrobin28, 2023-11-19:
We completely agree on this point 👍 and I'm grateful my neighbors too (or at least, most of the time, it is their ISP provided IAD that does). Having applied rules to make things works is a chance that should be preserved for one of our most precious gift: the holy Internet access 🤩  😁  👍

More seriously even when not thinking about the others, it is also obvious that having too much more power in a direction than in the other is pointless if both the device and the router have to talk and hear each other. Today we even have mobile phone calls automatically switching through VoWiFi for example, so having something balanced and reliable when reachable, is becoming even more essential than before, in addition of having something usable for everyone.

julienrobin28, 2023-11-20:
Before leaving all of this stuff definitely (and for not having the regret of not having it done), I finally spent the necessary amount of time digging for real into this, to get the real conviction that the finally obtained 17 dBm limited Tx-Power value at least was't a driver bug.  

__It turns out that 17 dBm is indeed the right value according to my ART partition when DFS region is set to ETSI.__  

### Final conclusion

Despite the really confusing behavior of 
 * showing unavailable values as available  
 * Tx-Power limit value being under what is permitted by both the hardware and regdomain  
 * statements about it to be an automatically applied 3 dB offset using the antenna gain  

Actually this is ends up being just the ART partition values that are limiting the applied Tx-Power value at the very final step of the Tx-Power set function, depending on WiFi mode and DFS region.

Hoping one day those explanations may help someone trying to understand the same issue!

By the way, setting the country code to `AU` was also setting the DFS region to ETSI (thus enforcing ETSI limits) which also has contributed to make this so confusing (but this may be another unrelated issue).  

### What I did:  

 * rebuilding OpenWrt with `CONFIG_PACKAGE_ATH_DEBUG` (+ `luci` `openssh-sftp-server` `nano` packages embedded by the way)  
 * putting `ath9k debug=0x00000224` instead of just `ath9k` into `/etc/modules.d/ath9k` to get `ATH_DBG_EEPROM | ATH_DBG_CONFIG | ATH_DBG_REGULATORY` flags enabled to print out `ath_dbg()` interesting messages from ath9k driver  
 * I even added others debug prints at different places into the source code of the ath9k, to track down what was done to my expected Tx-Power value  
 
### What I saw:  

 * the 17 dBm limitation is only applied according to the ART partition content - and nothing else  
 * no variable is modifying the Tx-Power I selected (unless it is above the maximum "shown as available" value by the driver), before it is passed to the final Tx-Power set function 
 * what I mean is: offsetting variables like `ant_gain` and `ant_reduction` values are 0  
 * reduction is applied at the very last step of the Tx-Power setting, from the `ar9003_eeprom.c` file, when the `ath9k_hw_ar9300_set_txpower()` function is calling the `ar9003_hw_set_power_per_rate_table()` function, which acts accordingly to the content of the ART partition.  

#### More details:

Depending on the value of `cfgCtl` parameter (containing information about the current DFS region and current WiFi mode), the according ART partition content limiting values are applied to the values of the `targetPowerValT2` table by `ar9003_hw_set_power_per_rate_table()`, before it is sent to the hardware.  

In my case, `reg->region` being set to `NL80211_DFS_ETSI` and `cfgCtl` being set accordingly, the maximum power level into the table end up being "25" for ETSI. Sending these values to the hardware gives a 17 dBm Tx-Power at the end, which seems to be how the hardware is interpreting the "25" value.  

This "25" value for example seems to be what I actually see having dumped my ART partitions to (kinda) human readable text [C46E1FEAF5C6-mtd5-art.txt](https://pix-server-sorel.luoss.fr/Manual/openwrt-debug/archer-c5/C46E1FEAF5C6-mtd5-art.txt) and [30B5C27F4616-mtd5-art.txt](https://pix-server-sorel.luoss.fr/Manual/openwrt-debug/archer-c5/30B5C27F4616-mtd5-art.txt). 

Having maximum value of "36" into the `targetPowerValT2` table for example (when `reg->region` is set to `NL80211_DFS_FCC` and channel set to 1) limits to 23 dBm of Tx-Power at the end.

brada4, 2023-11-20:
It is not shown by iw, but you can guess setting gain until power changes.