openwrt/openwrt #13998: 23.05 got runtime error

trombonist852, 2023-11-18:
### Describe the bug

An error message appears when accessing the router login interface.
[image]

### OpenWrt version

r0+27300-7fb31f2164

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

WanYou WanYouChunXiao

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

trombonist852, 2023-11-18:
> Invalid Version reported. `r0+27300-7fb31f2164` Is this from a clean repository?
 from immortalwrt at openwrt-23.05

> Invalid Release reported. `23.05-SNAPSHOT` Is this from a clean repository?

yes

Invalid Device reported. WanYou WanYouChunXiao Is this a supported model?

yes it's x86 router

brada4, 2023-11-18:
Your version tag is from some fork. You should report over there.
Something like luci-compat is missing, but you need to report there to fix their dependencies

closed 2023-11-18 by ynezz