openwrt/openwrt #14000: github-actions mislabeling issue as invalid

jmranger, 2023-11-18:
### Describe the bug

See [this log](https://github.com/openwrt/openwrt/actions/runs/6898856514/job/18769536860) from #13991.

On one side,  `target/linux/ath79/image/generic-ubnt.mk` contains
>DEVICE_VENDOR := Ubiquiti
>DEVICE_MODEL := Nanostation AC loco

so `scripts/dump-target-info.pl devices ath79/generic` returns
> ubnt_nanostation-ac-loco "Ubiquiti Nanostation AC loco"

but on the other, I believe that `/tmp/sysinfo/model` comes from the device tree, and `target/linux/ath79/dts/ar9342_ubnt_nanostation-ac-loco.dts` contains
> model = "Ubiquiti Nanostation AC loco (WA)";

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ath79/generic

### Device

Ubiquiti Nanostation AC loco

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-18:
ping @Ansuel

Ansuel, 2023-11-18:
@jmranger well I hoped who was submitting the device would match the dt value and the name... sad that this is not the case... will check with a script the inconsistency and see if I can find a solution. Thanks for confirming my fear!

brada4, 2023-11-18:
X86 gives dmi model there, just info, not supportability indicator.

Ansuel, 2023-11-21:
@brada4 made device validation optional and instead we add a tag for supported device.

closed 2023-11-21 by Ansuel