openwrt/openwrt #14003: Luci nftables page, displaying errors regarding 'iif'

a904055262, 2023-11-18:
### Describe the bug

raw rule:
iif "pppoe-wan" th dport 54321 @nh,128,32 0xc0a806e2 @nh,72,8 0x6 accept
and display:
Egress device id is pppoe-wanTransport header destination port is 54321 Network header bits 129-161 is 3232237282 Network header bits 73-81 is 6

Egress ??  Ingress？

### OpenWrt version

23.05.02

### OpenWrt release

23.05.02

### OpenWrt target/subtarget

x86_64

### Device

x86_64

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-18:
It is ingress indeed - but in which table it is located. Would make little sense having iif in output table. )and accept is missing or just in different font?

brada4, 2023-11-18:
ppoe device is dynamic, you shall use iifname. But bug is there.
Minimal example
```
chain luci_bug {
        type filter hook prerouting priority raw; policy accept;
        iif "lo"
}
```
Should be reported in ../luci though

a904055262, 2023-11-18:
> ppoe device is dynamic, you shall use iifname. But bug is there. Minimal example
> 
> ```
> chain luci_bug {
>         type filter hook prerouting priority raw; policy accept;
>         iif "lo"
> }
> ```
> 
> Should be reported in ../luci though

[image]

brada4, 2023-11-18:
Please report to github.com/openwrt/luci  ....
Your use case requires iifname,  but you can use no-op example code to demo issue over there.

closed 2023-11-18 by jow-