openwrt/openwrt #14004: Bananapi BPI-R3 SFP WOL not working

vrvrvrvrvr, 2023-11-19:
### Describe the bug

WOL is not working when using SFP interface for a LAN.
There is no LAN link over SFP in shutdown state to send magic packets for PC WOL. In shutdown state lights on endpoint device NIC are not blinking.
Switching back to LAN interface WOL works - which indicates issue on router / SFP interface side.
SFP module is TP-LINK TL-SM410U
Openwrt version 23.05.2 stable
Interface configs are 1:1 to match the ones of LAN.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Expected behaviour

WOL should work over SFP LAN interface (?)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-19:
SFP with WoL is almost non-existant.

vrvrvrvrvr, 2023-11-19:
Thanks! Didn't know that this is SFP hardware module specific feature. Would be very helpful if somebody could recommend one with WoL support. Not an easy way to find by googling..

brada4, 2023-11-19:
Such netcards barely exist. Certainly Intel says none supports WoL. Would be extreme to have 10W idle power and still call it sleep.

closed 2023-11-19 by vrvrvrvrvr

brada4, 2023-11-19:
Only one card aqn100
https://www.marvell.com/content/dam/marvell/en/public-collateral/ethernet-adaptersandcontrollers/marvell-ethernet-adapters-aqn100-product-brief.pdf

vrvrvrvrvr, 2023-11-19:
> Only one card aqn100 https://www.marvell.com/content/dam/marvell/en/public-collateral/ethernet-adaptersandcontrollers/marvell-ethernet-adapters-aqn100-product-brief.pdf

Thanks, but this SFP would not be supported by BPI R3 as only 2.5G modules are supported

brada4, 2023-11-19:
Any SFP module will work plugged in that card, but no other card or onboard port will keep laser on during sleep.

vrvrvrvrvr, 2023-11-19:
> Any SFP module will work plugged in that card, but no other card or onboard port will keep laser on during sleep.

what laser? I need copper output, not fiber.

brada4, 2023-11-19:
that is npt powered either.