openwrt/openwrt #14010: daemon.notice netifd: radio1 (1902): WARNING: Variable 'data' does not exist or is not an array/object

NilsRo, 2023-11-19:
### Describe the bug

Installed a fresh OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-23.306.39416-c86c256
and when enabled Wifi without changing something (and also later fully configured) I see netif errors in the log:

**daemon.notice netifd: radio1 (1902): WARNING: Variable 'data' does not exist or is not an array/object
daemon.notice netifd: radio0 (1901): WARNING: Variable 'data' does not exist or is not an array/object**

[Wax206.log](https://github.com/openwrt/openwrt/files/13404519/Wax206.log)

It happens to Netgear R7800, Zyxel ZyXEL NBG6817 and Wax206 devices. Does see that also in some logs posted for other devices here in github so it seems widely spread topic.

Simple set channel and mapped to 'guest' network for testing:

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/18000000.wmac'
        option channel '4'
        option band '2g'
        option htmode 'HT20'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'guest'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel '36'
        option band '5g'
        option htmode 'HE80'
        option cell_density '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'guest'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

This error happens to all my devices after activating Wifi. It seems not breaking Wifi but looks like an error as something should be set but is not correclty.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Netgear WAX206 (but also R7800 and NBG3817

### Image kind

Official downloaded image

### Steps to reproduce

Activate wifi

### Actual behaviour

New error messages that are not logged in earlier releases

### Expected behaviour

no error in netif

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

NilsRo, 2023-11-19:
Yes, changed device and added additional device in notes.

NilsRo, 2023-11-19:
> Invalid Device reported. `Netgear WAX206 (but also R7800 and NBG3817` Is this a supported model?

Yes, changed device and added additional device in notes.

brada4, 2023-11-19:
What is in netifd configuration at /etc/config/network ?

edrikk, 2023-11-19:
The same issue on Xiaomi Redmi Router AX6S (mediatek/mt7622).   Seen on (re)boot of the router.

```
Thu Nov 16 06:38:29 2023 daemon.notice netifd: radio0 (1805): WARNING: Variable 'data' does not exist or is not an array/object
Thu Nov 16 06:38:29 2023 daemon.notice netifd: radio1 (1807): WARNING: Variable 'data' does not exist or is not an array/object
```

brada4, 2023-11-20:
And a copy config/network conf file? The error primes up very early as if from that one.

edrikk, 2023-11-20:
Sure, here is mine.  Nothing big here, I have 3 AX6S serving as APs.
Nothing connected to WAN of course, and IPv4 network only:

```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd43:0b0c:09f8::/48'
        option packet_steering '1'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        option ipv6 '0'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        list dns '192.168.41.1'
        option ipaddr '192.168.41.3'
        option netmask '255.255.252.0'
        option gateway '192.168.41.1'

config device
        option name 'wan'
        option macaddr '5c:XX:XX:XX:XX:96'

config interface 'wan'
        option device 'wan'
        option proto 'dhcp'

config interface 'wan6'
        option device 'wan'
        option proto 'dhcpv6'

```

edrikk, 2023-11-20:
BTW, this is not new...  It's been going on for a while.  Here are some examples.  Nobody seems to have solved it...

[Forum post  - April 2023](https://forum.openwrt.org/t/problems-with-client-mode-on-snapshot-netifd-shows-lots-of-errors/156762)
[Forum post - Sept 2023](https://forum.openwrt.org/t/trouble-with-wifi-on-belkin-rt3200-23-05-0-rc4/173115)
[Forum post - Jan 2023](https://forum.openwrt.org/t/xiaomi-router-4a-100m-r4ac-wireless-issues-phy-is-undefined-for-device-radio0/148854)
[Forum post - July 2022](https://forum.openwrt.org/t/radio-phys-not-found-for-mt7622-since-recent-commits/132401/3)
[Forum post - Jan 2022](https://forum.openwrt.org/t/r6220-could-not-find-phy-for-device-radio0-mt76x2-issue-continues/116665)
[Forum post - July 2019](https://forum.openwrt.org/t/could-not-find-phy-for-device-radio0/41792/17)
etc.

[Issue 601](https://github.com/openwrt/mt76/issues/601)
[Issue 11725](https://github.com/openwrt/openwrt/issues/11725)
[Issue 13815](https://github.com/openwrt/openwrt/issues/13815)
[Issue 13701](https://github.com/openwrt/openwrt/issues/13701)

etc.

NilsRo, 2023-11-20:
Attached a new log and network config. It is shown if wpad is restating.

Also some other errors are there:
Mon Nov 20 08:25:13 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Mon Nov 20 08:25:13 2023 daemon.warn netifd: Wireless device 'radio0' setup failed, retry=3
Mon Nov 20 08:25:13 2023 daemon.warn netifd: Wireless device 'radio1' setup failed, retry=3

Mon Nov 20 08:25:42 2023 daemon.notice netifd: lan (2283): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "lan" } (Permission denied)

Perhaps not related but do not know....
[Wax206_2.log](https://github.com/openwrt/openwrt/files/13407665/Wax206_2.log)
[network.txt](https://github.com/openwrt/openwrt/files/13407666/network.txt)

NilsRo, 2023-11-20:
Also present in newest snapshot...

brada4, 2023-11-20:
may not be related, but still unpleaseant. 
Please rename /etc/capabilities/wpad.json and restart wpad to run without ujail sandbox. It should allow access to all files also those not mentioned in jail.

NilsRo, 2023-11-20:
Renamed...but same errors. ubus is not there but perhaps this is fixed in snapshot.

root@OpenWrt6:/etc/capabilities# ls
ntpd.json      wpad.json.bak

**Log**
Mon Nov 20 21:13:51 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Mon Nov 20 21:13:51 2023 daemon.warn netifd: Wireless device 'radio1' setup failed, retry=3
Mon Nov 20 21:13:51 2023 daemon.warn netifd: Wireless device 'radio0' setup failed, retry=3
Mon Nov 20 21:13:51 2023 daemon.notice hostapd: Set new config for phy wl0:
Mon Nov 20 21:13:51 2023 daemon.notice hostapd: Set new config for phy wl1:
Mon Nov 20 21:13:51 2023 daemon.notice wpa_supplicant[6771]: Set new config for phy wl0
Mon Nov 20 21:13:51 2023 daemon.notice wpa_supplicant[6771]: Set new config for phy wl1
Mon Nov 20 21:13:51 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Mon Nov 20 21:13:51 2023 daemon.notice netifd: Wireless device 'radio1' is now down
Mon Nov 20 21:13:51 2023 daemon.notice netifd: radio0 (6852): WARNING: Variable 'data' does not exist or is not an array/object
Mon Nov 20 21:13:51 2023 daemon.notice netifd: radio1 (6858): WARNING: Variable 'data' does not exist or is not an array/object
Mon Nov 20 21:13:51 2023 daemon.notice hostapd: Set new config for phy wl0:
Mon Nov 20 21:13:51 2023 daemon.notice hostapd: Set new config for phy wl1:

root@OpenWrt6:/var/run# ls -l
-rw-r--r--    1 root     root           572 Nov 20 08:34 config.md5
drwxr-xr-x    2 dnsmasq  dnsmasq         40 Nov 20 21:13 dnsmasq
-rw-r--r--    1 root     root             5 Nov 20 08:34 dropbear.1.pid
-rw-r--r--    1 root     root             0 Nov 20 21:13 fw4.lock
drwxr-xr-x    2 network  network        100 Nov 20 21:13 hostapd
-rw-r--r--    1 root     root           762 Nov 20 21:13 hostapd-wl0.conf
-rw-r--r--    1 root     root           762 Nov 20 21:12 hostapd-wl0.conf.prev
-rw-r--r--    1 root     root          1898 Nov 20 21:13 hostapd-wl1.conf
-rw-r--r--    1 root     root          1898 Nov 20 21:12 hostapd-wl1.conf.prev
-rw-r--r--    1 root     root            17 Nov 20 08:34 led.state
-rw-r--r--    1 root     root             4 Nov 20 08:34 logread.2.pid
drwx------    6 root     root           120 Nov 20 21:12 rpcd
drwxr-xr-x    2 ubus     ubus            60 Jan  1  1970 ubus
-rw-r--r--    1 root     root             5 Nov 20 08:34 udhcpc-br-lan.1.pid
-rw-r--r--    1 root     root             3 Nov 20 21:13 wdev-wl0.json
-rw-r--r--    1 root     root             3 Nov 20 21:13 wdev-wl1.json
drwxr-xr-x    2 network  network         60 Nov 20 21:13 wpa_supplicant

brada4, 2023-11-20:
The error text not an array/object is from ubus libraries.....
Why your phy-s are called wlX ? Is it OpenWRT?

NilsRo, 2023-11-20:
Is standard with OpenWrt 23.05.2 and my Wax206.

brada4, 2023-11-20:
do you get array/object warning with sandbox disabled ie json renamed?
you said still array/object at boot?

NilsRo, 2023-11-21:
The permission denied error is gone with disabled sandbox only. All other errors are still there also after reboot also array/object.

brada4, 2023-11-21:
(not the reported problem, the other one)

Add rw mount for sandbox process to delete control directory

Edit /etc/init.d/wpad adding new line (without PLUS sign)

```diff
                [ -x /sbin/ujail -a -e /etc/capabilities/wpad.json ] && {
                        procd_add_jail hostapd
+                        procd_add_jail_mount_rw /var/run/hostapd       
```

You can make equivalent change regarding wpa_supplicant below. Does not affect your setup.
Then move json back and try few restarts and confirm this error is gone.

I am puzzled it does not happen for me and i frankly dont see any overcustomisation on your part either.

edrikk, 2023-11-21:
Out of curiosity, what device(s) are you using @brada4?  

I rebooted my main router (Xiaomi AX3600 - ipq807) today, and noticed that it also popped the same message (I never paid attention before).

It seems it’s a general message, which may not actually cause any issues, and maybe should just be silenced.

But it is interesting that you don’t get it.

brada4, 2023-11-22:
I was stopping starting one ap interface. Particular logg text is hostapd removing directory with control sockets on own shutdown.
re200 if thst mstters.

edrikk, 2023-11-22:
I just tried on my friend's device that I manage for him (ramips/mt7621 (xiaomi,redmi-router-ac2100).  Running SNAPSHOT r24427-c4fe1bfc65, shows the same message as well:

`Sun Nov 19 18:25:08 2023 daemon.notice netifd: radio1 (2081): WARNING: Variable 'data' does not exist or is not an array/object`

Message only showing for radio1 as he has disabled radio0.

brada4, 2023-11-22:
the message is non-specific, null response from ubus.
you dont set country for wifi-s? But rest is quite default.

NilsRo, 2023-11-22:
> the message is non-specific, null response from ubus. you dont set country for wifi-s? But rest is quite default.

New spare testdevice....my original configuration which shows also the issue has everything set.

brada4, 2023-11-22:
Can you try one interface with auto channel, i cannot trigger error still BUT my wifis join bridge 15s after (5 without auto channel) br-lan is set up.

NilsRo, 2023-11-27:
Same behavior also with AUTO channel.

brada4, 2023-11-27:
You can try recording 'ubus monitor | tee -a /tmp/ubus.trace' if you manage to trigger error while at ssh.
Say disable wifi, restart, ssh in one session and start ubus log, try to bring interfaces up etc. The error message just says ubus returned empty message without saying what the request was.

NilsRo, 2023-11-27:
Restarting Wpad does also trigger the error so no restart is neccessary. Attached the ubus log. I cannot find any Null, Empty, etc. messages...but also does not know for what I have to look.
[ubus.txt](https://github.com/openwrt/openwrt/files/13481345/ubus.txt)

brada4, 2023-11-27:
:-S log off luci and try to restart wpad service instead, it like lists opkg packages 500 times in the mega log

NilsRo, 2023-11-28:
Actually recorded wpad start only. Also a new error message in the log but for your information only.
```
Tue Nov 28 23:33:46 2023 daemon.notice hostapd: wl1-ap0: AP-DISABLED
Tue Nov 28 23:33:46 2023 daemon.notice hostapd: wl1-ap0: CTRL-EVENT-TERMINATING
Tue Nov 28 23:33:47 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Tue Nov 28 23:33:47 2023 daemon.notice hostapd: nl80211: deinit ifname=wl1-ap0 disabled_11b_rates=0
Tue Nov 28 23:33:47 2023 kern.info kernel: [783707.226626] device wl1-ap0 left promiscuous mode
Tue Nov 28 23:33:47 2023 kern.info kernel: [783707.231471] br-lan: port 9(wl1-ap0) entered disabled state
Tue Nov 28 23:33:47 2023 daemon.notice netifd: Network device 'wl1-ap0' link is down
Tue Nov 28 23:33:47 2023 daemon.notice netifd: lan (2458): udhcpc: sending renew to server 192.168.177.1
Tue Nov 28 23:33:47 2023 daemon.notice netifd: lan (2458): udhcpc: lease of 192.168.177.53 obtained from 192.168.177.1, lease time 86400
Tue Nov 28 23:33:47 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Tue Nov 28 23:34:08 2023 daemon.err uhttpd[1632]: [info] luci: accepted login on / for root from 192.168.177.22
Tue Nov 28 23:34:15 2023 daemon.notice netifd: radio0 (14399): Command failed: ubus call hostapd config_set { "phy": "wl0", "config": "", "prev_config": "/var/run/hostapd-wl0.conf" } (Request timed out)
Tue Nov 28 23:34:15 2023 daemon.notice netifd: radio0 (14399): Command failed: Not found
Tue Nov 28 23:34:15 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Tue Nov 28 23:34:15 2023 daemon.notice netifd: radio0 (14584): WARNING: Variable 'data' does not exist or is not an array/object
Tue Nov 28 23:34:15 2023 daemon.notice netifd: radio1 (14411): Command failed: ubus call hostapd config_set { "phy": "wl1", "config": "", "prev_config": "/var/run/hostapd-wl1.conf" } (Request timed out)
Tue Nov 28 23:34:15 2023 daemon.notice netifd: radio1 (14411): Command failed: Not found
Tue Nov 28 23:34:15 2023 daemon.notice netifd: radio0 (14584): Command failed: Not found
Tue Nov 28 23:34:15 2023 daemon.notice netifd: radio0 (14584): Command failed: Not found
Tue Nov 28 23:34:15 2023 daemon.notice netifd: Wireless device 'radio1' is now down
Tue Nov 28 23:34:15 2023 daemon.notice netifd: radio1 (14609): WARNING: Variable 'data' does not exist or is not an array/object
Tue Nov 28 23:34:15 2023 daemon.notice netifd: radio1 (14609): Command failed: Not found
Tue Nov 28 23:34:15 2023 daemon.notice netifd: radio1 (14609): Command failed: Not found
Tue Nov 28 23:34:36 2023 daemon.notice wpa_supplicant[14806]: Successfully initialized wpa_supplicant
Tue Nov 28 23:34:36 2023 daemon.notice hostapd: Set new config for phy wl1: /var/run/hostapd-wl1.conf
Tue Nov 28 23:34:36 2023 daemon.notice hostapd: Restart interface for phy wl1
Tue Nov 28 23:34:36 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=DE ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=36 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935] ieee80211ac=1 vht_oper_chwidth=2 vht_oper_centr_freq_seg0_idx=50 vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][MAX-MPDU-7991][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=2 he_oper_centr_freq_seg0_idx=50 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_bss_color=5 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu_edca_qos_info_txop_request
Tue Nov 28 23:34:36 2023 kern.info kernel: [783756.251255] br-lan: port 6(wl1-ap0) entered blocking state
Tue Nov 28 23:34:36 2023 kern.info kernel: [783756.256878] br-lan: port 6(wl1-ap0) entered disabled state
```

[ubus (2).txt](https://github.com/openwrt/openwrt/files/13494053/ubus.2.txt)

brada4, 2023-11-28:
thanks, will follow up on WE

rany2, 2023-11-30:
See https://github.com/openwrt/openwrt/commit/9b56c27a8a08a5c40d38064b9046f42177c94768#commitcomment-124323371. It is a harmless warning.

brada4, 2023-11-30:
@rany2 lemme try 2 find source.

AI-rrr, 2024-01-29:
still there, get it with last snapshot

idcrisis, 2024-03-25:
Archer C20 V5 ( snapshot build )

Two problems: 

- After sysupgrade ( -n ) Wifi ( 2.4 ghz ) doesn't come up ( it comes up in the router but is not visible on the network )

- 5 Ghz doesn't show up at all in the router. 

First problem solved with a hard reset, soft reboot doesn't work. 

Both problems solved with upgrading to wpad-mesh-mbedtls.

Boczek7, 2024-04-21:
@NilsRo 

> Tue Nov 28 23:34:15 2023 daemon.notice netifd: radio0 (14584): Command failed: Not found

You may not have netifd utility installed.

professor-jonny, 2024-05-05:
I get this error on my Wallystech dr4029

wryun, 2024-08-28:
For clarity, my assumption is this is coming from mac80211.sh:
```
        json_select data && {                       
                json_get_var prev_rxantenna rxantenna
                json_get_var prev_txantenna txantenna                                     
                json_select ..        
        }  
```

this was added to OpenWrt 23 when the 'reload' behaviour was improved in 9720b094aef89802327683f25824820581fed0b9

Basically, using json_select in this manner (to detect whether data exists) will always issue a warning when data doesn't exist. i.e. the warning is nothing to worry about... which means the code should be changed to not generate it. I think this is best done via a json_is_a test. i.e.

```
json_is_a "$data" object && {
    json_select data
    ...
}
```

santutti, 2025-04-14:
Hello, it happened to me in my MF286d (QCA4019), with 24.10

"Sun Apr 13 10:29:00 2025 daemon.notice netifd: radio1 (6104): WARNING: Variable 'data' does not exist or is not an array/object"

has a workaround been found yet? 
Thank you!