openwrt/openwrt #14013: LuCI extremely slow on Raspberry PI 3 or 4

schuettecarsten, 2023-11-20:
### Describe the bug

LuCI is really slow on Raspberry Pi 3 or 4. The system itself works fine, routing is fine, no significant CPU usage during normal operation. No errors in logs. But when using LUCI, it takes seconds before pages are shown. CPU stays at <5%, but there seems to be a delay on each request when LUCI tries to generate page. This issue occurs on a Raspberry Pi 3 and 4 using the latest images. Downloaded and installed on a fast SD card and the error is there. So it's easy to reproduce. No idea what's going on there, but also no idea how to debug this.

### OpenWrt version

r24414-255d5c9bf8

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.5

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

schuettecarsten, 2023-11-20:
Please remove the "invalid" label, as RPi4 Model B is a supported device according to https://openwrt.org/toh/raspberry_pi_foundation/raspberry_pi

jow-, 2023-11-20:
Please try executing `/www/cgi-bin/luci` on the cli and see if you can notice an unusual delay there too. If yes, proceed with strace to see where it runs. You can also use `ucode -t /www/cgi-bin/luci` to get a detailed opcode trace (might be very verbose)

jow-, 2023-11-20:
Also please try checking your browser debuggers network tab to see which kind of requests are hanging.

From the screenshot above it looks like the LuCI interpreter itself works in the expected speed but that async requests via Ubus-HTTP gateway are stalling.
It would be interesting to know if the installation currently uses the accelerated gateway (http://.../ubus) or the fallback one (http://.../cgi-bin/luci/ubus).

schuettecarsten, 2023-11-20:
Running the luci command on the CLI generates an HTML response instantly.
This is how the timeline for the **status** page looks like:

[image]
[image]

The **network** page take longer to generate the first response:

[image]

When I try to run the ubus commands from CLI, I can see that `ubus call luci-rpc getWirelessDevices` and another ubus request that asks for config:wireless seem to cause the delays. Unfortunately, I have no idea how to run this command from CLI:

[image]

If I delete my wireless device from config, LuCI works fine. When I re-add the wifi AP, it slows down again.

0xFelix, 2023-11-20:
I noticed that installing `irqbalance` significantly speeds up the load time of LuCI on the Redmi AX6000. It may help on the RPI too?

leobsky, 2023-11-22:
Issues on snapshot started with the backport 6.5 update, since then wifi in the rpi is unusable.
 enabling wifi spam this logs:
```
[  357.742359] net_ratelimit: 30 callbacks suppressed
[  357.742380] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[  357.862331] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[  357.982297] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  358.102300] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  358.222267] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  358.229784] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  358.237266] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  358.244847] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
[  358.252288] brcmfmac: brcmf_set_channel: set chanspec 0xd064 fail, reason -52
[  358.259704] brcmfmac: brcmf_set_channel: set chanspec 0xd068 fail, reason -52
[  362.831110] net_ratelimit: 34 callbacks suppressed
[  362.831130] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  362.843463] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  362.850993] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  362.858416] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
[  362.865808] brcmfmac: brcmf_set_channel: set chanspec 0xd064 fail, reason -52
[  362.873217] brcmfmac: brcmf_set_channel: set chanspec 0xd068 fail, reason -52
[  362.880611] brcmfmac: brcmf_set_channel: set chanspec 0xd06c fail, reason -52
[  362.887999] brcmfmac: brcmf_set_channel: set chanspec 0xd070 fail, reason -52
[  362.895391] brcmfmac: brcmf_set_channel: set chanspec 0xd074 fail, reason -52
[  362.902782] brcmfmac: brcmf_set_channel: set chanspec 0xd078 fail, reason -52
[  369.499493] net_ratelimit: 46 callbacks suppressed
[  369.499512] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[  369.619489] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[  369.739456] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  369.859405] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  369.979416] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  369.986930] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  369.994414] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  370.001932] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
[  370.009361] brcmfmac: brcmf_set_channel: set chanspec 0xd064 fail, reason -52
[  370.016761] brcmfmac: brcmf_set_channel: set chanspec 0xd068 fail, reason -52
[  374.578351] net_ratelimit: 33 callbacks suppressed
[  374.578371] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  374.698330] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  374.705843] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  374.713328] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  374.720832] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
[  374.728244] brcmfmac: brcmf_set_channel: set chanspec 0xd064 fail, reason -52
[  374.735645] brcmfmac: brcmf_set_channel: set chanspec 0xd068 fail, reason -52
[  374.743039] brcmfmac: brcmf_set_channel: set chanspec 0xd06c fail, reason -52
[  374.750437] brcmfmac: brcmf_set_channel: set chanspec 0xd070 fail, reason -52
[  374.757825] brcmfmac: brcmf_set_channel: set chanspec 0xd074 fail, reason -52
[  381.097104] net_ratelimit: 47 callbacks suppressed
[  381.097125] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[  381.217069] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[  381.336899] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  381.456998] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  381.576999] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  381.584509] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  381.591994] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  381.599502] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
[  381.606921] brcmfmac: brcmf_set_channel: set chanspec 0xd064 fail, reason -52
[  381.614325] brcmfmac: brcmf_set_channel: set chanspec 0xd068 fail, reason -52
[  386.723081] net_ratelimit: 10 callbacks suppressed
[  386.723099] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[  386.843066] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[  386.966103] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  387.086135] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  387.206092] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  387.213594] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  387.221066] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  387.228567] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
[  387.235988] brcmfmac: brcmf_set_channel: set chanspec 0xd064 fail, reason -52
[  387.243392] brcmfmac: brcmf_set_channel: set chanspec 0xd068 fail, reason -52
[  393.685048] net_ratelimit: 50 callbacks suppressed
[... 19 lines cut ...]
[  398.758498] brcmfmac: brcmf_set_channel: set chanspec 0xd074 fail, reason -52
[... cut ...]
[  412.302485] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  412.422484] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  412.430030] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  412.437550] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  412.445144] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
[  412.452599] brcmfmac: brcmf_set_channel: set chanspec 0xd064 fail, reason -52
[  412.460014] brcmfmac: brcmf_set_channel: set chanspec 0xd068 fail, reason -52
```

hauke, 2023-11-26:
There is a discussion about the problem in brcmfmac on the Linux wireless mailing list, see https://lore.kernel.org/all/5c462fac-b27d-41c0-a62c-a8951bf445d2@gmx.net/T/

ptpt52, 2023-12-09:
mark to follow

ptpt52, 2023-12-09:
```
[    7.199688] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.208780] usbcore: registered new interface driver brcmfmac
[    7.214699] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin failed with error -2
[    7.226440] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin
[    7.470356] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    7.478382] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jul 29 2022 02:15:20 version 7.45.250 (70e9766 CY) FWID 01-e53e306b
[   12.870410] brcmfmac mmc1:0001:1 phy0-sta0: renamed from wlan0
[  145.710973] brcmfmac: brcmf_dump_obss: dump_obss error (-52)
[  146.592029] brcmfmac: brcmf_dump_obss: dump_obss error (-52)
[  148.253687] brcmfmac: brcmf_dump_obss: dump_obss error (-52)
[  148.890904] brcmfmac: brcmf_dump_obss: dump_obss error (-52)
[  149.369528] brcmfmac: brcmf_dump_obss: dump_obss error (-52)
[  154.570815] brcmfmac mmc1:0001:1 wlan0: renamed from phy0-sta0
[  739.859611] brcmfmac mmc1:0001:1 phy0-sta0: renamed from wlan0
```
AP mode get hang
STA mode works fine.

ptpt52, 2023-12-09:
any idea to fix the hang issue on AP mode for `brcmfmac`
@nbd168 @hauke

Noltari, 2024-01-26:
I've reproduced this on my Raspberry Pi 5 (same wifi as Raspberry Pi 4).
The issue is coming from iwinfo commands taking a lot to complete, not from LuCI

Raspberry Pi 5 running main/master:
```
root@rpi5:~# time iwinfo phy0-ap0 assoclist
No station connected
real    0m 2.65s
user    0m 0.00s
sys     0m 0.00s
```
```
root@rpi5:~# time iwinfo phy0-ap0 info
phy0-ap0  ESSID: "OpenWrt"
          Access Point: DA:3A:DD:XX:XX:XX
          Mode: Master  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 31 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -88 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ac/b/g/n
          Hardware: 02D0:4345 0000:0000 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: no  PHY name: phy0
real    0m 2.21s
user    0m 0.00s
sys     0m 0.00s
```

Raspberry Pi 4 running 23.05.2:
```
root@rpi4:~# time iwinfo phy0-ap0 assoclist
24:A1:60:XX:XX:XX  unknown / unknown (SNR 0)  2000 ms ago
        RX: 36.0 MBit/s                                16966 Pkts.
        TX: 72.2 MBit/s                              2092110 Pkts.
        expected throughput: unknown

24:A1:60:XX:XX:XX  unknown / unknown (SNR 0)  1000 ms ago
        RX: 18.0 MBit/s                                27948 Pkts.
        TX: 14.4 MBit/s                              2102572 Pkts.
        expected throughput: unknown

24:A1:60:XX:XX:XX  unknown / unknown (SNR 0)  7000 ms ago
        RX: 12.0 MBit/s                                48159 Pkts.
        TX: 14.4 MBit/s                              2119406 Pkts.
        expected throughput: unknown

70:03:9F:XX:XX:XX  unknown / unknown (SNR 0)  0 ms ago
        RX: 36.0 MBit/s                                53186 Pkts.
        TX: 72.2 MBit/s                              2111846 Pkts.
        expected throughput: unknown

real    0m 0.01s
user    0m 0.00s
sys     0m 0.00s
```
```
root@rpi4:~# time iwinfo phy0-ap0 info
phy0-ap0  ESSID: "XXX"
          Access Point: DE:A6:32:XX:XX:XX
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 31 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: 43.2 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/b/g/n
          Hardware: 02D0:A9A6 0000:0000 [Cypress CYW43455]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: no  PHY name: phy0
real    0m 0.04s
user    0m 0.01s
sys     0m 0.00s
```

Noltari, 2024-01-26:
While debugging the issue I've found out that it always hangs on `nl80211_get_noise`:
https://github.com/openwrt/iwinfo/blob/a34977c0760c93480491c8eb94da656b57d7f4cc/iwinfo_nl80211.c#L1693-L1694

However, sometimes it also hangs at the beginning on `nl80211_ifname2phy`:
https://github.com/openwrt/iwinfo/blob/a34977c0760c93480491c8eb94da656b57d7f4cc/iwinfo_nl80211.c#L779-L780

Noltari, 2024-01-26:
This is fixed by https://github.com/openwrt/openwrt/pull/14491/commits/42ccd5349ef5bc05b4dd0626c888a91362117d22

closed 2024-01-26 by openwrt-bot

reddy-quan, 2024-10-17:
Yes, now is 20241017, I am using the latest openwrt fw, it is so slow slow slow on my pi4.  I think this problem still exist!!!
Hostname	OpenWrt
Model	Raspberry Pi 4 Model B Rev 1.1
Architecture	ARMv8 Processor rev 3
Target Platform	bcm27xx/bcm2711
Firmware Version	OpenWrt 23.05.5 r24106-10cc5fcd00 / LuCI openwrt-23.05 branch git-24.264.56413-c7a3562

chylex, 2024-12-06:
Running into the same problem on a Pi 4, the web interface is constantly timing out... is there any way to increase the timeout? I would rather wait longer than have to retry almost every single action... this is almost unusable.