openwrt/openwrt #14019: filogic: WED bug and kernel  under heavy load

patrykk, 2023-11-20:
### Describe the bug

We get the kernel Ops with enabled WED ( wifi offloading). During heavy load on the 5GHz wifi we can get random kernel panic like below. Used option: options mt7915e wed_enable=Y. 

Below kernel log from console:
[ 1915.546516] ieee80211 phy0: WA: free done event
[ 1915.546516] 1022d640
[ 1915.546516] len = 20
[ 1915.546516] DW0 : 14 00 01 30
[ 1915.546516] DW1 : 01 e5 94 52
[ 1915.546516] DW2 : 00 c0 87 89
[ 1915.546516] DW3 : 72 00 00 41
[ 1915.546516] DW4 : 2d 84 ff 3f
[ 1964.918808] ieee80211 phy0: WA: free done event
[ 1964.918808] 1022dc80
[ 1964.918808] len = 112
[ 1964.918808] DW0 : 70 00 23 30
[ 1964.918808] DW1 : 05 0c 34 0c
[ 1964.918808] DW2 : 00 80 02 89
[ 1964.918808] DW3 : 04 20 00 41
[ 1964.918808] DW4 : bc 83 de 01
[ 1964.918808] DW5 : be 83 df 01
[ 1964.918808] DW6 : c0 83 e0 01
[ 1964.918808] DW7 : c2 83 ff 3f
[ 1964.918808] DW8 : 04 20 00 41
[ 1964.918808] DW9 : c3 03 e2 01
[ 1964.918808] DW10 : c5 03 e3 01
[ 1964.918808] DW11 : c7 03 e4 01
[ 1964.918808] DW12 : c9 83 ff 3f
[ 1964.918808] DW13 : 04 20 00 41
[ 1964.918808] DW14 : ca 83 e5 01
[ 1964.918808] DW15 : cc 83 e6 01
[ 1964.975482] ieee80211 phy0: WA: DW16 : ce 83 e7 01
[ 1964.975482] DW17 : d0 83 ff 3f
[ 1964.975482] DW18 : 04 20 00 41
[ 1964.975482] DW19 : d1 03 e9 01
[ 1964.975482] DW20 : d3 03 ea 01
[ 1964.975482] DW21 : d5 03 eb 01
[ 1964.975482] DW22 : d7 83 ff 3f
[ 1964.975482] DW23 : 04 20 00 41
[ 1964.975482] DW24 : d8 83 ec 01
[ 1964.975482] DW25 : da 83 1a 02
[ 1964.975482] DW26 : 36 84 1b 02
[ 1964.975482] DW27 : 38 84 ff 3f
[ 2131.716404] Unable to handle kernel paging request at virtual address ffffffffc0000000
[ 2131.724343] Mem abort info:
[ 2131.727132]   ESR = 0x0000000096000145
[ 2131.730869]   EC = 0x25: DABT (current EL), IL = 32 bits
[ 2131.736190]   SET = 0, FnV = 0
[ 2131.739239]   EA = 0, S1PTW = 0
[ 2131.742379]   FSC = 0x05: level 1 translation fault
[ 2131.747245] Data abort info:
[ 2131.750111]   ISV = 0, ISS = 0x00000145
[ 2131.753950]   CM = 1, WnR = 1
[ 2131.756914] swapper pgtable: 4k pages, 39-bit VAs, pgdp=0000000048a80000
[ 2131.763608] [ffffffffc0000000] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[ 2131.772309] Internal error: Oops: 96000145 [#1] SMP
[ 2131.777178] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgenc
[ 2131.777335]  arc4 cbc encrypted_keys uas usb_storage uhci_hcd ohci_platform ohci_hcd dm_mirror dm_region_hash dm_log dm_mod dax trusted tpm oid_registry asn1_encoder asn1_decoder leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gn
[ 2131.888403] CPU: 2 PID: 1029 Comm: napi/phy0-10 Not tainted 5.15.134 #0
[ 2131.894998] Hardware name: ASUS TUF-AX6000 (DT)
[ 2131.899511] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 2131.906452] pc : dcache_clean_poc+0x1c/0x34
[ 2131.910629] lr : arch_sync_dma_for_device+0x1c/0x24
[ 2131.915489] sp : ffffffc00948bc20
[ 2131.918786] x29: ffffffc00948bc20 x28: ffffffc009427000 x27: 0000000000004710
[ 2131.925904] x26: ffffff8004150080 x25: 0000000000000000 x24: ffffffc00942b710
[ 2131.933022] x23: 0000000045148800 x22: 0000000000000002 x21: ffffff8000217c10
[ 2131.940139] x20: 00000000000006c0 x19: 0000000000000000 x18: 0000000000000000
[ 2131.947255] x17: 0000000000001c80 x16: ffffffc009403000 x15: 0000000000000e40
[... 391 lines cut ...]
r24114-b7f1459fef

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

ASUS TUF-AX6000

### Image kind

Self-built image

### Actual behaviour

Kernel Ops with Self-build and official image.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

zhaojh329, 2023-11-21:
It will occur soon if run iperf3 with 5G and 2.4G concurrent.

brada4, 2023-11-21:
wed is off on most hardware capable of enabling it by means of dts. Just get back to stable default?

patrykk, 2023-11-21:
We can simple disable WED by just removing module parameter but nevertheless there is a bug which have to be investigated. 
@brada4 Maybe you have more this kind of "advices" and just disable function when BUG you find ;)

brada4, 2023-11-22:
WED will never obey bonding failover or balancing. It is device to device DMA. 
 BTW your netcard MAC is not read from flash.

zhaojh329, 2023-12-01:
https://github.com/openwrt/mt76/issues/830

The issue is bound to occur with `wed` enabled follow these steps.

```
STA1 ---2g---
             \
                MT7986 -- wired LAN ----- PC
             /
STA2 ---5g---
```

Assume the IP address of PC is `192.168.1.100`

1. Run iperf3 in PC
```
iperf -s -p 9000
``` 
2. Run another iperf3 in PC
```
iperf -s -p 9001
``` 
3. Run iperf3 in STA1
```
iperf -c 192.168.1.100 -p 9000 -P 10 -t 1000
```
5. Run iperf3 in STA2
```
iperf -c 192.168.1.100 -p 9001 -P 10 -t 1000
```

brada4, 2023-12-01:
Yes, if particular SoC is not capable of full speed device-to-device DMA then WED will fail if enabled (not by default). More graceful error handling may come with time like fallback if dev2dev DMA fails, or it is tested on the go etc, but for now keep it off.

@padavanonly thanks for appreciation, i am just re-iterating hard letter in kernel flowtable offload docs.

patrykk, 2023-12-01:
Maybe some kind of solution (partial) could be introduce option to enable WED only for some band or device?

brada4, 2023-12-04:
you need to build kernel with CONFIG_HOTPLUG_CPU optiob to have offlining interface

mpratt14, 2023-12-07:
Also, this is worth trying on kernel 6.1, if you're able to build with that configured

brada4, 2023-12-07:
Easier to push nosmp or maxcpus=1 boot options.

patrykk, 2023-12-13:
Problem has been solved with commit https://github.com/openwrt/mt76/commit/f559adf1849c8af91f5a5eb670f4ed2c24988898

patrykk, 2023-12-13:
mt76 changes merged to the main branch
commit : https://github.com/openwrt/openwrt/commit/8f782ed07dc23b64bb24f63ce31d939ee6afb5ad

closed 2023-12-13 by patrykk