openwrt/openwrt #14023: CONFIG_MOLD=y is displayed twice in scripts/diffconfig.sh

hgl, 2023-11-21:
### Describe the bug

See `Steps to reproduce`

### OpenWrt version

r24415-9062e5faae

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

Default string Default string

### Image kind

Self-built image

### Steps to reproduce

Run `make defconfig && scripts/diffconfig.sh`

### Actual behaviour

two `CONFIG_MOLD=y` are displayed

### Expected behaviour

Only one CONFIG_MOLD=y is displayed

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hgl, 2023-11-21:
Not sure if the bot is being helpful here. It's just a generic x86 machine.

Ansuel, 2023-11-21:
@hgl yes made the device validation optional to handle this.