openwrt/openwrt #14024: pppoe-wan MTU setting not respected for mini-jumbo frames

ThxAndBye, 2023-11-21:
### Describe the bug

The MTU set in `/etc/config/network` for `interface 'wan'` is not respected when the interface is (re)started.

Full config for `wan`:
```
config interface 'wan'
        option device 'eth1'
        option proto 'pppoe'
        option username 'XXXX@provider.tld'
        option password 'xXxxXxx'
        option peerdns '0'
        option ipv6 '0'
        option mtu '1500'

config interface 'wan6'
        option proto 'dhcpv6'
        option device '@wan'
        option reqaddress 'try'
        option reqprefix 'auto'
        option peerdns '0'

config device
        option name 'pppoe-wan'
        option mtu '1500'
        option mtu6 '1500'
```

`ifconfig` for `pppoe-wan` interface:
```
pppoe-wan Link encap:Point-to-Point Protocol  
          inet addr:xx.xx.xx.xx  P-t-P:xx.xx.xx.xx  Mask:255.255.255.255
          inet6 addr: fe80::c420:f50f:d63c:5c4a/128 Scope:Link
          inet6 addr: 2001:db8:85a3:8d3:1319:8a2e:370:7348/64 Scope:Global
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1492  Metric:1
          RX packets:884 errors:0 dropped:0 overruns:0 frame:0
          TX packets:993 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:3 
          RX bytes:300256 (293.2 KiB)  TX bytes:150985 (147.4 KiB)
```

Using `ifconfig pppoe-wan mtu 1500 up` to set the configured MTU on pppoe-wan works and the connection also works with the mini-jumbo frames, tested via https://www.speedguide.net/analyzer.php :
```
« SpeedGuide.net TCP Analyzer Results » 
Tested on: 2023.11.21 03:53 
IP address: xx.xx.xx.xx 
Client OS/browser: Mac OS (Firefox 119.0) 
 
TCP options string: 020405b4010303060101080ad7b559380000000004020000 
MSS: 1460 
MTU: 1500 
TCP Window: 131712 (not multiple of MSS) 
RWIN Scaling: 6 bits (2^6=64) 
Unscaled RWIN : 2058 
Recommended RWINs: 64240, 128480, 256960, 513920, 1027840 
BDP limit (200ms): 527 Mbps (53 Megabytes/s) 
BDP limit (500ms): 211 Mbps (21 Megabytes/s) 
MTU Discovery: ON 
TTL: 50 
Timestamps: ON 
SACKs: ON 
IP ToS: 00000000 (0) 

```

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi Compute Module 4 Rev 1.1

### Image kind

Official downloaded image

### Steps to reproduce

Set Physical WAN interface to 1508 MTU
Set PPPoE interface to 1500 MTU

use `ifup wan` or 
`ubus call network.interface.wan down` followed by `ubus call network.interface.wan up` to restart the wan interface

MTU for PPPoE is set to 1492 instead of 1500

### Actual behaviour

`pppoe-wan` MTU gets set to 1492

### Expected behaviour

`pppoe-wan` MTU gets set to 1500

### Additional info

The CM4 is running on a DFRobot Router board. So my setup has two physical interfaces. The WAN interface is the additional RTL8111 chip attached via PCIe as it allowed to set a 1508 MTU; the internal Broadcom one refused higher than 1500 MTU.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-21:
it is limited by lower device configuration. Rise mtu there first, with ifconfig on the fly to assure it connects after reboot if something goes wrong.

ThxAndBye, 2023-11-22:
> it is limited by lower device configuration. Rise mtu there first

I've set physical interface `eth1` to 1508 MTU, the `pppoe-wan` device to 1500 and the wan also to 1500, which other devices need to be configured accordingly? 
The PPPoE connection is the last link in the chain that isn't configured correctly according to the config files. Configuring `pppoe-wan` manually does work and the connection then is operating at 1500 MTU.

brada4, 2023-11-22:
i do not see any 1508 set on eth1 in config you posted.
you can do it in gui,  or make section similar to pppoe-wan but eth1 and 1508

ThxAndBye, 2023-11-22:
I didn't include the part for `eth1` as I've only included the parts of the config concerning the part that isn't working. I've only mentioned that the physical MTU is set to 1508.  

Here is the config concerning the configuration of the physical interface though:. 
```

config device
        option name 'eth1'
        option macaddr '82:34:xx:xx:xx:xx'
        option mtu '1508'

```

brada4, 2023-11-22:
Add this to your WAN section.
`option pppd_options 'debug'`
 it seems it takes negotiated MTU from other pppd which is on MTU 1500 LAN? In this case MTU is not passed to pppd but (expected but not) set with ~~ ifconfig.

jow-, 2023-11-22:
The PPPoE proto MTU setting is known broken - it applies to too many things at once. Afair it is both translated to pppd mru/mtu options and taken by netifd to set the resulting netdev MTU.

brada4, 2023-11-22:
Log shall show PPP extension negotiation, likely mentioning 1492 few times.

Oorweeg, 2023-12-02:
As @brada4 has mentioned, you need to look at the full pppoe debug output or do a tcpdump of the PPP coming up and view in wireshark. I believe even when using ‘ppp-max-payload’ configuration the ‘MTU’ will still be 1492, this is part of the RFC. There is a separate ppp-max-payload tag where the higher values are defined and the remote side will be looking (if it supports the extensions) to increase the MTU above 1492.

brada4, 2023-12-02:
It negotiates down only, should expose if change is needed on ppp server.

Oorweeg, 2023-12-02:
> It negotiates down only, should expose if change is needed on ppp server.

You are thinking of normal PPP MTU negotiation, baby jumbo support is not part of normal MTU negotiation and is a separate process that overrides the normal PPP MTU negotiation rules when the ppp-max-payload tag is in the request and allows negotiation above 1492. Assuming the server side is configured to accept this and supports these extensions. See RFC4638.

Setting a ppp interface mtu of 1500 shouldn’t change the MTU sent in a pppoe request, the max here is still 1492, but it should set an additional ppp-max-payload tag to 1500 indicating it want to negotiate a larger mtu. This will still fail though if the server isn’t configured to accept this. You cant force it unless its supported both sides.

brada4, 2023-12-02:
Yes, logs should expose server capped at 1492 or us being bugg and insist on 1492.

vincejv, 2024-04-07:
I'm using PPPoE mini jumbo frames since forever, can't repro... on top of eth1 MTU, the other ethernet port where eth1 is connected to must also be > 1500, i think this is a configuration issue with OP