openwrt/openwrt #14027: Meraki MR24 Bad Default Network Configuration

farmergreg, 2023-11-21:
### Describe the bug

The network settings are lost during an upgrade and/or fresh install. The device becomes unreachable via ethernet and jtag must be used to recover.

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

apm821xx/nand

### Device

Meraki MR24

### Image kind

Official downloaded image

### Steps to reproduce

1. Install 19.07.10 (factory default settings)
2. Upgrade to 23.05.2.

IMPORTANT: You can choose to keep the current settings, or reset to factory default. Both ways will trigger this bug!

### Actual behaviour

When the board boots back up, the /etc/config/network file will have no IP address assigned to lan

### Expected behaviour

lan should have 192.168.1.1 assigned to it.

### Additional info

Here is the bad /etc/config/network:

	config interface 'loopback'
		option device 'lo'
		option proto 'static'
		option ipaddr '127.0.0.1'
		option netmask '255.0.0.0'

	config globals 'globals'
		option ula_prefix 'fdcb:2c75:fddd::/48'

	config device
		option name 'br-lan'
		option type 'bridge'
		list ports 'eth0'

	config interface 'lan'
		option device 'br-lan'
		option proto 'dhcp'

	config interface 'lan6'
		option device 'br-lan'
		option proto 'dhcpv6'

Here is an example of a good /etc/config/network

	config interface 'loopback'
		option ifname 'lo'
		option proto 'static'
		option ipaddr '127.0.0.1'
		option netmask '255.0.0.0'

	config globals 'globals'
		option ula_prefix 'fdf0:6055:f7b1::/48'

	config interface 'lan'
		option type 'bridge'
		option ifname 'eth0'
		option proto 'static'
		option ipaddr '192.168.1.1'
		option netmask '255.255.255.0'
		option ip6assign '60'

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

farmergreg, 2023-11-21:
I realized that I wasn't testing with 23.05.0 and not 23.05.2 like I  thought. Unfortunately, my usb serial board just let out some magic smoke, so I can't perform any more analysis until I get a new one.

raenye, 2023-11-21:
Duplicate of #13922.

Since OpenWrt 22.x NAS devices are configured using DHCP; please see abb4083fa06841ef5b094aee24a3b1f58403516b.
To get static IP, use failsafe mode (no JTAG/UART needed) or use `uci-default` in image builder.

brada4, 2023-11-22:
mr24 is not nas

raenye, 2023-11-22:
But that commit explicitly includes mr24

brada4, 2023-11-22:
@chunkeey mushave overlooked something. hundred other 1 eth wifi devices start with .1.1

farmergreg, 2023-11-22:
in some ways, the current behavior makes a lot of sense because the MR24 is a wireless access point. It isn't likely to be used as a router. The new behavior surprised me because other APs like the MR18  default to the standard openwrt 192.168.1.1/24 configuration. I suppose the best thing would be to either make the MR24 act like the other devices, or convert the other wireless access points over to the new behavior too.

brada4, 2023-11-22:
there is lot of 1 eth range extenders , this change seems more taargeted at NAS boxes and one router slipped in by mistake.
lets wait few weeks to see wait patch author intended.

chunkeey, 2023-11-22:
multiple things here.

https://github.com/openwrt/openwrt/commit/abb4083fa06841ef5b094aee24a3b1f58403516b affects "new installs / factory resets on 23.5" . Since OP upgraded from 19.07 the default "192.168.1.1" ethernet configuration would have been in place and this isn't something that gets overwritten unless you tell sysupgrade to disregard the old configuration. 

One suggestion: Please upgrade from one release to the next. I.e.: go from 19.07->21.02->22.03->23.05 instead of skipping the releases in between. 

Since the MR24 is an AP. I don't see why "dhcp" is a "bad" default.  Because now the OpenWRT MR24 is now more closely following what the original Meraki Firmware is doing. That said, one area that I would like to see an improvement would be that if DHCP fails to get an address, then the udhcpc (or netifd?) should try to setup zeroconf as a fallback. (but this is exceedingly OT).

The MR24 doesn't have a ethernet-switch hardware.  The sole ethernet-mac is directly connected to an AT8035 PHY transceiver chip which can only drive the sole RJ45 port. But if you want "ethX.1", you can setup the VLAN with ID 1.
(There's more to the story about ethX.1 / ethX.2 .)

closed 2023-11-22 by chunkeey

brada4, 2023-11-22:
nothing to do with vlans, poor user with windows and cable tries to set their first wifi, no dhcp server around.

farmergreg, 2023-11-23:
@chunkeey I've thought about this some more, and I think the code should be reverted to the original behavior with a static ip address for the MR24. 

Here is why:

- All of the other supported Meraki access points start up with 192.168.1.1 on the eth interface when openwrt boots. To have the MR24 behaving differently is confusing. For example, I also have MR18's that I use with openwrt. These start up with 192.168.1.1 instead of dhcp on the ethernet interface.
- The possibility of someone directly connecting the MR24 to their internet connection and using it as @brada4 described.