openwrt/openwrt #14030: kernel: mac80211: ieee80211_link_info_change_notify warning with batman-adv over 802.11s

raenye, 2023-11-21:
### Describe the bug

I have multiple devices running `batman-adv` over `802.11s` mesh.
Some (but not all) of them (running different architectures) have the following warning appearing *once* in `dmesg`,
It happens *immediately* after `wpa_supplicant` reports (in `logread`):
```
phy0-mesh0: mesh plink with xx:xx:xx:xx:xx:xx established
```
The message *does not* repeat upon disabling and re-enabling WiFi. `batman-adv` works fine. 

### D-Link DAP-1620-B1 (`ramips/mt7621`, kernel 5.15.138):
```
[   41.810117] ------------[ cut here ]------------
[   41.819404] WARNING: CPU: 1 PID: 130 at backports-6.5/net/mac80211/main.c:283 ieee80211_link_info_change_notify+0x14c/0x15c [mac80211]
[   41.843824] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet batman_adv nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7615e mt7615_common mt76_connac_lib mt76 mac80211 cfg80211 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon compat ledtrig_pattern ledtrig_oneshot ledtrig_activity cryptodev sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 crypto_hw_eip93 hmac des_generic libdes cmac cbc authencesn authenc leds_gpio ledtrig_transient gpio_button_hotplug crc32c_generic
[   41.988486] CPU: 1 PID: 130 Comm: kworker/1:5 Not tainted 5.15.138 #0
[   42.001344] Workqueue: events cfg80211_stop_iface [cfg80211]
[   42.012722] Stack : 80850000 80083d84 00000000 00000004 8077cd50 80ff3cc4 80fc8f00 8004b718
[   42.029384]         80850000 80ff3cfb 00000001 80ff3b10 82e00d40 00000001 80ff3cd0 80c41880
[   42.046041]         00000000 00000000 8077cd50 80ff3b10 ffffefff 00000000 ffffffea 00000000
[   42.062689]         80ff3b1c 000000fd 808569c8 ffffffff 8077cd50 00000001 00000000 82f025b0
[   42.079344]         00000009 00000040 81010b05 80fc8f40 00000018 804020ec 00000004 80a10004
[   42.095995]         ...
[   42.100867] Call Trace:
[   42.105723] [<80007db4>] show_stack+0x28/0xf0
[   42.114415] [<80376b40>] dump_stack_lvl+0x60/0x80
[   42.123795] [<8002df4c>] __warn+0x9c/0x124
[   42.131959] [<8002e030>] warn_slowpath_fmt+0x5c/0xac
[   42.141851] [<82f025b0>] ieee80211_link_info_change_notify+0x14c/0x15c [mac80211]
[   42.156968] [<82f768c8>] ieee80211_mesh_work+0x1a8/0x30c [mac80211]
[   42.169547] [<82e00e00>] cfg80211_stop_iface+0x1bc/0x208 [cfg80211]
[   42.182115]
[   42.185187] ---[ end trace 33f7f8634247f2b5 ]---
```

### D-Link DAP-1720 (`ath79/generic`, kernel 6.1.62):
```
[   88.333155] ------------[ cut here ]------------
[   88.338102] WARNING: CPU: 0 PID: 252 at backports-6.5/net/mac80211/main.c:283 ieee80211_link_info_change_notify+0x14c/0x15c [mac80211]
[   88.350909] Modules linked in: ath9k ath9k_common nft_fib_inet nf_flow_table_inet batman_adv ath9k_hw ath10k_pci ath10k_core ath nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 cfg80211 usbnet usbhid rng_core nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon hid_generic compat hid evdev input_core ledtrig_usbport cryptodev rfkill crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic seqiv jitterentropy_rng drbg hmac cmac authencesn authenc usb_storage fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug usbcore nls_base usb_common crc16 mii crc32c_generic
[   88.432820] CPU: 0 PID: 252 Comm: kworker/0:5 Not tainted 6.1.62 #0
[   88.439319] Workqueue: events wiphy_new_nm [cfg80211]
[   88.444647] Stack : 81000205 80eee8c0 81879d1c 800c0e80 00000000 806dc088 80790000 80eee880
[   88.453302]         825025b4 00000009 80c08e5c 800c0f64 80d53668 00000001 81879cd8 55b40d0d
[   88.461956]         00000000 00000000 806dc088 81879bf0 ffffefff 00000000 00000000 ffffffea
[   88.470611]         000000dc 81879bfc 000000dc 807999c0 00000001 806dc088 00000000 825025b4
[   88.479267]         00000009 80c08e5c 81000205 80eee8c0 00000018 8038e400 00000000 80950000
[   88.487922]         ...
[   88.490450] Call Trace:
[   88.492976] [<80066d4c>] show_stack+0x28/0xf0
[   88.497504] [<806006a8>] dump_stack_lvl+0x38/0x60
[   88.502385] [<80085268>] __warn+0x9c/0xe8
[   88.506534] [<80085310>] warn_slowpath_fmt+0x5c/0xac
[   88.511679] [<825025b4>] ieee80211_link_info_change_notify+0x14c/0x15c [mac80211]
[   88.519700] [<8257540c>] ieee80211_mesh_work+0x18c/0x314 [mac80211]
[   88.526385] [<82400b10>] wiphy_new_nm+0x81c/0x85c [cfg80211]
[   88.532340]
[   88.534247] ---[ end trace 0000000000000000 ]---
```

### Xiaomi Redmi Router AX6S (`mediatek/mt7622`, kernel 6.1.61):
Does not show the warning.

### OpenWrt version

r24433+21-2634d3f855

### OpenWrt release

SNAPSHOT
[... 274 lines cut ...]
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_umdns=y
CONFIG_PACKAGE_vim=y
CONFIG_PACKAGE_vnstat2=y
CONFIG_PACKAGE_vnstati2=y
CONFIG_PACKAGE_wifischedule=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_zlib=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_WPA_MBO_SUPPORT=y
CONFIG_WPA_RFKILL_SUPPORT=y
CONFIG_ZLIB_OPTIMIZE_SPEED=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-11-21:
I suspect  `package/kernel/mac80211/patches/subsys/311-mac80211-fix-mesh-id-corruption-on-32-bit-systems.patch` has something to do with it, since both affected targets are 32-bit while the unaffected target is 64-bit.

Would appreciate your opinion @nbd168; thanks.

mpratt14, 2023-12-02:
I experience this as well when the signal is very bad
it's something to do with flags changing in the interface between being down / losing a connection / gaining a connection
(in the trace you can see cfg80211_stop_iface)

If you have external antennas, and you remove them, you can expect it to happen a lot, every time wireless comes up, due to losing connections like crazy, and the interface freaks out

I would try moving locations or adjusting RSSI threshold

the WARN family of functions often cause a scare because it prints the whole stack instead of just a one-liner...
perhaps this is another warning that deserves silencing (or a one-line error)...

it's probably going to take upstream a long time to figure this out, and I think there's more than one issue, but it's not critical to operation
There is a proposed fix for a similar warning, but everyone's disagreeing about what to do
https://patchwork.kernel.org/project/linux-wireless/patch/20230901035301.3473463-1-shaozhengchao@huawei.com/

mpratt14, 2023-12-02:
specifically, at least for me, its trigger by this event

` * @MESH_WORK_MBSS_CHANGED: rebuild beacon and notify driver of BSS changes`

mpratt14, 2023-12-02:
in the mailing list, someone suggested using bitmap_zero() function instead of memset() by itself

might not make a difference, but it's worth a try maybe? i don't know...

raenye, 2023-12-02:
Thanks for the elaborate reply :)
The patch you mentioned is already in OpenWrt, so it does not prevent the warning. 
Which memset do you suggest to change?

xavifr, 2024-02-13:
I have the same issue:

### TP-Link Deco M4R v4 (ramips/mt7621)
```
[   28.503004] ------------[ cut here ]------------
[   28.507684] WARNING: CPU: 3 PID: 265 at backports-6.5/net/mac80211/main.c:283 ieee80211_link_info_change_notify+0x14c/0x15c [mac80211]
[   28.520012] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet batman_adv pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 crypto_hw_eip93 hmac des_generic libdes cmac authencesn authenc leds_gpio gpio_button_hotplug crc32c_generic
[   28.590412] CPU: 3 PID: 265 Comm: kworker/3:3 Not tainted 5.15.148 #0
[   28.596869] Workqueue: events cfg80211_stop_iface [cfg80211]
[   28.602675] Stack : 808c0000 80083bac 00000000 00000004 807d9bf4 81cfdcc4 81da2680 8004b5cc
[   28.611059]         808b0000 81cfdcfb 00000001 81cfdb10 82f00d40 00000001 81cfdcd0 81489880
[   28.619430]         00000000 00000000 807d9bf4 81cfdb10 ffffefff 00000000 ffffffea 00000000
[   28.627788]         81cfdb1c 00000136 808bca50 ffffffff 807d9bf4 00000001 00000000 830025b0
[   28.636145]         00000009 000000c0 81026c05 81da26c0 00000018 80413800 0000000c 80a7000c
[   28.644497]         ...
[   28.646944] Call Trace:
[   28.649378] [<80007db4>] show_stack+0x28/0xf0
[   28.653750] [<803854dc>] dump_stack_lvl+0x60/0x80
[   28.658456] [<8002df2c>] __warn+0x9c/0x124
[   28.662563] [<8002e010>] warn_slowpath_fmt+0x5c/0xac
[   28.667531] [<830025b0>] ieee80211_link_info_change_notify+0x14c/0x15c [mac80211]
[   28.675215] [<83076998>] ieee80211_mesh_work+0x1a8/0x30c [mac80211]
[   28.681605] [<82f00e00>] cfg80211_stop_iface+0x1bc/0x208 [cfg80211]
[   28.688005] 
[   28.689568] ---[ end trace d3ba1f57aaa741e9 ]---
```

### TP-Link Deco M4R v1 (ath79/generic)
```
[   58.594556] ------------[ cut here ]------------
[   58.599566] WARNING: CPU: 0 PID: 275 at backports-6.5/net/mac80211/main.c:283 ieee80211_link_info_change_notify+0x14c/0x15c [mac80211]
[   58.612888] Modules linked in: ath9k ath9k_common pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet batman_adv ath9k_hw ath10k_pci ath10k_core ath pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat sha512_generic seqiv jitterentropy_rng drbg hmac cmac gpio_button_hotplug crc16 crc32c_generic
[   58.677429] CPU: 0 PID: 275 Comm: kworker/0:3 Not tainted 5.15.148 #0
[   58.684201] Workqueue: events wiphy_new_nm [cfg80211]
[   58.689531] Stack : 807b0000 80f79680 82002578 00000009 80c0885c 800c0670 80cb4880 807b1e83
[   58.698420]         809732d4 00000113 81873d14 807b0000 00000000 00000001 81873ce8 6e25155d
[   58.707928]         00000000 00000000 806f7358 81873b68 ffffefff 00000000 00000000 ffffffea
[   58.716856]         000000e0 81873b74 000000e0 807b80b8 80970000 00000009 00000000 82002578
[   58.725600]         00000009 80c0885c 81000205 80f796c0 00000018 8039cba4 00000000 80970000
[   58.734277]         ...
[   58.736816] Call Trace:
[   58.739338] [<80066b40>] show_stack+0x28/0xf0
[   58.743985] [<80084ed8>] __warn+0x9c/0x124
[   58.748247] [<80084fbc>] warn_slowpath_fmt+0x5c/0xac
[   58.753433] [<82002578>] ieee80211_link_info_change_notify+0x14c/0x15c [mac80211]
[   58.762237] [<82075a68>] ieee80211_mesh_work+0x1a8/0x330 [mac80211]
[   58.769078] [<81b00aec>] wiphy_new_nm+0x7f8/0x838 [cfg80211]
[   58.775079] 
[   58.776623] ---[ end trace 07fd36cf3777b97f ]---
[   59.374512] device phy0-mesh0 left promiscuous mode
```

Both running with latest snapshot (OpenWrt SNAPSHOT r25132-dfc1e8cfee / LuCI Master git-24.040.70477-23ebdb3) with kernel 5.15.148.
I have a mesh between them without batman.

The mesh connection works fine for some hours and then stops working until I reboot the devices.

maurerle, 2025-01-15:
I just got this as well on a gluon/openwrt-24.10 node running on a CPE210 v1:

```
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.796752] ------------[ cut here ]------------
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.801621] WARNING: CPU: 0 PID: 390 at backports-6.12.6/net/mac80211/main.c:408 ieee80211_link_info_change_notify+0x154/0x164 [mac80211]
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.815037] Modules linked in: ath9k(O) ath9k_common(O) iptable_nat batman_adv(O) ath9k_hw(O) ath(O) xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_CT wireguard nf_nat nf_conntrack mac80211(O) lzo libchacha20poly1305 ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211(O) xt_time xt_tcpudp xt_quota xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_comment xt_addrtype xt_TCPMSS xt_LOG poly1305_mips nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress libcurve25519_generic libcrc32c iptable_mangle iptable_filter ip_tables ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 compat(O) chacha_mips sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow act_skbedit act_mirred act_gact ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 dummy ip_tunnel veth vxlan udp_tunnel ip6_udp_tunnel sha512_generic seqiv
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.815623]  sha3_generic jitterentropy_rng drbg kpp hmac geniv rng cmac crypto_acompress act_police cls_basic sch_tbf sch_ingress gpio_button_hotplug(O) crc16 crc32c_generic
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.925476] CPU: 0 PID: 390 Comm: kworker/u2:5 Tainted: G           O       6.6.69 #0
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.933453] Workqueue: events_unbound wiphy_delayed_work_pending [cfg80211]
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.940682] Stack : 81addd3c 829941f0 81c052d0 80740828 80810000 81c05280 82a02a50 00000009
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.949209]         829941f0 800be064 808a0000 80740828 81c1dc28 00000001 81addd00 d3ac5a92
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.957727]         00000000 00000000 80740828 81addc20 ffffefff 00000000 00000000 ffffffea
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.966246]         000000df 81addc2c 000000df 80816330 00000001 80740828 00000000 82a02a50
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.974763]         00000009 829941f0 81c052d0 61c88647 00000018 80379ed8 00000000 809d0000
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.983282]         ...
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.985774] Call Trace:
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.988255] [<800666b4>] show_stack+0x28/0xf0
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.992711] [<8065e840>] dump_stack_lvl+0x38/0x60
Tue May  7 20:47:55 2024 kern.warn kernel: [  228.997505] [<80084c1c>] __warn+0xb0/0xe4
Tue May  7 20:47:55 2024 kern.warn kernel: [  229.001589] [<80084cac>] warn_slowpath_fmt+0x5c/0xac
Tue May  7 20:47:55 2024 kern.warn kernel: [  229.006649] [<82a02a50>] ieee80211_link_info_change_notify+0x154/0x164 [mac80211]
Tue May  7 20:47:55 2024 kern.warn kernel: [  229.014605] [<829003ec>] wiphy_delayed_work_pending+0x3ec/0x540 [cfg80211]
Tue May  7 20:47:55 2024 kern.warn kernel: [  229.021756]
Tue May  7 20:47:55 2024 kern.warn kernel: [  229.025112] ---[ end trace 0000000000000000 ]---
```

Though I do not see any problems with it :)

MisharUser, 2025-03-15:
ramips/mt7621

OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.072.79561~41a338d

```
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.648187] ------------[ cut here ]------------
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.653078] WARNING: CPU: 0 PID: 44 at backports-6.12.6/net/mac80211/main.c:408 0x83a02a78 [mac80211@d219c6b6+0x8a000]
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.664099] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet batman_adv(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_bridge nf_conntrack mt7915e(O) mt76_connac_lib(O) mt76(O) mac80211(O) cfg80211(O) slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat(O) ip_tunnel vxlan udp_tunnel ip6_udp_tunnel sha512_generic sha256_generic sha1_generic seqiv sha3_generic drbg md5 crypto_hw_eip93 hmac geniv rng des_generic libdes cmac authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug(O) usbcore nls_base usb_common crc32c_generic
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.744376] CPU: 0 PID: 44 Comm: kworker/u9:2 Tainted: G           O       6.6.73 #0
Sat Mar 15 01:36:22 2025 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Sat Mar 15 01:36:22 2025 daemon.notice hostapd: Reloaded settings for phy phy1
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.752165] Workqueue: events_unbound 0x8390106c [cfg80211@1ec3f85a+0x46000]
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.759292] Stack : 00000000 0000002c 80943440 80841920 00000000 00000000 00000000 00000000
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.767652]         00000000 00000000 00000000 00000000 00000000 00000001 819afce8 8184b840
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.776031]         819afd80 00000000 00000000 819afbc0 00000038 80822a84 ffffffea 00000000
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.784485]         819afbcc 00000160 8094b060 ffffffff 80841920 819afcc8 00000000 83a02a78
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.792924]         00000009 00000000 80943440 81806020 00000018 80496db4 01813597 018135d7
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.801343]         ...
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.803830] Call Trace:
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.803862] [<80822a84>] 0x80822a84
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.809853] [<83a02a78>] 0x83a02a78 [mac80211@d219c6b6+0x8a000]
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.815835] [<80496db4>] 0x80496db4
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.819325] [<800073b8>] 0x800073b8
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.822818] [<800073c0>] 0x800073c0
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.826353] [<83a02a78>] 0x83a02a78 [mac80211@d219c6b6+0x8a000]
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.832365] [<807f8c54>] 0x807f8c54
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.835918] [<8002d9b0>] 0x8002d9b0
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.839514] [<83a02a78>] 0x83a02a78 [mac80211@d219c6b6+0x8a000]
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.845565] [<8002db0c>] 0x8002db0c
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.849229] [<83a02a78>] 0x83a02a78 [mac80211@d219c6b6+0x8a000]
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.855265] [<83a7d0cc>] 0x83a7d0cc [mac80211@d219c6b6+0x8a000]
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.861561] [<83901148>] 0x83901148 [cfg80211@1ec3f85a+0x46000]
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.867588] [<808245bc>] 0x808245bc
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.871217] [<80048db8>] 0x80048db8
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.874888] [<800498fc>] 0x800498fc
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.878436] [<80053698>] 0x80053698
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.881991] [<80824940>] 0x80824940
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.885545] [<80049560>] 0x80049560
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.889095] [<80053698>] 0x80053698
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.892637] [<8005379c>] 0x8005379c
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.896180] [<80053698>] 0x80053698
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.899730] [<80053698>] 0x80053698
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.903282] [<80002a58>] 0x80002a58
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.906845]
Sat Mar 15 01:36:22 2025 kern.warn kernel: [   38.908809] ---[ end trace 0000000000000000 ]---
```

pktpls, 2025-05-13:
Recently started getting this on a Ubiquiti UniFi AC Mesh (ath79/generic) with OpenWrt 24.10-SNAPSHOT r28601-9e9f6875bc