openwrt/openwrt #14042: Regression: 1TB SD card does no longer function in SD card reader with OpenWRT 23.05.2

BorreSanders, 2023-11-22:
### Describe the bug

After upgrading from OpenWRT 22.03.5 to 23.05.2 I can no longer access SD card contents from a 1TB SD card inserted in SD card slot.
- Kernel log shows errors with built-in SD card reader
- If the same SD card is used via an external SD card reader connected to USB the card is readable
- Whether the file system is FAT32 or exFAT does not make a difference
- A SD card of a smaller size (32GB) does function correctly in the built-in SD card slot
- Problem with large capacity SD card in built-in SD card slot did not exist with 22.03.5 (as far as I know), since I used the same card there without any issues.

Seems that there is a software change that breaks compatibility of the SD card reader with large capacity SD cards.
See attached kernel log file for some details.

[kernel log.txt](https://github.com/openwrt/openwrt/files/13442942/kernel.log.txt)

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt76x8

### Device

RAVPower RP-WD009

### Image kind

Official downloaded image

### Steps to reproduce

1. Install OpenWRT 22.03.5 on RAVPower RP-WD009
2. Insert 1TB SD card in built-in SD card slot
3. Observe that card is recognized, and one can generate a valid config for a mount point
4. Upgrade OpenWRT to 23.05.2
5. Observe that card is no longer recognized, and one can generate a valid config for a mount point

### Actual behaviour

SD card not recognized when inserted in built-in SD card slot, and contents inaccessible

### Expected behaviour

SD card recognized and content accessible

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-22:
Can you help with posting those errors, also try to check parameters in /sys/modules for sdio and mmc modules

BorreSanders, 2023-11-22:
Hello Brada4. Thanks for checking in. I'm not sure what errors you are referring to. I attached the kernel logs, aren't the errors what is shown there? Also, I'm not an expert at OpenWRT and Linux. Can you perhaps give me some instruction, e.g.  command-line, using which I can provide you the info needed for sys/module/mmc*?

brada4, 2023-11-22:
You can hide MAC addresses and peoples names, whole dmesg would help best. (you can try compare mmc0 init messages with previous version if possible)
Spmething along lines of bulk or crc or voltage modes.

BorreSanders, 2023-11-23:
Thanks. I'll check and report dmesg later this week

BorreSanders, 2023-11-24:
Found info on listing contents of files in all subdirectories [here](https://stackoverflow.com/questions/21277963/how-to-display-contents-of-all-files-under-a-directory-on-the-screen-using-unix). Will try and compare mmc and sdio parameters this way.

brada4, 2023-11-24:
It is just module size difference
You can try changing just one module parameter to change IO mode of mmc/sdcard:
`/etc/modules.d/mmc`
```
mmc_block
mmc_core use_spi_crc=N
```
You can do more with usermode mmc-tools like to figure out unstable IO modes, but lets try checkbox first.

BorreSanders, 2023-11-24:
Hello, I tried adding use_spi_crc=N to the mmc_core statement, but noticed no effect:
[    4.077221] init: - preinit -
[    5.393697] random: jshn: uninitialized urandom read (4 bytes read)
[    5.574435] random: jshn: uninitialized urandom read (4 bytes read)
[    5.642973] random: jshn: uninitialized urandom read (4 bytes read)
[    6.869029] msdc0 -> XXX CMD<48> wait xfer_done<512> timeout!! <- msdc_do_request() : L<1216> PID<kworker/0:5><0x5d>
[    6.889941] msdc0 ->     DMA_SA   = 0x18f1000 <- msdc_do_request() : L<1217> PID<kworker/0:5><0x5d>
[    6.907876] msdc0 ->     DMA_CA   = 0x18f1000 <- msdc_do_request() : L<1218> PID<kworker/0:5><0x5d>
[    6.925815] msdc0 ->     DMA_CTRL = 0x6100 <- msdc_do_request() : L<1219> PID<kworker/0:5><0x5d>
[    6.943234] msdc0 ->     DMA_CFG  = 0x3 <- msdc_do_request() : L<1220> PID<kworker/0:5><0x5d>
[    6.960144] mmc0: error -145 reading general info of SD ext reg
[    6.971886] mmc0: error -145 whilst initialising SD card
I welcome any suggestion for continued analysis. Thanks for your guidance!

brada4, 2023-11-24:
Take out parameter, other parameters are via command line - best if you have smaller/cheaper card to eventually hard lock up?
install mmc-utils and try to manipulate transfer modes of the card:
https://git.kernel.org/pub/scm/utils/mmc/mmc-utils.git/tree/man/mmc.1

Something like blkid should ID filesystems or show errors you get at present, if you get past that you can check stability of card. What I suspect recent MMC driver added some new transfer routines that your card or controller claim to support but in reality does not.

brada4, 2023-11-24:
Obviouly note which parameter needs to be flipped

BorreSanders, 2023-11-24:
I removed the parameter again, rebooted the device and installed mmc-utils and blkid. A different SD card with capacity of 256GB is recognized and listed when using blkid. When inserting the 1TB card I can not run mmc because of the same error as before. Also blkid does not list the 1TB card. Not sure what to try/do next.

brada4, 2023-11-24:
switxh back to v22, and scan mmc related upstream git logs,  trying to bisect issue. Dts files for your router have not changed for years.

BorreSanders, 2023-11-24:
Not sure what's the nature of a mmc change to look for in git and how. Reverted to release 22.03.5 as interim containment, hoping for a solution someday.

brada4, 2023-11-24:
That involves following by means of recompilation  mmc-core/mmc-blk kernel changes between 22 and 23 kernels and pointing to change that breaks the mmc....

BorreSanders, 2023-11-25:
I get the needed approach, and unfortunately personally am not able to address that. I hope that this bug report may serve others in the future, yet I need to stop here.

BorreSanders, 2023-11-26:
Did one more check. Problem exists already with OpenWrt 23.05.0-rc1 (r23069-e2701e0f33). Same error as with 23.05.2 is then reported. May be that helps to narrow down the research scope for someone in the future.

brada4, 2023-11-26:
snapshot?

BorreSanders, 2023-11-27:
[rc1 sysupgrade package](https://downloads.openwrt.org/releases/23.05.0-rc1/targets/ramips/mt76x8/openwrt-23.05.0-rc1-ramips-mt76x8-ravpower_rp-wd009-squashfs-sysupgrade.bin)

brada4, 2023-11-27:
i mean if snaoshot fixes the problem,

BorreSanders, 2023-11-27:
I'm willing to try snapshots with some binary-search style testing of different snapshot versions, if that helps to narrow down the window of where the issue was introduced. Assuming you are referring to sysupgrade package snapshots, where can I find those for this particular device?

BorreSanders, 2023-11-27:
Result: problem is still present (mmc0: error -145)

dmesg logging:
BusyBox v1.36.1 (2023-11-26 01:14:04 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r24498-b2f1c6ed52
 -----------------------------------------------------
root@OpenWrt:~# dmesg
[    0.000000] Linux version 5.15.139 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24498-b2f1c6ed52) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Sun Nov 26 01:14:04 2023
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7628AN ver:1 eco:2
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
[    0.000000] MIPS: machine is RAVPower RP-WD009
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55440K/65536K available (6027K kernel code, 608K rwdata, 1304K rodata, 1204K init, 217K bss, 10096K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] intc: using register map from devicetree
[    0.000000] CPU Clock: 580MHz
[    0.000000] timer_probe: no matching timers found
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
[    0.000003] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
[    0.015476] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.087661] pid_max: default: 32768 minimum: 301
[    0.097967] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.112326] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.137421] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.156824] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.170448] pinctrl core: initialized pinctrl subsystem
[    0.183001] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.195464] thermal_sys: Registered thermal governor 'step_wise'
[    0.468751] PCI host bridge to bus 0000:00
[    0.488599] pci_bus 0000:00: root bus resource [mem 0x20000000-0x2fffffff]
[    0.502188] pci_bus 0000:00: root bus resource [io  0x10160000-0x1016ffff]
[    0.515778] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.531485] pci 0000:00:00.0: [14c3:0801] type 01 class 0x060400
[    0.543316] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[    0.555689] pci 0000:00:00.0: reg 0x14: [mem 0x20200000-0x2020ffff]
[    0.568153] pci 0000:00:00.0: supports D1
[    0.575968] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    0.589456] pci 0000:01:00.0: [14c3:7650] type 00 class 0x028000
[    0.601300] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000fffff]
[    0.613782] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    0.626116] pci 0000:01:00.1: [14c3:8650] type 00 class 0x0d1100
[... 185 lines cut ...]
[   70.198930] br-lan: port 3(phy0-ap0) entered blocking state
[   70.210025] br-lan: port 3(phy0-ap0) entered forwarding state
[   70.221596] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   70.632952] br-lan: port 3(phy0-ap0) entered disabled state
[   71.276999] br-lan: port 3(phy0-ap0) entered disabled state
[   71.289181] device phy0-ap0 left promiscuous mode
[   71.298744] br-lan: port 3(phy0-ap0) entered disabled state
[   72.435681] phy0-sta0: authenticate with cc:32:e5:60:44:5f
[   72.446768] phy0-sta0: 80 MHz not supported, disabling VHT
[   72.553423] phy0-sta0: send auth to cc:32:e5:60:44:5f (try 1/3)
[   72.567076] phy0-sta0: authenticated
[   72.612914] phy0-sta0: associate with cc:32:e5:60:44:5f (try 1/3)
[   72.657972] phy0-sta0: RX AssocResp from cc:32:e5:60:44:5f (capab=0x1031 status=0 aid=17)
[   72.674552] phy0-sta0: associated
[   73.837713] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-sta0: link becomes ready
[   73.992011] br-lan: port 3(phy0-ap0) entered blocking state
[   74.003155] br-lan: port 3(phy0-ap0) entered disabled state
[   74.014628] device phy0-ap0 entered promiscuous mode
[   74.533626] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[   74.547062] br-lan: port 3(phy0-ap0) entered blocking state
[   74.558171] br-lan: port 3(phy0-ap0) entered forwarding state
[  146.760386] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
[  146.773838] br-lan: port 2(phy1-ap0) entered blocking state
[  146.784960] br-lan: port 2(phy1-ap0) entered forwarding state
root@OpenWrt:~#

brada4, 2023-11-27:
You can try to read "ext reg" with mmc command line tool, maybe it helps crash it, say if some command certainly crashes at least you know where sd driver may not to go. It is upstream issue, feel free to report there, and campaign for backport ;-)

BorreSanders, 2023-11-28:
__

> You can try to read "ext reg" with mmc command line tool, maybe it helps crash it, say if some command certainly crashes at least you know where sd driver may not to go. It is upstream issue, feel free to report there, and campaign for backport ;-)

What upstream issue would you suggest I report, and where should I do so?

brada4, 2023-11-28:
regression introduced between kernel x.y.z and a.b.c that this sd controller does not read anymore that sd card.

BorreSanders, 2023-11-28:
I'm going to do one more check with another brand 1TB card.

brada4, 2023-11-28:
`https://bugzilla.kernel.org`
Seems not to be wiring problem like not getting bigger than 32GB cards, just that at some point 'read ext reg' was mandated to access card.

BorreSanders, 2023-11-29:
I'm not sure about that. For my understanding, is "mmc extcsd read /dev/mmcblk0" using mmc-utils the same as what you refer to as reading ext reg? I'm asking because I observed that this command failed for the 32GB card too, yet the card was mountable.

brada4, 2023-11-29:
Probably something before that locks up card or controller.

BorreSanders, 2023-12-02:
Did cross-check with Sandisk 1TB micro-SD card in internal SD card reader slot (mmc). Same problem as with Lexar card. Conclusion: problem is not specific to a SD card brand, but must be related to the card storage capacity size.

BorreSanders, 2023-12-02:
I browsed bugzilla for a duplicate report, but could not find a specific match.
My guess is that the issue I'm facing can be related to this, but I'm not sure: https://lore.kernel.org/stable/ZUQ_wm22gO7lLZ3N@codewreck.org/
I'm not sure what exactly to report, and what fix to ask.
Can you perhaps help me formulate a proper kernel bug report based on findings of this investigation?

brada4, 2023-12-02:
You can report you are unable to access card, or identify missing quirk removed that causes it.

quirks are like this device announces 2.5v support but switches off when using it.

fkroessbacher, 2023-12-06:
Hello Out there, i've got the same PRoblem with an Mikrotik 750G3 and a 128GByte SD Card

BorreSanders, 2023-12-06:
Thanks for adding your case. I submitted a kernel bug report as suggested here.
See https://bugzilla.kernel.org/show_bug.cgi?id=218236
Hope it helps!

brada4, 2023-12-07:
Thats above capabilities of OpenWRT.... You can try snapshot with kernel 6.1 (which receives same support level as any other longterm branch)

BorreSanders, 2023-12-07:
Do you mean downloading latest snapshot (which is based on kernel version 5.15.141), and somehow replacing the kernel with 6.1? If so, I'm not aware how I can arrange that. Can you provide (a reference to) instructions how to arrange Openwrt with kernel 6.1?

brada4, 2023-12-07:
I dont know what broke but at one point 6.1 was already made default now it is back.
Just wait a month or so when it gets to 6.1

shenghuang147, 2024-01-19:
I've encountered the same issue on Ramips MT7621, where a 256GB TF card couldn't be identified. I attempted to build it using both `v23.05.2` and mainline, but none of them was able to recognize the 256GB TF card. 
The model of the TF card is `SanDisk Extreme Pro Micro SDXC UHS-I U3 A2 V30 Memory Card (256GB)`
Currently, I only have an 8GB and a 256GB TF card at my disposal. I can confirm that the 8GB TF card works as expected, but the 256GB one can't be recognized.

<details>
  <summary>Main Kernel Log</summary>
  <pre><code>
[    0.000000] Linux version 5.15.147 (ubuntu@openwrt) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24855-3891355015) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Wed Jan 17 20:56:35 2024
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is JDCloud RE-CP-02
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s14512 r8192 d22352 u45056
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113536
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Writing ErrCtl register=00000a20
[    0.000000] Readback ErrCtl register=00000a20
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 442924K/458752K available (7039K kernel code, 622K rwdata, 1444K rodata, 1248K init, 216K bss, 15828K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.008028] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066209] pid_max: default: 32768 minimum: 301
[    0.071607] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.078817] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.090315] rcu: Hierarchical SRCU implementation.
[    0.095792] smp: Bringing up secondary CPUs ...
[    0.101159] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.101185] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.101200] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.101244] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.160535] Synchronize counters for CPU 1: done.
[    0.192664] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.192685] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.192697] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.192723] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.251798] Synchronize counters for CPU 2: done.
[    0.282453] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.282473] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.282483] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.282514] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.336977] Synchronize counters for CPU 3: done.
[    0.366831] smp: Brought up 1 node, 4 CPUs
[    0.375935] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.385733] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.392811] pinctrl core: initialized pinctrl subsystem
[    0.399464] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.406082] thermal_sys: Registered thermal governor 'step_wise'
[    0.435627] clocksource: Switched to clocksource GIC
[    0.442094] NET: Registered PF_INET protocol family
[    0.447177] IP idents hash table entries: 8192 (order: 4, 65536 bytes, linear)
[... 518 lines cut ...]
[   33.684628] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   33.696323] mt7530-mdio mdio-bus:1f lan1: configuring for phy/gmii link mode
[   33.706262] br-lan: port 1(lan1) entered blocking state
[   33.711745] br-lan: port 1(lan1) entered disabled state
[   33.719658] device lan1 entered promiscuous mode
[   33.724337] device eth0 entered promiscuous mode
[   33.760315] mt7530-mdio mdio-bus:1f lan2: configuring for phy/gmii link mode
[   33.768976] br-lan: port 2(lan2) entered blocking state
[... cut ...]
[   35.438231] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control rx/tx
[   35.446409] br-lan: port 1(lan1) entered blocking state
[   35.451688] br-lan: port 1(lan1) entered forwarding state
[   35.458703] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   69.754615] done.
[   69.756621] jffs2: notice: (1992) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[   69.919170] overlayfs: upper fs does not support tmpfile.
  </code></pre>
</details>

jhswartz, 2024-01-25:
To make use of UHS cards with 23.05.2, I [patched](https://github.com/openwrt/openwrt/files/14056500/0002-mtk-mmc-sd-ignore-cmd48.txt) the _mtk-mmc_ driver's _msdc_do_request()_ function to ignore requests for CMD48. This works, but probably the best approach would be to make sure that extension register access is handled correctly in _mtk-mmc_.

Before
```
[   10.596295] msdc0 -> XXX CMD<48> wait xfer_done<512> timeout!! <- msdc_do_request() : L<1216> PID<kworker/1:3><0xac>
[   10.617328] msdc0 ->     DMA_SA   = 0x36b1000 <- msdc_do_request() : L<1217> PID<kworker/1:3><0xac>
[   10.635368] msdc0 ->     DMA_CA   = 0x36b1000 <- msdc_do_request() : L<1218> PID<kworker/1:3><0xac>
[   10.653407] msdc0 ->     DMA_CTRL = 0x6100 <- msdc_do_request() : L<1219> PID<kworker/1:3><0xac>
[   10.670944] msdc0 ->     DMA_CFG  = 0x3 <- msdc_do_request() : L<1220> PID<kworker/1:3><0xac>
[   10.687958] mmc0: error -145 reading general info of SD ext reg
[   10.699777] mmc0: error -145 whilst initialising SD card
```

After
```
[    8.047488] mmc0: new high speed SDHC card at address aaaa
[    8.076588] mmcblk0: mmc0:aaaa SK32G 29.7 GiB
[    8.087580]  mmcblk0: p1 p2
```

I think the _mtk-mmc_ driver's difficulty with making sense of CMD48 responses seems to have only become an issue after [this set of kernel patches](https://patchwork.kernel.org/project/linux-mmc/cover/20210421103154.169410-1-ulf.hansson@linaro.org/) introduced a call from _mmc_sd_init_card()_ to the (then new) _sd_read_ext_regs()_ function.

_sd_read_ext_regs()_ attempts to read from the card's extension registers if a bit that indicates that the card supports CMD48 is set in the SCR, otherwise it returns without error if CMD48 isn't supported.

shenghuang147, 2024-01-26:
> To make use of UHS cards with 23.05.2, I [patched](https://github.com/openwrt/openwrt/files/14056500/0002-mtk-mmc-sd-ignore-cmd48.txt) the _mtk-mmc_ driver's _msdc_do_request()_ function to ignore [...]

Thank you for your patch. I have tested it and can confirm that this patch works well in the mainline

<details>
  <summary>Main Kernel Log</summary>
  <pre><code>
[    0.000000] Linux version 5.15.147 (ubuntu@openwrt) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24855-3891355015) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Fri Jan 26 04:26:52 2024
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is JDCloud RE-CP-02
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s14512 r8192 d22352 u45056
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113536
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Writing ErrCtl register=00073020
[    0.000000] Readback ErrCtl register=00073020
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 442924K/458752K available (7039K kernel code, 622K rwdata, 1444K rodata, 1248K init, 216K bss, 15828K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.008035] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066220] pid_max: default: 32768 minimum: 301
[    0.071629] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.078840] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.090500] rcu: Hierarchical SRCU implementation.
[    0.096015] smp: Bringing up secondary CPUs ...
[    0.101439] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.101463] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.101478] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.101524] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.160804] Synchronize counters for CPU 1: done.
[    0.193464] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.193483] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.193495] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.193523] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.252353] Synchronize counters for CPU 2: done.
[    0.283026] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.283048] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.283059] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.283091] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.337539] Synchronize counters for CPU 3: done.
[    0.367399] smp: Brought up 1 node, 4 CPUs
[    0.376559] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.386360] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.393447] pinctrl core: initialized pinctrl subsystem
[    0.400117] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.406743] thermal_sys: Registered thermal governor 'step_wise'
[    0.436670] clocksource: Switched to clocksource GIC
[    0.443170] NET: Registered PF_INET protocol family
[    0.448259] IP idents hash table entries: 8192 (order: 4, 65536 bytes, linear)
[... 247 lines cut ...]
mtdblock2    31:2    0   256K  1 disk 
mtdblock3    31:3    0  15.4M  0 disk 
mtdblock4    31:4    0   2.7M  1 disk 
mtdblock5    31:5    0  12.7M  1 disk /rom
mtdblock6    31:6    0   7.6M  0 disk /overlay
mmcblk0     179:0    0 238.3G  0 disk 
└─mmcblk0p1 179:1    0 238.3G  0 part 
root@OpenWrt:/# mount /dev/mmcblk0p1 /mnt/
[  246.249059] exFAT-fs (mmcblk0p1): invalid fs_name
[  246.253842] exFAT-fs (mmcblk0p1): failed to read boot sector
[  246.259483] exFAT-fs (mmcblk0p1): failed to recognize exfat type
[  246.283682] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
root@OpenWrt:/# lsblk 
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mtdblock0    31:0    0   256K  1 disk 
mtdblock1    31:1    0    64K  0 disk 
mtdblock2    31:2    0   256K  1 disk 
mtdblock3    31:3    0  15.4M  0 disk 
mtdblock4    31:4    0   2.7M  1 disk 
mtdblock5    31:5    0  12.7M  1 disk /rom
mtdblock6    31:6    0   7.6M  0 disk /overlay
mmcblk0     179:0    0 238.3G  0 disk 
└─mmcblk0p1 179:1    0 238.3G  0 part /mnt
  </code></pre>
</details>

MikHulk, 2024-02-11:
Hi 

I have this problem too on rammips mt-7621 and a 566GB card. How can I use your patch ? I have upgrades my system from firmware image here https://downloads.openwrt.org/releases/23.05.2/targets/ramips/mt7621/

jhswartz, 2024-02-11:
> Hi
> 
> I have this problem too on rammips mt-7621 and a 566GB card. How can I use your patch ? I have upgrades my system from firmware image here https://downloads.openwrt.org/releases/23.05.2/targets/ramips/mt7621/

You'll need to build OpenWRT for your target/board from source, with the patch applied.

MikHulk, 2024-02-11:
I have never done that . Can I do that from any platform ? 

I downgraded to 22.05 now. And I am afraid I won't have time for the next few days. Maybe, Is there a new factory image in progress with this patch ?

brada4, 2024-02-11:
Next sysupgrade release will include the fix. Building from source involves installing a Linux virtual machine and a lot of time.

MikHulk, 2024-02-11:
Does one need a mips emulator as qemu or something ?

Documentation given for that  does not mention the vm https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem

Is it possible to build from mac OS too. But I have 3 linux . Why do I need a vm ?

MikHulk, 2024-02-11:
I will try later from a container I think.

MikHulk, 2024-02-11:
I'm mostly afraid of spending time on the `make menuconfig` . TBH

jhswartz, 2024-02-11:
> Does one need a mips emulator as qemu or something ?
There's no need for MIPS emulation.

> Is it possible to build from mac OS too. But I have 3 linux . Why do I need a vm ?
If you have a suitable Linux distribution running on real hardware (ie. an amd64 box), then you don't need it installed in a virtual machine.

MikHulk, 2024-02-11:
Is there a way to extract the configuration for the build from my running system ? my hardware is a zbtlink_zbt-wg3526-32m . I try to find it in source but found nothing. So I guess I need to know exactly what is my hardware. How can I know that ?

MikHulk, 2024-02-11:
humm ok so I guess I need to target mt-7620 with packages provided here https://firmware-selector.openwrt.org/?version=23.05.2&target=ramips%2Fmt7621&id=zbtlink_zbt-wg3526-32m 

I will try. Thanks a lot.

jhswartz, 2024-02-11:
Select `Zbtlink ZBT-WG3526 32M` from the `Target Profile` menu.

You can find out about the default package configuration from the device definition for `zbtlink_zbt-wg3526-32m` in `target/linux/ramips/image/mt7621.mk`.

For any other packages you want, you'll have to select them yourself.

MikHulk, 2024-02-11:
ok got it https://github.com/openwrt/openwrt/blob/2253645411d98621466e4ba5e92595aeff3b413d/target/linux/ramips/image/mt7621.mk#L3025 

Thanks a lot. I was looking into mt7620... I don't know why I thought it was this one...

fkroessbacher, 2024-03-25:
I've tried with the new 23.05.03, and got the samw issue

BorreSanders, 2024-03-25:
Thanks for trying and notification of the result. Too bad the fix is not included yet. After earlier messages in the ticket thread I was under the impression that inclusion in next release was secured. Not sure how inclusion triage works for Openwrt. Can anyone advise how to sollicit fix inclusion? Thanks in advance!

jhswartz, 2024-04-03:
> Thanks for trying and notification of the result. Too bad the fix is not included yet. After earlier messages in the ticket thread I was under the impression that inclusion in next release was [...]

I'd be happy to prepare and submit the patch, however might be appropriate for this distribution.

What's the procedure to follow to make that happen?

BorreSanders, 2024-04-11:
Today tried snapshot build with kernel 6.1.82, hoping the issue may be solved already for a future version. Unfortunately, Error -145 is still present: "mmc0: error -145 reading general info of SD ext reg".
So having proposed patch accepted might be a great solution!
Can anyone advise @jhswartz how to address this?

jhswartz, 2024-05-08:
I use the Linux mainline MediaTek SD/MMC Card Driver driver (drivers/mmc/host/mtk-sd.c) in my own Linux (v6.x) + busybox distribution and it works well.

Recently, I needed to get one of my client's MT7621 based targets to boot OpenWRT (kernel and root filesystem, no initramfs) from an SD card via U-Boot in SPI-NOR flash.

I found that OpenWRT 23.05.2's default buildroot and kernel (v5.15.137) configuration didn't work for this purpose out of the box, mostly because kmod-sdhci-mt7620 is a kernel module that would only be present if I use an initramfs and because I do not want to make use of extroot/overlayfs trickery.
 
So I thought, why not see if the mainline driver will work unmodified with OpenWRT 23.05.2's v5.15.137 kernel? Turns out that it works and requires no driver code modification.

To give it a try:

1. Modify your target's device tree source file to include the following nodes as children of /:

```
        mmc_fixed_3v3: regulator-3v3 {
                compatible = "regulator-fixed";

                enable-active-high;

                regulator-always-on;
                regulator-max-microvolt = <3300000>;
                regulator-min-microvolt = <3300000>;
                regulator-name = "mmc_power";
        };

        mmc_fixed_1v8_io: regulator-1v8 {
                compatible = "regulator-fixed";

                enable-active-high;

                regulator-always-on;
                regulator-max-microvolt = <1800000>;
                regulator-min-microvolt = <1800000>;
                regulator-name = "mmc_io";
        };

        mmc: mmc@1e130000 {
                compatible = "mediatek,mt7620-mmc";
                reg = <0x1e130000 0x4000>;

                bus-width = <4>;

                cap-mmc-highspeed;
                cap-sd-highspeed;

                clocks = <&sysc MT7621_CLK_SHXC>,
                         <&sysc MT7621_CLK_50M>;
                clock-names = "source", "hclk";

                disable-wp;

                interrupt-parent = <&gic>;
                interrupts = <GIC_SHARED 20 IRQ_TYPE_LEVEL_HIGH>;

                max-frequency = <48000000>;

                pinctrl-names = "default", "state_uhs";
                pinctrl-0 = <&sdhci_pins>;
                pinctrl-1 = <&sdhci_pins>;

                vmmc-supply = <&mmc_fixed_3v3>;
                vqmmc-supply = <&mmc_fixed_1v8_io>;
        };
```

2. Remove, or comment out, the `&sdhci` node if present, so that remains disabled.

3. Find your board's definition in `target/linux/ramips/image/mt7621.mk` and remove _kmod-sdhci-mt7620_ from DEVICE_PACKAGES.

4. Run `make menuconfig` and go to _Kernel modules -> Other modules_, and deselect _kmod-mmc_ and _kmod-sdhci-mt7620_ if necessary.

5. Run `make kernel_menuconfig` and enable the following configuration definitions. Make sure the indicators are * and M.

```
        Device Drivers
          MMC/SD/SDIO card support
    <*>     MMC block device driver
    <*>     MediaTek SD/MMC Card Interface support
```

6. Finally, run `make` with whatever options you need to rebuild your distribution.

Success looks something like:
```
[    1.508203] mmc0: new high speed SDHC card at address aaaa
[    1.523878] mmcblk0: mmc0:aaaa SK32G 29.7 GiB 
[    1.535445]  mmcblk0: p1
```

BorreSanders, 2024-06-06:
Without having a development environment I'm not sure how to support moving forward with a fix, but found this guide on submitting patches in the core packages: https://openwrt.org/submitting-patches
Can anyone take initiative to propose fix inclusion?

strayge, 2024-06-15:
Also reproduced this error.

GL.iNet GL-MT1300
OpenWrt 23.05.3 r23809-234f1a2efa
SanDisk 128GB MAX Endurance

Fixed with [patch](https://github.com/openwrt/openwrt/issues/14042#issuecomment-1910769942) from this thread.

I have builded `kmod-sdhci-mt7620` from release branch with [config from the same release](https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem#using_official_build_config).

This way builded kernel modules has the same hash as release so I installed patched `kmod-sdhci-mt7620_5.15.150-1_mipsel_24kc.ipk` on device with official `23.05.3` firmware without full reinstall and preserve compatibility with official kernel modules.

Patched ipk uploaded [here](https://mega.nz/file/k4JQXIAL#ABILx-zZ5DkG_NbeYleYi_wv3LVjAYMLIz6CjmmwziU)

tmn505, 2024-07-08:
#15896 will very likely fix this issue, so anyone affected should test it.

jhswartz, 2024-07-08:
+1 for using upstream!

meesterwu, 2024-07-13:
> #15896 will very likely fix this issue, so anyone affected should test it.

Working for me! Used kmod-mmc-mtk. Excellent!!!

[3 more replies]