openwrt/openwrt #14048: ipq40xx: kernel WARNING: phy0-ap0.sta1: Failed check-sdata-in-driver check, flags: 0x1

guidosarducci, 2023-11-23:
### Describe the bug

Sometimes will see a kernel warning in `dmesg` related to the wireless driver:
```
 WARNING: CPU: 3 PID: 1309 at backports-6.1.24/net/mac80211/driver-ops.h:611 0xbf4963ac [mac80211@05829ad7+0x7f000]
 phy0-ap0.sta1: Failed check-sdata-in-driver check, flags: 0x1
```
What triggers this is unclear, but it seems to occur more often while running `tcpdump` on another wlan interface. It might also be relevant that the `phy0-ap0` is a 5 GHz AP in WDS mode. 

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys EA8300 (Dallas)

### Image kind

Official downloaded image

### Steps to reproduce

Seems to occur running `tcpdump`, e.g. on a wlan iface
`# tcpdump -i phy1-ap1`

### Actual behaviour

Sometimes observe a kernel WARNING:

[691238.690172] ------------[ cut here ]------------
[691238.690485] WARNING: CPU: 3 PID: 1309 at backports-6.1.24/net/mac80211/driver-ops.h:611 0xbf4963ac [mac80211@05829ad7+0x7f000]
[691238.694175] phy0-ap0.sta1: Failed check-sdata-in-driver check, flags: 0x1
[691238.706318] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount ath10k_pci ath10k_core ath xt_state xt_helper xt_conntrack xt_connmark xt_connbytes xt_CT wireguard pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_flow_table nf_conntrack_netlink nf_conntrack mac80211 libchacha20poly1305 ipt_REJECT curve25519_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbhid ts_kmp ts_fsm ts_bm slhc sch_cake poly1305_arm nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_objref nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_counter nf_tables nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic iptable_mangle iptable_filter ipt_ECN ip_tables hwmon
[691238.708170]  hid_generic crc_ccitt compat chacha_neon act_ipt sch_teql sch_sfq sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_pedit act_csum libcrc32c em_ipset cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact hid evdev input_core ledtrig_usbport xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ifb dummy ip6_udp_tunnel udp_tunnel sit tunnel4 ip_tunnel sha512_arm kpp ghash_arm_ce cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug crc32c_generic
[691238.856538] CPU: 3 PID: 1309 Comm: hostapd Not tainted 5.15.134 #0
[691238.878717] Hardware name: Generic DT based system
[691238.885071] Function entered at [<c030d3a8>] from [<c0309784>]
[691238.889829] Function entered at [<c0309784>] from [<c060cd64>]
[691238.895731] Function entered at [<c060cd64>] from [<c0322744>]
[691238.901633] Function entered at [<c0322744>] from [<c0322820>]
[691238.907535] Function entered at [<c0322820>] from [<bf4963ac>]
[691238.913486] Function entered at [<bf4963ac>] from [<bf46210c>]
[691238.919344] Function entered at [<bf46210c>] from [<bf462474>]
[691238.925246] Function entered at [<bf462474>] from [<bf41a0b4>]
[691238.931162] Function entered at [<bf41a0b4>] from [<c084cd78>]
[691238.937050] Function entered at [<c084cd78>] from [<c084be48>]
[691238.942953] Function entered at [<c084be48>] from [<c084c484>]
[691238.948853] Function entered at [<c084c484>] from [<c084b510>]
[691238.954755] Function entered at [<c084b510>] from [<c084b7e4>]
[691238.960659] Function entered at [<c084b7e4>] from [<c07b5918>]
[691238.966563] Function entered at [<c07b5918>] from [<c07b74ec>]
[691238.972464] Function entered at [<c07b74ec>] from [<c07b7654>]
[691238.978368] Function entered at [<c07b7654>] from [<c0300040>]
[691238.984270] Exception stack(0xc32cffa8 to 0xc32cfff0)
[691238.990178] ffa0:                   00000000 00000000 0000000e bee9a938 00000000 00000000
[691238.995391] ffc0: 00000000 00000000 020a0820 00000128 00000004 bee9a980 00000001 00000000
[691239.003634] ffe0: bee9a8e0 bee9a8d0 b6ef8b9c b6ef800c
[691239.011942] ---[ end trace 9a54bcd53eb578dc ]---

### Expected behaviour

[... 527 lines cut ...]
[691238.954755] Function entered at [<c084b510>] from [<c084b7e4>]
[691238.960659] Function entered at [<c084b7e4>] from [<c07b5918>]
[691238.966563] Function entered at [<c07b5918>] from [<c07b74ec>]
[691238.972464] Function entered at [<c07b74ec>] from [<c07b7654>]
[691238.978368] Function entered at [<c07b7654>] from [<c0300040>]
[691238.984270] Exception stack(0xc32cffa8 to 0xc32cfff0)
[691238.990178] ffa0:                   00000000 00000000 0000000e bee9a938 00000000 00000000
[691238.995391] ffc0: 00000000 00000000 020a0820 00000128 00000004 bee9a980 00000001 00000000
[691239.003634] ffe0: bee9a8e0 bee9a8d0 b6ef8b9c b6ef800c
[691239.011942] ---[ end trace 9a54bcd53eb578dc ]---
[691243.691634] br-lan: port 8(phy0-ap0.sta1) entered disabled state
[691243.743194] device phy0-ap0.sta1 left promiscuous mode
[691243.743360] br-lan: port 8(phy0-ap0.sta1) entered disabled state
[691251.169861] br-lan: port 8(phy0-ap0.sta1) entered blocking state
[691251.169924] br-lan: port 8(phy0-ap0.sta1) entered disabled state
[691251.175341] device phy0-ap0.sta1 entered promiscuous mode
[691251.181376] br-lan: port 8(phy0-ap0.sta1) entered blocking state
[691251.186437] br-lan: port 8(phy0-ap0.sta1) entered forwarding state
[716682.002791] ath10k_ahb a000000.wifi: mac flush vdev 1 drop 0 queues 0x2 ar->paused: 0x0  arvif->paused: 0x0
[976698.777166] ath10k_ahb a000000.wifi: mac flush vdev 1 drop 0 queues 0x2 ar->paused: 0x0  arvif->paused: 0x0
[1037340.979775] ath10k_ahb a000000.wifi: mac flush vdev 1 drop 0 queues 0x2 ar->paused: 0x0  arvif->paused: 0x0

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

NPCuse, 2023-11-23:
can you share your /etc/config/network

xback, 2023-11-23:
Already fixed with [this PR](https://github.com/openwrt/openwrt/pull/13620)
I've tested it and it works.

The solution has also already been sent to upstream.
As soon as they approve it, it should get merged very fast.

If you want to try it yourself, feel free to build an image using my [staging tree](https://git.openwrt.org/?p=openwrt/staging/xback.git;a=summary), which contains this fix

xback, 2023-11-23:
Duplicate: #13620

ynezz, 2024-01-12:
> Duplicate: #13620

@xback feel free to close such issues directly next time, thanks! :)

ynezz, 2024-01-12:
Ah, @xback this is not a duplicate actually, this is for 23.05 release with 6.1 based backports (the original fix was for 6.5 backports), but I'm quite puzzled since those commits in the fixes tags:

* Fixes: 0b75a1b1e42e ("wifi: mac80211: flush queues on STA removal")
* Fixes: d00800a289c9 ("wifi: mac80211: add flush_sta method") 

are both from 6.4 kernel, so there must be something else in 6.1 causing this issue and fixed, @oldium any idea?

oldium, 2024-01-16:
@ynezz https://github.com/openwrt/openwrt/blob/openwrt-23.05/package/kernel/mac80211/patches/subsys/331-wifi-mac80211-flush-queues-on-STA-removal.patch

oldium, 2024-01-16:
The problematic patch _wifi: mac80211: flush queues on STA removal_ is part of [23.05](https://github.com/openwrt/openwrt/blob/openwrt-23.05/package/kernel/mac80211/patches/subsys/331-wifi-mac80211-flush-queues-on-STA-removal.patch) and also [22.03](https://github.com/openwrt/openwrt/blob/openwrt-22.03/package/kernel/mac80211/patches/subsys/348-wifi-mac80211-flush-queues-on-STA-removal.patch) and [21.02](https://github.com/openwrt/openwrt/blob/openwrt-21.02/package/kernel/mac80211/patches/subsys/411-wifi-mac80211-flush-queues-on-STA-removal.patch),

guidosarducci, 2025-09-20:
@ynezz Could you please update this with a `release/24.10.2` tag? It continues to happen even now, and it would be nice to address it in the next update if possible.

```
root@AP8302753657:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='24.10.2'
DISTRIB_REVISION='r28739-d9340319c6'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 24.10.2 r28739-d9340319c6'
DISTRIB_TAINTS=''
```

```
[2972603.790843] ath10k_ahb a000000.wifi: Spurious quick kickout for STA e0:9d:31:2e:f6:18
[2972604.337199] ath10k_ahb a000000.wifi: Spurious quick kickout for STA e0:9d:31:2e:f6:18
[2972604.821046] ath10k_ahb a000000.wifi: Spurious quick kickout for STA e0:9d:31:2e:f6:18
[2972605.373794] ath10k_ahb a000000.wifi: Spurious quick kickout for STA e0:9d:31:2e:f6:18
[2972605.845425] ath10k_ahb a000000.wifi: Spurious quick kickout for STA e0:9d:31:2e:f6:18
[2972606.391157] ath10k_ahb a000000.wifi: Spurious quick kickout for STA e0:9d:31:2e:f6:18
[2972606.897552] ath10k_ahb a000000.wifi: Spurious quick kickout for STA e0:9d:31:2e:f6:18
[3813340.256474] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[3813340.358883] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[3813340.461266] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[3813340.563671] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[3813340.666075] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[3813340.768475] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[3813340.870878] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[3813340.973271] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[3813341.075677] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[3813341.178076] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[3813343.241614] ieee80211 phy0: Hardware restart was requested
[3813345.604303] ath10k_warn: 21 callbacks suppressed
[3813345.604334] ath10k_pci 0000:01:00.0: Unknown eventid: 36933
[3813345.852734] ------------[ cut here ]------------
[3813345.852788] WARNING: CPU: 1 PID: 9857 at backports-6.12.6/net/mac80211/driver-ops.c:327 0xbf590168 [mac80211@5ceddda7+0x77000]
[3813345.856727] phy0-ap0.sta1: Failed check-sdata-in-driver check, flags: 0x1
[3813345.868386] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount ath10k_pci(O) ath10k_core(O) ath(O) xt_state xt_helper xt_conntrack xt_connmark xt_connbytes xt_CT wireguard pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat nf_flow_table_inet nf_flow_table nf_conntrack_netlink nf_conntrack mac80211(O) libchacha20poly1305 ipt_REJECT curve25519_neon cfg80211(O) xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbhid ts_kmp ts_fsm ts_bm slhc sch_cake poly1305_arm nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nf_tables nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic iptable_mangle iptable_filter ipt_ECN ip_tables hid_generic crc_ccitt compat(O) chacha_neon act_ipt sch_teql sch_sfq sch_multiq sch_gred
[3813345.868929]  sch_fq sch_ets sch_codel em_text em_nbyte em_meta em_cmp act_skbmod act_simple act_pedit act_csum libcrc32c em_ipset cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact hid evdev input_core ledtrig_usbport xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ifb dummy ip6_udp_tunnel udp_tunnel sit tunnel4 ip_tunnel sha512_arm kpp ghash_arm_ce cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug(O) crc32c_generic
[3813346.017031] CPU: 1 PID: 9857 Comm: kworker/1:1 Tainted: G           O       6.6.93 #0
[3813346.038391] Hardware name: Generic DT based system
[3813346.046202] Workqueue: events_freezable 0xbf58b158 [mac80211@5ceddda7+0x77000]
[3813346.051417] Function entered at [<c030d1d8>] from [<c0308ec8>]
[3813346.058785] Function entered at [<c0308ec8>] from [<c0a5208c>]
[3813346.064774] Function entered at [<c0a5208c>] from [<c03216a4>]
[3813346.070763] Function entered at [<c03216a4>] from [<c03217c4>]
[3813346.076756] Function entered at [<c03217c4>] from [<bf590168>]
[3813346.082781] Function entered at [<bf590168>] from [<bf5cc334>]
[3813346.088734] Function entered at [<bf5cc334>] from [<bf58b238>]
[3813346.094721] Function entered at [<bf58b238>] from [<c0339284>]
[3813346.100712] Function entered at [<c0339284>] from [<c0339fb4>]
[3813346.106703] Function entered at [<c0339fb4>] from [<c0342580>]
[3813346.112691] Function entered at [<c0342580>] from [<c030012c>]
[3813346.118679] Exception stack(0xc4451fb0 to 0xc4451ff8)
[3813346.124671] 1fa0:                                     00000000 00000000 00000000 00000000
[3813346.129626] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[3813346.137956] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[3813346.146938] ---[ end trace 0000000000000000 ]---
[3813346.158768] ath10k_pci 0000:01:00.0: device successfully recovered
```